Joshua-Riek/ubuntu-rockchip #809: Orange PI 5 plus with official Orange Pi 8852BE wifi/bt is not creating the hci0 Bluetooth device

teststation5, 2024-05-18:
I am a big fan of your images.  The hardware accelerated ffmeg is critical for my project and works very well.  I have a 22.04 server build working flawlessly on an Orange Pi 5B.  I am now trying to get the same setup with an Orange PI plus and the official bt/wifi m2 card because my project my require dual wired nics in some configurations  The wifi for the card appears to work fine, but the Bluetooth device does not appear to be recognized at any point in the boot process and when I go to bring up the hci0 device, it cannot be found.  I have tried the latest 22.04 and 24.04 builds and a few assorted older ones, but the issue is consistent in all of them.  I did try the Orange PI official Ubuntu Sammy release and the Bluetooth does show up so I know the hardware works and can be loaded into ubuntu.  I can't use the Orange PI official image because it does not have the hardware accelerated ffmeg so I have to figure out the missing piece to get the Bluetooth working with these images.  I captured some key diagnostic from the failing 24.04 image and then the same commands on the Orange PI ubuntu for comparison.  Any tips on what I could do to get it working or even where to start looking would be greatly appreciated.

ubuntu@ubuntu:~$ cat /proc/version
Linux version 6.1.0-1013-rockchip (buildd@bos01-arm64-013) (aarch64-linux-gnu-gcc (Ubuntu 13.2.0-23ubuntu4) 13.2.0, GNU ld (GNU Binutils for Ubuntu) 2.42) #13-Ubuntu SMP Thu May  9 21:20:56 UTC 2024

ubuntu@ubuntu:~$ dmesg | grep -e tooth -e 8852 -e btusb
[   10.666012] Bluetooth: Core ver 2.22
[   10.666042] Bluetooth: HCI device and connection manager initialized
[   10.666054] Bluetooth: HCI socket layer initialized
[   10.666063] Bluetooth: L2CAP socket layer initialized
[   10.666075] Bluetooth: SCO socket layer initialized
[   11.837625] [BT_RFKILL]: bluetooth_platdata_parse_dt: get property: uart_rts_gpios = 148.
[   11.838094] [BT_RFKILL]: bluetooth_platdata_parse_dt: get property: BT,reset_gpio = 21.
[   11.838543] [BT_RFKILL]: bluetooth_platdata_parse_dt: get property: BT,wake_gpio = 150.
[   11.838986] [BT_RFKILL]: bluetooth_platdata_parse_dt: get property: BT,wake_host_irq = 0.
[   16.560671] rtw89_8852be 0002:21:00.0: loaded firmware rtw89/rtw8852b_fw-1.bin
[   16.560821] rtw89_8852be 0002:21:00.0: enabling device (0000 -> 0003)
[   16.567470] rtw89_8852be 0002:21:00.0: Firmware version 0.29.29.5 (da87cccd), cmd version 0, type 5
[   16.567484] rtw89_8852be 0002:21:00.0: Firmware version 0.29.29.5 (da87cccd), cmd version 0, type 3
[   16.846646] rtw89_8852be 0002:21:00.0: chip rfe_type is 1
[   16.887442] rtw89_8852be 0002:21:00.0 wlP2p33s0: renamed from wlan0
[   26.580909] rtw89_8852be 0002:21:00.0: loaded firmware rtw89/rtw8852b_fw-1.bin
[   26.585408] rtw89_8852be 0002:21:00.0: Firmware version 0.29.29.5 (da87cccd), cmd version 0, type 5
[   26.585428] rtw89_8852be 0002:21:00.0: Firmware version 0.29.29.5 (da87cccd), cmd version 0, type 3
[   26.870326] rtw89_8852be 0002:21:00.0: chip rfe_type is 1
[   26.902201] rtw89_8852be 0002:21:00.0 wlP2p33s0: renamed from wlan0

ubuntu@ubuntu:~$ hciconfig -a
**empty response

ubuntu@ubuntu:~$ lsmod
Module                  Size  Used by
qrtr                   36864  4
rtw89_8852be           16384  0
rtw89_8852b           282624  1 rtw89_8852be
rtw89_pci              45056  1 rtw89_8852be
binfmt_misc            20480  1
rtw89_core            303104  2 rtw89_8852b,rtw89_pci
panfrost               61440  0
pwm_fan                20480  0
goodix_ts              28672  0
input_leds             16384  0
drm_shmem_helper       20480  1 panfrost
gpu_sched              32768  1 panfrost
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
squashfs               53248  4
dm_multipath           28672  0
fuse                  114688  1
dm_mod                110592  3 dm_multipath
nfnetlink              20480  2
ip_tables              28672  0
ipv6                  471040  32
raid10                 53248  0
raid456               143360  0
async_raid6_recov      16384  1 raid456
async_memcpy           16384  1 raid456
async_pq               16384  1 raid456
async_xor              16384  2 async_pq,raid456
async_tx               16384  5 async_pq,async_memcpy,async_xor,raid456,async_raid6_recov
raid1                  40960  0
raid0                  20480  0
multipath              16384  0
linear                 16384  0
md_mod                135168  6 raid1,raid10,raid0,linear,raid456,multipath
r8169                  77824  0
adc_keys               16384  0
pwm_bl                 16384  0

ubuntu@ubuntu:~$ lsusb
[... 64 lines cut ...]
btintel                40960  1 btusb
rtw89_8852b           303104  1 rtw89_8852be
bluetooth             733184  29 btrtl,btmtk,btintel,btbcm,bnep,btusb
rtw89_pci              53248  1 rtw89_8852be
rtw89_core            270336  2 rtw89_8852b,rtw89_pci
pwm_fan                24576  0
sch_fq_codel           20480  3
fuse                  126976  1
ip_tables              32768  0
r8169                  86016  0

orangepi@orangepi5plus:~$ lsusb
Bus 002 Device 002: ID 05e3:0620 Genesys Logic, Inc. USB3.2 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0bda:b85b Realtek Semiconductor Corp. Bluetooth Radio
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 005 Device 003: ID 060b:0540 Solid Year DeltaCo TB-106U Keyboard
Bus 005 Device 002: ID 1a40:0801 Terminus Technology Inc. USB 2.0 Hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

teststation5, 2024-05-22:
I was able to narrow the start of this issue to somewhere between builds 1.27 and 1.28.   I was also able to rule out the rtl8852 reload service that was added, so it seems it must be something in the rather large kernel update from 5.10.160-13 to 5.10.160-15.  I also tried a test of 1.27 and 1.28 with an Intel ATX210 and the bluetooth works with that card in 1.27 and not in 1.28 so it doesn't seem to be specific to the realtek driver, but more likely the usb connection to the bluetooth.   Does any of that help narrow down the issue?

Joshua-Riek, 2024-05-22:
I know this works, the Orange Pi 5 Plus was used during the entire development process of Ubuntu 24.04.

closed 2024-05-22 by Joshua-Riek

teststation5, 2024-05-23:
Well, what I could I possibly be doing wrong that the hci0 device is not recognized in 22.04 1.28 and every build after that I tried and is recognized in 22.04 1.27 and every build before that support orange pi 5 plus?   I took the downloaded images did nothing to them other than sudo apt update and sudo apt upgrade.  In the attached dmesg, one has an init of hci0 and the other clearly does not.   hciconfig -a shows a device in 1.28 and shows nothing in 1.27.  

[dmesg.22.04.04.1.28.log](https://github.com/Joshua-Riek/ubuntu-rockchip/files/15409972/dmesg.22.04.04.1.28.log)
[dmesg.22.04.04.1.27.log](https://github.com/Joshua-Riek/ubuntu-rockchip/files/15409973/dmesg.22.04.04.1.27.log)

teststation5, 2024-05-23:
So doing a compare of those two dmesg logs, I noticed some logs from the BT_RFKILL process on the working 1.27 image that were not on the 1.28 image.  I have no experience with rfkill so I researched a bit and tried just a empty rfkill from the command prompt of the 1.28 build and returned this:

ubuntu@ubuntu:~$ sudo rfkill
ID TYPE      DEVICE          SOFT      HARD
 0 bluetooth bt_default   blocked unblocked
 1 wlan      phy0       unblocked unblocked

hmmm... looks like maybe bluetooth is soft blocked?  so I tried

ubuntu@ubuntu:~$ sudo rfkill unblock 0

and, what do you know, the hci0 device appears!!

ubuntu@ubuntu:~$ hciconfig -a
hci0:   Type: Primary  Bus: USB
        BD Address: 40:9C:A7:66:EF:5D  ACL MTU: 1021:6  SCO MTU: 255:12
        UP RUNNING
        RX bytes:1852 acl:0 sco:0 events:209 errors:0
        TX bytes:41829 acl:0 sco:0 commands:209 errors:0
        Features: 0xff 0xff 0xff 0xfe 0xdb 0xfd 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
        Link policy: RSWITCH HOLD SNIFF PARK
        Link mode: PERIPHERAL ACCEPT
        Name: 'ubuntu'
        Class: 0x000000
        Service Classes: Unspecified
        Device Class: Miscellaneous,
        HCI Version: 5.2 (0xb)  Revision: 0xdfb7
        LMP Version: 5.2 (0xb)  Subversion: 0x91cb
        Manufacturer: Realtek Semiconductor Corporation (93)

So I guess the moral to the story is that something in the images from 1.28  are either rfkill soft blocking the bluetooth device and will require an unblock from a startup service or something in the 1.27 images and lower are soft unblocking bluetooth.  

I am curious as to which and why, but whatever, I think I can get things working now!

aledomu, 2024-12-28:
Apart from the braindead rfkill issue, there is yet some problem with the device module initialization order. If I unblock `bt_default` like @teststation5 did, everything is fine. But if I restart, WiFi is not even recognized. I have to softblock `bt_default` again and then restart to get WiFi back. Even worse: this does not always happen, although the "fix" does work always.

aledomu, 2025-01-22:
This still fails even in Linux 6.1.84, vendor version. Now the difference is that I have to block both the `hci0` device and `bt_default`.