Joshua-Riek/ubuntu-rockchip #813: I was unable to boot version 24.04 on the orange pi 5 installed on the SSD (M.2 SATA)

douglasdock, 2024-05-18:
I was unable to boot version 24.04 on the orange pi 5 installed on the SSD (M.2 SATA)

FreeWilly2022, 2024-05-19:
Based on the image you are using, I assume you have a Orange Pi 5 base model. If you don't, then you are using the wrong image. This is Ubuntu, not Debian, so I don't understand that part of your question. The OPi5 has a SPI flash, so you need to copy the image to the SSD using:

sudo ubuntu-rockchip-install /dev/nvme0n1

and u-boot to the SPI flash using:

sudo u-boot-install-mtd

douglasdock, 2024-05-19:
Hello friend, thanks for the answer and I believe I understood.
I'll try to fix it here and I'll give you feedback

douglasdock, 2024-05-19:
Hello, I installed "ubuntu 24.04" on the SD card and now I wanted to boot to SSDM2, however when I execute the command:

sudo ubuntu-rockchip-install /dev/nvme0n1

It is returned that it does not have the SSD, with the following information:

"/usr/bin/ubuntu-rockchip-install: block device '/dev/nvme0n1' not found"

I installed "GParted" and I can't find the SSD, on the company's official "Debian" system I know how to activate the SSD, but here I don't have the "Orangepi-config" command.

Would anyone know how to activate the SSD?

Thank you very much

Note: congratulations on the work

thanhtantran, 2024-06-03:
Hello,

Orange Pi 5 default support M.2 NVME then if you use M.2 SATA the system will not recognize it.

douglasdock, 2024-06-04:
Hello everything is fine?
It is possible to use the M.2 SSD.
It was discussed here:
https://github.com/Joshua-Riek/ubuntu-rockchip/issues/817

closed 2024-06-04 by douglasdock