Joshua-Riek/ubuntu-rockchip #814: The USB3.0 port of the orange pi 5 pro did not work  with ubuntu24.04

ugawahm, 2024-05-19:
Device: Orange pi 5 pro
Image: Ubuntu-24.04-preinstalled-desktop-arm64-orangepi-5-pro.img.xz
Problem: One of the four USB ports is not working properly, which is the USB3.0 port.

FreeWilly2022, 2024-05-19:
@ugawahm can you please describe what "not working properly" means specifically, and maybe include the results of the lsusb command if that is relevant. For example what did you connect to that port, how did it not work, is there any other issues/conditions that could have affected the port?

ugawahm, 2024-05-20:
@FreeWilly2022 
Here are the results of the lsusb command in different  images.(Orangepi5pro_1.0.2_debian_bookworm_desktop_xfce_linux5.10.160 and Ubuntu-24.04-preinstalled-desktop-arm64-orangepi-5-pro)
[image]
[image]
T be specific，when I inserted the device (such as U disk, keyboard and mouse) into the USB 3.0 port, nothing happened.

bagong, 2024-06-13:
I just set up a OPI+ system with 6.1.0.1016, The USB3 port does work.

````
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 002: ID 05e3:0620 Genesys Logic, Inc. GL3523 Hub
Bus 002 Device 003: ID 090c:1000 Silicon Motion, Inc. - Taiwan (formerly Feiya Technology Corp.) Flash Drive
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver
````
I haven't tested it extensively, but you can copy files, and it seems to be quite fast.

Two remarks however: coincidentally I had my keyboard-dongle in the neighboring USB slot, and once I had inserted the USB-drive into the slot below, the keyboard/mouse contact was interrupted. I could pull it out and insert the dongle into one of the back USB-ports and the connection was reestablished. However it seems there is some interference, as the mouse drags noticeably when some other USB-activity goes on. I've noticed that in the past on the 5.1 kernel, but didn't bother. I think however one might want to keep the issue at the back of the head as it seems to point to some deeper issue, not fatal, but irritating.

The other question is whether to enable USB boot is planned/in reach?

Best
Rainer

marcoexpo, 2024-07-03:
Ho everyone, I am encountering the same issue on my Orange pi 5 pro with Ubuntu 24.04. Dis someone solve the problem? Thank you for your help!

Joshua-Riek, 2024-07-03:
Please update your system with apt, i fixed this with the commit https://github.com/Joshua-Riek/linux-rockchip/commit/cc972a2683c329ad8454148134829123dc1c92e2

closed 2024-07-03 by Joshua-Riek