Joshua-Riek/ubuntu-rockchip #815: orange pi 5 plus use kodi  ，When playing 4k, the screen goes black, but the sound and subtitles are displayed normally

dengqz, 2024-05-19:
orange pi 5 plus use kodi  ，When playing 4k, the screen goes black, but the sound and subtitles are displayed normally
[image]

ubuntu24

dengqz, 2024-05-19:
[image]

dengqz, 2024-05-19:
[image]

dengqz, 2024-05-19:
[image]

dengqz, 2024-05-19:
[image]

dengqz, 2024-05-19:
[image]

dengqz, 2024-05-19:
My monitor has a 1080p resolution of 60hz     
orange pi plus 32G    
SD install
HDMI 2.0 Wire

nguyenhunga5, 2024-05-21:
@dengqz I using the patch in this tutorial and now Kodi fully working with 4k, h265.
You can try to build Kodi with the apply patch from the link bellow.

https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Rendering

kkein, 2024-05-21:
very good!can you please give more details?

nguyenhunga5, 2024-05-22:
> very good!can you please give more details?

Yes, there is my steps:

1. Get the source code
- Make sure git is installed:
sudo apt install git
- Change to your home directory:
cd $HOME
- Clone Kodi's current master branch:
git clone https://github.com/xbmc/xbmc kodi

2. Get build dependencies manually
- sudo apt install ccache
- sudo apt install distcc
- sudo apt install debhelper autoconf automake autopoint gettext autotools-dev cmake curl default-jre doxygen gawk gcc gdc gperf libasound2-dev libass-dev libavahi-client-dev libavahi-common-dev libbluetooth-dev libbluray-dev libbz2-dev libcdio-dev libp8-platform-dev libcrossguid-dev libcurl4-openssl-dev libcwiid-dev libdbus-1-dev libdrm-dev libegl1-mesa-dev libenca-dev libflac-dev libfmt-dev libfontconfig-dev libfreetype6-dev libfribidi-dev libfstrcmp-dev libgcrypt-dev libgif-dev libgles2-mesa-dev libgl1-mesa-dev libglu1-mesa-dev libgnutls28-dev libgpg-error-dev libgtest-dev libiso9660-dev libjpeg-dev liblcms2-dev libltdl-dev liblzo2-dev libmicrohttpd-dev libmysqlclient-dev libnfs-dev libogg-dev libpcre3-dev libplist-dev libpng-dev libpulse-dev libshairplay-dev libsmbclient-dev libspdlog-dev libsqlite3-dev libssl-dev libtag1-dev libtiff5-dev libtinyxml-dev libtinyxml2-dev libtool libudev-dev libunistring-dev libva-dev libvdpau-dev libvorbis-dev libxmu-dev libxrandr-dev libxslt1-dev libxt-dev lsb-release meson nasm ninja-build python3-dev python3-pil python3-pip rapidjson-dev swig unzip uuid-dev zip zlib1g-dev
- sudo apt install libcec-dev libfmt-dev liblirc-dev
- sudo apt install libflatbuffers-dev
- sudo apt install libglew-dev libwayland-dev libxkbcommon-dev waylandpp-dev wayland-protocols
- sudo apt install libgbm-dev libinput-dev libxkbcommon-dev
- sudo apt install doxygen libcap-dev libsndio-dev libmariadbd-dev
- sudo apt install libtexturePacker-dev
- sudo apt install libcap-dev
- sudo apt install libcec-dev
- sudo apt install libavcodec-dev libavfilter-dev libavformat-dev libavutil-dev libswscale-dev libswresample-dev libpostproc-dev
- sudo apt install libdav1d-dev
- sudo apt install mpv ffmpeg

3. Follow this link to config build and build https://github.com/xbmc/xbmc/blob/master/docs/README.Linux.md#4-build-kodi
- My build config build: 
   - mkdir $HOME/kodi-build
   - cd $HOME/kodi-build
   - cmake ../kodi -DCMAKE_INSTALL_PREFIX=/usr/local -DCORE_PLATFORM_NAME="x11 wayland gbm" -DAPP_RENDER_SYSTEM=gles -B.
   - cmake --build . -- VERBOSE=1 -j$(getconf _NPROCESSORS_ONLN)
   - sudo usermod -a -G input,video <username>
   - sudo make install

4. After that you need logout and login with Kodi session.
5. This is my setting:
[image]
6. Result
[image]
[image]

nguyenhunga5, 2024-05-22:
> kodi is allready patched, the user is likely trying to play HDR videos which do not work

It's strange, Kodi working with hdr10 on my side.
https://youtu.be/ejPhwgw9R6g

dengqz, 2024-05-23:
> > very good!can you please give more details?
> 
> Yes, there is my steps:
> 
> 1. Get the source code
> 
> * Make sure git is installed:
>   sudo apt install git
> * Change to your home directory:
>   cd $HOME
> * Clone Kodi's current master branch:
>   git clone https://github.com/xbmc/xbmc kodi
> * Download patchs:
>   
>   * Goto home dir: cd $HOME
>   * wget https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/patches/kodi/0001-windowing-gbm-Dynamic-plane-selection.patch
>   * wget https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/patches/kodi/0002-VideoLayerBridgeDRMPRIME-Use-crop-fields-to-render-t.patch
>   * Apply patchs
>     
>     * cd $HOME/kodi
>     * git apply ~/0001-windowing-gbm-Dynamic-plane-selection.patch
>     * git apply ~/0002-VideoLayerBridgeDRMPRIME-Use-crop-fields-to-render-t.patch
> 
> 2. Get build dependencies manually
> 
> * sudo apt install ccache
> * sudo apt install distcc
> * sudo apt install debhelper autoconf automake autopoint gettext autotools-dev cmake curl default-jre doxygen gawk gcc gdc gperf libasound2-dev libass-dev libavahi-client-dev libavahi-common-dev libbluetooth-dev libbluray-dev libbz2-dev libcdio-dev libp8-platform-dev libcrossguid-dev libcurl4-openssl-dev libcwiid-dev libdbus-1-dev libdrm-dev libegl1-mesa-dev libenca-dev libflac-dev libfmt-dev libfontconfig-dev libfreetype6-dev libfribidi-dev libfstrcmp-dev libgcrypt-dev libgif-dev libgles2-mesa-dev libgl1-mesa-dev libglu1-mesa-dev libgnutls28-dev libgpg-error-dev libgtest-dev libiso9660-dev libjpeg-dev liblcms2-dev libltdl-dev liblzo2-dev libmicrohttpd-dev libmysqlclient-dev libnfs-dev libogg-dev libpcre3-dev libplist-dev libpng-dev libpulse-dev libshairplay-dev libsmbclient-dev libspdlog-dev libsqlite3-dev libssl-dev libtag1-dev libtiff5-dev libtinyxml-dev libtinyxml2-dev libtool libudev-dev libunistring-dev libva-dev libvdpau-dev libvorbis-dev libxmu-dev libxrandr-dev libxslt1-dev libxt-dev lsb-release meson nasm ninja-build python3-dev python3-pil python3-pip rapidjson-dev swig unzip uuid-dev zip zlib1g-dev
> * sudo apt install libcec-dev libfmt-dev liblirc-dev
> * sudo apt install libflatbuffers-dev
> * sudo apt install libglew-dev libwayland-dev libxkbcommon-dev waylandpp-dev wayland-protocols
> * sudo apt install libgbm-dev libinput-dev libxkbcommon-dev
> * sudo apt install doxygen libcap-dev libsndio-dev libmariadbd-dev
> * sudo apt install libtexturePacker-dev
> * sudo apt install libcap-dev
> * sudo apt install libcec-dev
> * sudo apt install libavcodec-dev libavfilter-dev libavformat-dev libavutil-dev libswscale-dev libswresample-dev libpostproc-dev
> * sudo apt install libdav1d-dev
> * sudo apt install mpv ffmpeg
> 
> 3. Follow this link to config build and build https://github.com/xbmc/xbmc/blob/master/docs/README.Linux.md#4-build-kodi
> 
> * My build config build:
>   
>   * mkdir $HOME/kodi-build
>   * cd $HOME/kodi-build
>   * cmake ../kodi -DCMAKE_INSTALL_PREFIX=/usr/local -DCORE_PLATFORM_NAME="x11 wayland gbm" -DAPP_RENDER_SYSTEM=gles -B.
>   * cmake --build . -- VERBOSE=1 -j$(getconf _NPROCESSORS_ONLN)
>   * sudo usermod -a -G input,video
>   * sudo make install
> 
> 4. After that you need logout and login with Kodi session.
> 5. This is my setting:
>    ![IMG_4528](https://private-user-images.githubusercontent.com/853275/332636274-6b559c03-3080-4406-aacb-6ac30da2a5e3.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY0Mjc1NDUsIm5iZiI6MTcxNjQyNzI0NSwicGF0aCI6Ii84NTMyNzUvMzMyNjM2Mjc0LTZiNTU5YzAzLTMwODAtNDQwNi1hYWNiLTZhYzMwZGEyYTVlMy5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUyMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MjNUMDEyMDQ1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NWJjMDVkNDdiY2Q2ZWU2OWMwNWQ0MGE0ZGFjZWY2OGNmZTAzMzFkMTExZDVlN2M3ZTZlMWRjMjhhYjY2NGE3OSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.I8-cpsUg_9DsUi6ya5uHIQqdB0Sx39clw1yznGSGIg8)
> 6. Result
>    ![IMG_4529](https://private-user-images.githubusercontent.com/853275/332636418-43755f91-a68c-4d82-8c39-e9a42b9da128.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY0Mjc1NDUsIm5iZiI6MTcxNjQyNzI0NSwicGF0aCI6Ii84NTMyNzUvMzMyNjM2NDE4LTQzNzU1ZjkxLWE2OGMtNGQ4Mi04YzM5LWU5YTQyYjlkYTEyOC5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUyMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MjNUMDEyMDQ1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YmZiMjE1ZjZlMzkwN2Y4NjZmZDhiOTk0NzM0ZGU1MjdkOGY2OWUzZDc3NTJkMjU5NGZiOTFmMThiNzg2YmEyOSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.kZXYBj3QiZvWrNJb8mDqnWb0eRWXSRiNEJaJnhJLUHE)
>    ![IMG_4530](https://private-user-images.githubusercontent.com/853275/332636438-02489241-bb6b-4711-8dc3-c42177468fe6.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY0Mjc1NDUsIm5iZiI6MTcxNjQyNzI0NSwicGF0aCI6Ii84NTMyNzUvMzMyNjM2NDM4LTAyNDg5MjQxLWJiNmItNDcxMS04ZGMzLWM0MjE3NzQ2OGZlNi5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUyMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MjNUMDEyMDQ1WiZYLUFte [... cut ...]
The previous video can be played but there is no sound； and after playing for a few seconds, there was no picture again；Black screen again；I'm not sure if the reason is due to not installing libtexturePacker dev

Video files are processed using the NFS protocol
I installed the system and executed Kodi compilation without doing any other operations；

I plan to try compiling again；

Here are my hardware and environment：
orange pi 5 plus 32G
[ubuntu-24.04-preinstalled-desktop-arm64-orangepi-5-plus.img.xz](https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v2.1.0/ubuntu-24.04-preinstalled-desktop-arm64-orangepi-5-plus.img.xz) Ubuntu 24.04 LTS Desktop with Linux 6.1

dengqz, 2024-05-23:
> kodi is allready patched, the user is likely trying to play HDR videos which do not work

Thank you very much for your reply.
I plan to try compiling again.
Following the steps of nguyenhunga5, it seems that it can be played and there is a screen display, but there are still some issues

nguyenhunga5, 2024-05-23:
> I tried following your steps, but it still hasn't been completely resolved.
>
> `sudo apt install libtexturePacker-dev` When executing this step, the dependency cannot be found；In the end, I didn't [...]

Sorry I forgot add this command:
- sudo apt install libdisplay-info-dev

about `sudo apt install libtexturePacker-dev` this is my mistake :D, we no need to install it. It will build when we call build Kodi source.

nguyenhunga5, 2024-05-23:
- About sound, this is my setting
[image]
- Login setting:
[image]

dengqz, 2024-05-25:
> * About sound, this is my setting
>   ![IMG_4537](https://private-user-images.githubusercontent.com/853275/333009783-06afa07e-7fce-4b9f-a397-0ecbf64935ea.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY2NTQwMjMsIm5iZiI6MTcxNjY1MzcyMywicGF0aCI6Ii84NTMyNzUvMzMzMDA5NzgzLTA2YWZhMDdlLTdmY2UtNGI5Zi1hMzk3LTBlY2JmNjQ5MzVlYS5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MjVUMTYxNTIzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NWU5MGM2NjI4ZjI0Mjc5ZTBhNDg0YTNkMmRlZTYwY2Y0MDYwZjIyN2NmOGJmNGQ4M2RjOTU4ZjEyYTk4NThjMCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.i8LjtqoaPzQJDZVksgdSRBPOH6fgR99BcG3TBMxNyk0)
> * Login setting:
>   ![IMG_4536](https://private-user-images.githubusercontent.com/853275/333009748-c33bb383-5524-4b1c-8789-f5f085686344.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY2NTQwMjMsIm5iZiI6MTcxNjY1MzcyMywicGF0aCI6Ii84NTMyNzUvMzMzMDA5NzQ4LWMzM2JiMzgzLTU1MjQtNGIxYy04Nzg5LWY1ZjA4NTY4NjM0NC5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MjVUMTYxNTIzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTQ4ZmRiZmQxMTk4ODZiNjM3YzljMDlhZjcyYjUwODc2N2Q2YmNmMjU5OWU5OTA4M2MzMTVmM2M2NzU2YTE3MyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.I3FapYksySECFouB0fZop5nncDbvCNWGddPA28ekwFs)

I compiled version 21 and now it can play normally, but I found that my Xbox 360 controller doesn't seem to recognize it
In GBM mode, I want to use Xbox360 to control it
[image]
[image]

wmport, 2024-05-25:
> my steps:
GOOD !!!

nguyenhunga5, 2024-05-25:
> I compiled version 21 and now it can play normally, but I found that my Xbox 360 controller doesn't seem to recognize it
>
> In GBM mode, I want to use Xbox360 to control it
>
> [image]
>
> [image]
Glad to hear that. I don't have Xbox 360 controller. So I cannot test this case. Maybe you need add user to group like add mouse and keyboard (I guess)

Bluesmanuk, 2024-05-26:
It's great that there seems to be a mostly working Kodi.

I was wondering whether anybody has built a package that could be posted to make for an easy installation.

dengqz, 2024-05-28:
> > > * About sound, this is my setting
> > 
> > 
> > > ![IMG_4537](https://private-user-images.githubusercontent.com/853275/333009783-06afa07e-7fce-4b9f-a397-0ecbf64935ea.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY2NTQwMjMsIm5iZiI6MTcxNjY1MzcyMywicGF0aCI6Ii84NTMyNzUvMzMzMDA5NzgzLTA2YWZhMDdlLTdmY2UtNGI5Zi1hMzk3LTBlY2JmNjQ5MzVlYS5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MjVUMTYxNTIzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NWU5MGM2NjI4ZjI0Mjc5ZTBhNDg0YTNkMmRlZTYwY2Y0MDYwZjIyN2NmOGJmNGQ4M2RjOTU4ZjEyYTk4NThjMCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.i8LjtqoaPzQJDZVksgdSRBPOH6fgR99BcG3TBMxNyk0)
> > 
> > 
> > > * Login setting:
> > 
> > 
> > > ![IMG_4536](https://private-user-images.githubusercontent.com/853275/333009748-c33bb383-5524-4b1c-8789-f5f085686344.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY2NTQwMjMsIm5iZiI6MTcxNjY1MzcyMywicGF0aCI6Ii84NTMyNzUvMzMzMDA5NzQ4LWMzM2JiMzgzLTU1MjQtNGIxYy04Nzg5LWY1ZjA4NTY4NjM0NC5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MjVUMTYxNTIzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTQ4ZmRiZmQxMTk4ODZiNjM3YzljMDlhZjcyYjUwODc2N2Q2YmNmMjU5OWU5OTA4M2MzMTVmM2M2NzU2YTE3MyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.I3FapYksySECFouB0fZop5nncDbvCNWGddPA28ekwFs)
> > 
> > 
> > I compiled version 21 and now it can play normally, but I found that my Xbox 360 controller doesn't seem to recognize it
> > In GBM mode, I want to use Xbox360 to control it
> > ![屏幕截图 2024-05-26 002241](https://private-user-images.githubusercontent.com/22885116/333817570-a539473d-00d1-4053-81c6-1ae00aef7b5f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY4NTk4MTIsIm5iZiI6MTcxNjg1OTUxMiwicGF0aCI6Ii8yMjg4NTExNi8zMzM4MTc1NzAtYTUzOTQ3M2QtMDBkMS00MDUzLTgxYzYtMWFlMDBhZWY3YjVmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI4VDAxMjUxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZlNTJlZDc5MzRmYWJlOWJhZmE3Y2QyNWJkNzY3YWU4MTMxODNmNWExNGVmMDQ0Y2ZjNjFiY2Q1ZjU1NjI0MzcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.plP0M_lbfnUo97GLd6huL54h8z5f2n_AkdpgqTfm7gs)
> > ![屏幕截图 2024-05-26 002540](https://private-user-images.githubusercontent.com/22885116/333817708-0ae38297-3c09-4c2a-ba3f-2fb027db0ad5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY4NTk4MTIsIm5iZiI6MTcxNjg1OTUxMiwicGF0aCI6Ii8yMjg4NTExNi8zMzM4MTc3MDgtMGFlMzgyOTctM2MwOS00YzJhLWJhM2YtMmZiMDI3ZGIwYWQ1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI4VDAxMjUxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNlYzJiODUyNmRhZmY2NjhhNjQzNzBjMzJkYWY1ZDNjOWUzNzI3MWFhMWZjYTYwMmRlZDMzZjFkYjc4ZjNmOTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.h_qNuCzYK8R1_bKQ_TwfnPNa1HRNtiXRm1S0s2p9mcQ)
> 
> Glad to hear that. I don't have Xbox 360 controller. So I cannot test this case. Maybe you need add user to group like add mouse and keyboard (I guess)

 
`cd $HOME/kodi`
`sudo make -j$(getconf _NPROCESSORS_ONLN) -C tools/depends/target/binary-addons PREFIX=/usr/local ADDONS="peripheral.joystick"`
need build add-ons  
now I can use the Xbox 360 controller to control Kodi

dengqz, 2024-05-28:
> It's great that there seems to be a mostly working Kodi.
> 
> I was wondering whether anybody has built a package that could be posted to make for an easy installation.

I am not very familiar with publishing private packages on Ubuntu, and it is possible that Ubuntu rockchip will release new packages in future updates @Joshua-Riek @nguyenhunga5

Joshua-Riek, 2024-05-28:
Packaging is a little complex and I have the above patches already in the included Kodi version, so i have no idea why there is an issue in the first place. But building Kodi from GitHub sources is not the same as building Kodi as a package for Ubuntu. I simply don't have the time right now to debug Kodi or explain the Ubuntu packaging process, but I will eventually update the package.

kkein, 2024-05-28:
@Joshua-Riek  Compile kodi like this way in 2404 LTS v2.1.0,  kodi  works just NOT as good as the default MPV tool of v2.1.0 release. For example :  8K HDR  240 FPS Dolby vision - Special OLED demo ,  MPV plays  OK with low cpu ;  but kodi has voice but no video.   

Would you please check the diff with Kodi and MPV?      Thanks a lot for your so good ubuntu-rockchip release!

Bluesmanuk, 2024-06-01:
Thanks for the replies and I happily accept that packaging is likely quite complex and will take time.

So I followed the instructions to install manually, which seemed to go well apart from the very last part.

After logging off and back on, when I enter my password I am informed that my username is not listed and will not go any further.

Uglymotha, 2024-06-01:
Hi,

Great stuff can you add: libcdio++-dev libiso9660++-dev to de dependencies, otherwise you may have some issues playing old fashioned cd's.

Also, while kodi is displaying HDR content fine it does not seem to detect HDR capable displays correctly. In system info -> video, Display Supported HDR Types is always none and the option to toggle HDR is not present in Player settings. I have tested with two HDR capable displays.

Bluesmanuk, 2024-06-15:
I finally got Kodi running as well as it can under v2.2.1 24.04 (does not work with 22.04)

This is the specific procedure that I followed that pretty much matches what has already been written above but expanded it to encompass what I think is the whole process that I followed.

sudo apt update
sudo apt upgrade
sudo apt install git
cd $HOME
git clone https://github.com/xbmc/xbmc kodi
cd $HOME
wget https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/patches/kodi/0001-windowing-gbm-Dynamic-plane-selection.patch
wget https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/patches/kodi/0002-VideoLayerBridgeDRMPRIME-Use-crop-fields-to-render-t.patch
cd $HOME/kodi
git apply ~/0001-windowing-gbm-Dynamic-plane-selection.patch
git apply ~/0002-VideoLayerBridgeDRMPRIME-Use-crop-fields-to-render-t.patch
sudo apt install ccache
sudo apt install distcc

sudo apt install debhelper autoconf automake autopoint gettext autotools-dev cmake curl default-jre doxygen gawk gcc gdc gperf libasound2-dev libass-dev libavahi-client-dev libavahi-common-dev libbluetooth-dev libbluray-dev libbz2-dev libcdio-dev libp8-platform-dev libcrossguid-dev libcurl4-openssl-dev libcwiid-dev libdbus-1-dev libdrm-dev libegl1-mesa-dev libenca-dev libflac-dev libfmt-dev libfontconfig-dev libfreetype6-dev libfribidi-dev libfstrcmp-dev libgcrypt-dev libgif-dev libgles2-mesa-dev libgl1-mesa-dev libglu1-mesa-dev libgnutls28-dev libgpg-error-dev libgtest-dev libiso9660-dev libjpeg-dev liblcms2-dev libltdl-dev liblzo2-dev libmicrohttpd-dev libmysqlclient-dev libnfs-dev libogg-dev libpcre3-dev libplist-dev libpng-dev libpulse-dev libshairplay-dev libsmbclient-dev libspdlog-dev libsqlite3-dev libssl-dev libtag1-dev libtiff5-dev libtinyxml-dev libtinyxml2-dev libtool libudev-dev libunistring-dev libva-dev libvdpau-dev libvorbis-dev libxmu-dev libxrandr-dev libxslt1-dev libxt-dev lsb-release meson nasm ninja-build python3-dev python3-pil python3-pip rapidjson-dev swig unzip uuid-dev zip zlib1g-dev

sudo apt install libcec-dev libfmt-dev liblirc-dev
sudo apt install libflatbuffers-dev
sudo apt install libglew-dev libwayland-dev libxkbcommon-dev waylandpp-dev wayland-protocols
sudo apt install libgbm-dev libinput-dev libxkbcommon-dev
sudo apt install doxygen libcap-dev libsndio-dev libmariadbd-dev
sudo apt install libcap-dev
sudo apt install libcec-dev
sudo apt install libavcodec-dev libavfilter-dev libavformat-dev libavutil-dev libswscale-dev libswresample-dev libpostproc-dev
sudo apt install libdav1d-dev
sudo apt install mpv ffmpeg
sudo apt install libdisplay-info-dev

mkdir $HOME/kodi-build
cd $HOME/kodi-build
cmake ../kodi -DCMAKE_INSTALL_PREFIX=/usr/local -DCORE_PLATFORM_NAME=wayland -DAPP_RENDER_SYSTEM=gles -B.
cmake --build . -- VERBOSE=1 -j$(getconf _NPROCESSORS_ONLN)
sudo usermod -a -G input,video <your username, no < or > required>
sudo make install
sudo poweroff

Restart the machine
Logoff
Select your user profile, where you will see a little white button at the bottom right of the screen. Click on it.
Select kodi GBM and enter your password.
Kodi will load.
Change your settings as required and additionally, under Player, turn on 'Allow hardware acceleration with DRM Prime and then select Prime Render Method to 'Direct To Plane'
Test video playback.

Note that the next time that you start the machine, it will boot directly into Kodi.

All videos should play with hardware acceleration (including AV1). I have not yet found anything that doesn't.

HDR 10 bit videos do play but whether there will be any triggering of the mode on a TV I do not know as I don't have an HDR TV.

8K videos do not play correctly, playing with a zoomed in look that cannot be corrected. However, as far as I am concerned, right now, 8K is a gimmick, so not much of a loss at this time.

If looking very closely, you may find that AV1 and some 4k will not be 100% smooth, although I find that it doesn't affect the viewing pleasure as it is not immediately obvious.

The Kodi version that will be installed is 22 dirty from what seems likely to be the latest nightly. Whether things may be better under a Kodi 21 stable or nightly build I do not know as I don't know how to specifically select any specific version to build from.

All in all though this is as close as we can get to perfect Kodi playback on RK3588, so a definite step in the right direction.

Bluetooth audio is not automatically present as an option for me but I need to do more research to find out why.

lukaszsobala, 2024-07-16:
Kodi packaged with the current 24.04 release lacks the option to select DRM PRIME as the renderer, and without it, 4k HDR won't work for me. It exists in the Kodi that Armbian provided for 22.04, not sure about this (JR) Ubuntu 22.04.

To get 4k HDR working, I needed to run kodi using the the Mali blobs (`malirun`). I'm not sure what it needs now.

Bluesmanuk, 2024-07-19:
Installing a new release to test, after applying all the most recent updates, which takes things 1020, I ran the same processes but when I get to:

cmake ../kodi -DCMAKE_INSTALL_PREFIX=/usr/local -DCORE_PLATFORM_NAME=wayland -DAPP_RENDER_SYSTEM=gles -B.

I get errors:

-- Could NOT find Iso9660pp (missing: ISO9660PP_LIBRARY ISO9660PP_INCLUDE_DIR) (Required is at least version "2.1.0")
-- Could NOT find MDNS (missing: MDNS_LIBRARY MDNS_INCLUDE_DIR)
-- Could NOT find Pipewire (missing: PIPEWIRE_LIBRARY PIPEWIRE_INCLUDE_DIR SPA_INCLUDE_DIR) (Required is at least version "0.3.50")
CMake Error at /usr/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Exiv2 (missing: EXIV2_LIBRARY EXIV2_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindExiv2.cmake:120 (find_package_handle_standard_args)
  cmake/scripts/common/Macros.cmake:378 (find_package)
  cmake/scripts/common/Macros.cmake:390 (find_package_with_ver)
  CMakeLists.txt:277 (core_require_dep)

I can't find anything that points me in a direction for a fix.

lukaszsobala, 2024-07-19:
@Bluesmanuk you just need to install the build dependencies (here, for example: libiso9660-dev, libmdnsd-dev and a few others)

Bluesmanuk, 2024-07-20:
Thanks.

Therein lay my initial issues, attempting to interpret the errors into practical commands to execute.

So, with your  2 very useful example, I got those, pipewire and Exiv2 installed.

But I still get the errors

CMake Error at /usr/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Exiv2 (missing: EXIV2_LIBRARY EXIV2_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindExiv2.cmake:120 (find_package_handle_standard_args)
  cmake/scripts/common/Macros.cmake:378 (find_package)
  cmake/scripts/common/Macros.cmake:390 (find_package_with_ver)
  CMakeLists.txt:277 (core_require_dep)

And other errors that are listed above these in white lettering

-- Could NOT find ClangFormat (missing: CLANG_FORMAT_EXECUTABLE)
-- Building internal TexturePacker
-- Shipping internal TexturePacker
-- Internal TexturePacker will be executed during build
-- Could NOT find libcec (missing: libcec_DIR)
-- Could NOT find MDNS (missing: MDNS_LIBRARY MDNS_INCLUDE_DIR)
-- Could NOT find Pipewire (missing: PIPEWIRE_LIBRARY PIPEWIRE_INCLUDE_DIR SPA_INCLUDE_DIR) (Required is at least version "0.3.50")

I did find that there was a newer Exiv2-0.28.2 but installing this seemed impossible, including the need to use something actually called ninja

It is beginning to look as if this really is rocket science.

lukaszsobala, 2024-07-20:
@Bluesmanuk what distro are you using? It's really easy, just install whatever it tells you. If it's debian/Ubuntu, just write `sudo apt install libexiv2-dev libpipewire*-dev libcec*-dev`

You need anything it lists, ending with "dev".

MDNS is irrelevant.

Bluesmanuk, 2024-07-21:
I'm using the 24.04

Interpreting the errors to define the naming conventions for installation of the dependencies just doesn't see, to have consistent logic that I can fathom.

However, the examples that you provided enabled compilation to complete, so thank you for the guidance.

I'm now back at an older problem that is random and sometimes works and sometimes not after a build, that of logging in successfully.

After selecting kodi on gm, when entering my password I am informed that my username is not listed and will not go any further.

This is despite adding myself to the video group using the sudo usermod -a -G input,video command and verifying that I am a member of that group.

lukaszsobala, 2024-07-21:
The logic is consistent. You read the output, find out what libraries are missing and install the `dev` versions of them. Just use the tab key and Google. 

Regarding your problem, I don't really understand what it is as I never even knew what gbm was. I use X11 or wayland.

Bluesmanuk, 2024-07-21:
We will have to agree to disagree on the interpretation as googling sent me in all kinds of directions when searching based on ay one element of the error.

The kodi on gbm is the option that is picked from the white dot on the lower right hand side of the screen that loads Kodi with the correct parameters for hardware acceleration, in particular the option to select direct to plane.

There is a screenshot of the login option provided above in a post by nguyenhunga5 on May 23.

I will keep persevering and much appreciate the useful help that you have provided.

wmport, 2024-07-21:
> sudo usermod -a -G input,video

 sudo usermod -a -G video  <user>

Tiemichael, 2024-07-22:
Thank you @nguyenhunga5 for your Kodi-build recipe! It's extremely helpful and works well on my NanoPC-T6 running Ubuntu24.04.

The only problems I am facing are with sound which is extremely soft.
My NanoPcCT6 is connected to a SONY-soundsystem, and in my previous installations on eg Raspberry I use Passthrough.
1) cannot see any Passthrough option
2) System -> Audio -> Audio output device shows me 'HDMI1 Stereo, HDMI Stereo (PIPEWIRE)' and nothing like 'SONY on HDMI1',
3) my previous installation uses 'Pulseaudio' - how can I test / change to Pulseaudio ?

Thanks for your help!

nguyenhunga5, 2024-07-23:
@Tiemichael this is my current setting. In my case "HDMIO Stereo, Analog Output (PULSFAUDIO)" is available after installation, no need to install any package.
[image]

wmport, 2024-07-23:
Built the package for myself about a month ago, will install all dependencies

https://drive.google.com/file/d/1XwciSZUCF4MtT4-soalwuysga2pDakLN/view?usp=sharing

Tiemichael, 2024-07-24:
> @Tiemichael this is my current setting. In my case "HDMIO Stereo, Analog Output (PULSFAUDIO)" is available after installation, no need to install any package. ![291788CE-8004-40DC-B8FA-E4A62F23E2B3_1_102_a](https://private-user-images.githubusercontent.com/853275/351383536-b821b6e4-b426-4447-8166-a097e00cd399.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE3ODU2NDMsIm5iZiI6MTcyMTc4NTM0MywicGF0aCI6Ii84NTMyNzUvMzUxMzgzNTM2LWI4MjFiNmU0LWI0MjYtNDQ0Ny04MTY2LWEwOTdlMDBjZDM5OS5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcyNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MjRUMDE0MjIzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Yzg3ZDdlOThiYTA1ZDIxMTRmZDNjZGUyNzQ2YzM2OTRmNTE2N2UzYjNlNjc4ZmI2ZTIzYjYzYzdiYjJkOGIzNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ._3yLL5pHc392iuFLf2rJYbHsMXbEch5Fsa3d4eTljOo)

@nguyenhunga5 : Thanks!
I still have not figured out how to install / setup multichannel audio output. But this seems more a Ubuntu than a Kodi issue

lukaszsobala, 2024-07-29:
> Built the package for myself about a month ago, will install all dependencies
> kodi-22.0-ALPHA1_arm64.deb for 24.04
> 
> https://drive.google.com/file/d/1XwciSZUCF4MtT4-soalwuysga2pDakLN/view?usp=sharing
> 

@wmport I have tried this version, as well as the ona packaged by @Joshua-Riek and both have no option for enabling DRM PRIME decoding (the only one visible is "Auto Detect"). How can I fix this?

wmport, 2024-07-30:
Prime render only.
Switch direct plane and EGL

lukaszsobala, 2024-07-30:
@wmport i don't have these options visible anywhere... The Expert panel is activated and the only option I see is "Auto Detect". I can make HDR videos play by forcing mpv as an external player in kodi, but this is a clunky workaround.

wmport, 2024-07-30:
https://github.com/Joshua-Riek/ubuntu-rockchip/issues/815#issuecomment-2123764529

photo  5 .  ....

wmport, 2024-07-30:
I suspect that you are not logged into a GBM session.
[image]

lukaszsobala, 2024-07-31:
Whenever I try to log in to "Kodi on GBM", the display just stops responding (even the mouse stops moving). The board does not hang because I can still access it via ssh perfectly.

nguyenhunga5, 2024-07-31:
> Whenever I try to log in to "Kodi on GBM", the display just stops responding (even the mouse stops moving). The board does not hang because I can still access it via ssh perfectly.

> sudo usermod -a -G video < user >

Do you try this?

lukaszsobala, 2024-07-31:
Doesn't the `rockchip-multimedia-config` package set this up? I am pretty sure it was added and then I added it again anyway, will try though.

Bluesmanuk, 2024-08-01:
I finally got things working, even though I don't quite know how.

Kodi updated to the latest version via the updater and I was able to see how things are.

Both EGL and Direct To Plane have advantages and disadvantages and there is no full function but at least I can see the current overall position in readiness for newer updates and versions that can be tracked.

AlexD81, 2024-09-18:
could someone build new version of kodi pls?   the one uploaded above doesn't works...  When i login at login screen, kodi's interface if frosen (or not responsive)      Tried to build it myself using instructions above, but getting errors, this way is not for regular users

lukaszsobala, 2024-09-18:
@AlexD81 same problem here. I never found a way around this.

Bluesmanuk, 2024-09-19:
Whenever there is a new build I run through the process, sometimes seeing new errors but with some research, managing to complete the compilation process.

However, Kodi performance is the same. Not bad by any means, just missing some common features already available on other hardware.

I always keep an eye on what Kodi, LibreELEC and Collabora (especially them) are doing with respect to all the kernel developments that are being sent upstream for mainline support.

I know that as soon as the right conditions are met then LibreELEC will quickly add support for the RK3588.

There is still some key work to be done and the latest H265 work by Collabora is WIP (work in progress).

This link will be particularly beneficial in following progress, where you can also see what has yet to be tackled.

https://gitlab.collabora.com/hardware-enablement/rockchip-3588/notes-for-rockchip-3588/-/blob/main/mainline-status.md?ref_type=heads

I believe that until more progress is made in certain areas, any active work to make Kodi fully functional cannot realistically happen.

Personally I don;t believe that we will see anything complete this year but that is just my speculation based upon the progress that we have seen.

Without Collabora, I think that this SOC would be dead in the water in terms of complete functionality and all the hard work and progress from great people like Joshua would likely never have gotten off the ground because there would not have been a solid enough base to begin with in order to make any progress at all.

So bottom line for me is that it is a waiting game to see if and when full functionality will come to the RK3588.

AlexD81, 2024-09-19:
almost perfect solution would be using mpv as player for kodi, i even got  dts/ac3  bitstream working for usb>spdif adapter, but  in ubuntu 24,  there are no working builds of kodi ,  
kodi installed via apt is crashing on start,   other builds have very sluggish ui, dont know why.    The custom build uploaded above starts ok, ui is ok, and also it works fine with mpv as player, but it's alpha,   i would prefer last stable,  and i would stick with it for a long time.

Bluesmanuk, 2024-09-19:
I did experiment with using MPV as the player from within Kodi but I found no improvement.

lukaszsobala, 2024-09-20:
@Bluesmanuk @AlexD81 no, mpv as a kodi player definitely works, and works very well, with hw acceleration.

The problem is that the resume time is not saved, which makes it a bit useless as a home media center thing. It is quite annoying, as I remember there was a time that Kodi itself worked with hardware acceleration... not sure what changed since then.

Bluesmanuk, 2024-09-21:
@lukaszsobala

Sadly it doesn't perform any better for me but in terms of the resume issue I may have come across something that might help and I have confirmed works in MPV for Windows.

Add the line below to your mpv.conf. I had to create the file in the Windows version.

**_save-position-on-quit_**

I also tested uosc in the windows version to add some rudimentary GUI controls.

wxzmz, 2024-09-21:
> almost perfect solution would be using mpv as player for kodi, i even got dts/ac3 bitstream working for usb>spdif adapter, but in ubuntu 24, there are no working builds of kodi , kodi installed via apt is crashing on start, other builds have very sluggish ui, dont know why. The custom build uploaded above, starts ok, ui is ok, and also it works fine with mpv as player, but it's alpha, i would prefer last stable, and i would stick with it for a long time. another problem is missing autoframerate, solved it by mapping commands to rc keys and using this tool for framerate switch https://github.com/jadahl/gnome-monitor-config.git

Kodi plug-in MPV has another problem, how to solve it perfectly with SMB

AlexD81, 2024-09-22:
@nguyenhunga5   don't add shares via kodi's  built-in smb.  Use your ubuntu mapped share folder as source for kodi.    Also, imo,  you have to use remote control with mini-keyboard.    like MX3 or better. Then you'll get full freedom with control  mpv,  for example, to quit mpv that running on top of kodi, press 'q'       Everything works fine, i can share mpv conf file that allows perfectly smooth playback of 24p movies.  I was waiting for libreelec, but now it's not really necessary, i like mpv over kodi approch better,   much more possibilities.  and mpv adds tonemapping that kodi is unable to do.  Just need someone to compile latest stable build of kodi with patches  , without them ui of kodi is sluggish for unknown reason

wxzmz, 2024-09-22:
> @nguyenhunga5 don't add shares via kodi's built-in smb. Use your ubuntu mapped share folder as source for kodi. Also, imo, you have to use remote control with mini-keyboard. like MX3 or better. Then you'll get full freedom with control mpv, for example, to quit mpv that running on top of kodi, press 'q' Everything works fine, i can share mpv conf file that allows perfectly smooth playback of 24p movies. I was waiting for libreelec, but now it's not really necessary, i like mpv over kodi approch more, much more possibilities. for example, mpv adds tonemapping that kodi is unable to do. Just need someone to compile latest stable build of kodi with patches , without them ui of kodi is sluggish for unknown reason

Indeed, I have been using mpv recently, mpv as a kodi external player, or as a celluloid, the compatibility is much better than the current patched kodi native effect.

'Q' is not a problem. I set the home button of the infrared/bluetooth remote control as the exit button of mpv.

AlexD81, 2024-09-22:
in my setup  of mpv over kodi  is still 1 problem left.  Need advice how to toggle bitstream on/off  via key press on rc.  
the commands for bitstream  are looking like this 
audio-device='alsa/iec958:CARD=Audio,DEV=0' 
audio-spdif=ac3,dts

probably it should be done via profiles in mpv conf ?  but can't figure out how to do it and  toggle profiles via key

Bluesmanuk, 2024-09-23:
I decided to try using MPV as an external player and created the playercorefactory.xml as described above and confirmed with the Kodi wiki.

If I run kodi -fs I get an error

_ERROR: Unable to create GUI. Exiting
'this->recurse > 0' failed at ../src/pipewire/thread-loop.c:63 do_unlock()_

When I run Kodi from the GUI desktop icon it loads but when I attempt to play a file something flashes up at the top of the screen that is too short to read and then nothing happens.

AlexD81, 2024-09-23:
@Bluesmanuk  what ubuntu version do yo uuse?  I remember I had problemns with 22,  but with 24  with last updates everything is fine.    Btw  when I clean everything " sudo apt purge kodi\* "        and then install custom build above,     then it automatically updates to latest stable (actually it downgrades)   and suprisingly this last stable working fine with no issues.

[24 more replies]