Joshua-Riek/ubuntu-rockchip #817: SATA is not work in Ubuntu 24.04

douglasdock, 2024-05-19:
I can't access SATA SSD on Ubuntu 24.04

I read in a message here on the forum that it would be possible to add SATA this way:

added 'overlays=sata' to /boot/firmware/ubuntuEnv.txt

But I can't find the location or the file.

hakotani-o, 2024-05-20:
Hello, douglasdock.

How to SATA boot orange pi 5
1. Write ubuntu-22.04-preinstalled-desktop-arm64-orangepi-5.img.xz to the SD card.
2. Insert SD card into orange pi 5 and boot
3. Run sudo u-boot-install-sata-mtd.
4. Download ubuntu-22.04-preinstalled-desktop-arm64-orangepi-5.img.xz.
5. Write ubuntu-22.04-preinstalled-desktop-arm64-orangepi-5.img.xz to /dev/sda.
6. Mount /dev/sda1 to /mnt.
   sudo mount /dev/sda1 /mnt
7 Add two lines to /mnt/etc/default/u-boot.
U_BOOT_FDT="device-tree/rockchip/rk3588s-orangepi-5.dtb"
U_BOOT_FDT_OVERLAYS="device-tree/rockchip/overlay/orangepi-5-sata.dtbo"
8. Edit /mnt/boot/extlinux/extlinux.conf.
fdt /lib/firmware/6.1.0-1013-rockchip/device-tree/rockchip/rk3588s-orangepi-5.dtb
fdtoverlays /lib/firmware/6.1.0-1013-rockchip/device-tree/rockchip/overlay/orangepi-5-sata.dtbo
Add.

I think it will look like this after editing:
I think the UUIDs are different.

label l0
menu label Ubuntu 24.04 LTS 6.1.0-1013-rockchip
linux /boot/vmlinuz-6.1.0-1013-rockchip
initrd /boot/initrd.img-6.1.0-1013-rockchip
fdt /lib/firmware/6.1.0-1013-rockchip/device-tree/rockchip/rk3588s-orangepi-5.dtb
fdtoverlays /lib/firmware/6.1.0-1013-rockchip/device-tree/rockchip/overlay/orangepi-5-sata.dtbo
append root=UUID=e7c855d5-7db6-4bea-a77e-438fd2afefa8 rootwait rw console=ttyS2,1500000 console=tty1 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memoryquiet splash plymouth.ignore-serial-consoles

9. sudo umount /mnt  and power off
10. Remove the SD card.
11. power on

douglasdock, 2024-05-26:
Hi, how are you? Thank you for your help and work.
I'm trying to do the tutorial above, but I can't do the command:

"sudo mount /dev/sda1 /mnt"

When I run the command it returns:
"mount: /mnt: special device /dev/sda1 does not exist.
       dmesg(1) may have more information after failed mount system call."

I don't know what to do, if it's possible and anyone has any ideas, I'll be grateful.

[image]

douglasdock, 2024-05-26:
And if I use the official Ubuntu image of the orange pi 5, I can identify the SSD and execute the command:

sudo mount /dev/sda1 /mnt

[image]

douglasdock, 2024-05-26:
Bom dia amigo, tudo bem?
Eu utilizei o sistema oficial para conseguir editar como falado no tutorial a cima:

para conseguir editar os arquivos eu usei os seguintes comandos:

sudo nano /mnt/etc/default/u-boot
sudo nano /mnt/boot/extlinux/extlinux.conf

[image]

**### And it worked!**

Thank you very much and here's a tip for anyone having problems.

digby5, 2024-05-31:
noticed that extlinux.conf. points to 1013 device-tree even though 1014 is installed on my system. i see the 1014 directory available. is this a bug or intended?

hakotani-o, 2024-06-01:
Hello.
My orange pi 5 is running ubuntu-rockchip , not oficial.

ubuntu@ubuntu:~$ cat /boot/extlinux/extlinux.conf 
## /boot/extlinux/extlinux.conf
##
## IMPORTANT WARNING
##
## The configuration of this file is generated automatically.
## Do not edit this file manually, use: u-boot-update

default l0
menu title U-Boot menu
prompt 1
timeout 20

label l0
	menu label Ubuntu 24.04 LTS 6.1.0-1014-rockchip
	linux /boot/vmlinuz-6.1.0-1014-rockchip
	initrd /boot/initrd.img-6.1.0-1014-rockchip
	fdt /lib/firmware/6.1.0-1014-rockchip/device-tree/rockchip/rk3588s-orangepi-5.dtb
	fdtoverlays  /lib/firmware/6.1.0-1014-rockchip/device-tree/rockchip/overlay/orangepi-5-sata.dtbo
	append root=UUID=a3b00cb3-ba66-4706-a9d3-f7c99c185d98 rootwait rw console=ttyS2,1500000 console=tty1 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memoryquiet splash plymouth.ignore-serial-consoles

label l0r
	menu label Ubuntu 24.04 LTS 6.1.0-1014-rockchip (rescue target)
	linux /boot/vmlinuz-6.1.0-1014-rockchip
	initrd /boot/initrd.img-6.1.0-1014-rockchip
	fdt /lib/firmware/6.1.0-1014-rockchip/device-tree/rockchip/rk3588s-orangepi-5.dtb
	append root=UUID=a3b00cb3-ba66-4706-a9d3-f7c99c185d98 rootwait rw console=ttyS2,1500000 console=tty1 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memoryquiet splash plymouth.ignore-serial-consoles single
	

label l1
	menu label Ubuntu 24.04 LTS 6.1.0-1013-rockchip
	linux /boot/vmlinuz-6.1.0-1013-rockchip
	initrd /boot/initrd.img-6.1.0-1013-rockchip
	fdt /lib/firmware/6.1.0-1013-rockchip/device-tree/rockchip/rk3588s-orangepi-5.dtb
	fdtoverlays  /lib/firmware/6.1.0-1013-rockchip/device-tree/rockchip/overlay/orangepi-5-sata.dtbo
	append root=UUID=a3b00cb3-ba66-4706-a9d3-f7c99c185d98 rootwait rw console=ttyS2,1500000 console=tty1 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memoryquiet splash plymouth.ignore-serial-consoles

label l1r
	menu label Ubuntu 24.04 LTS 6.1.0-1013-rockchip (rescue target)
	linux /boot/vmlinuz-6.1.0-1013-rockchip
	initrd /boot/initrd.img-6.1.0-1013-rockchip
	fdt /lib/firmware/6.1.0-1013-rockchip/device-tree/rockchip/rk3588s-orangepi-5.dtb
	append root=UUID=a3b00cb3-ba66-4706-a9d3-f7c99c185d98 rootwait rw console=ttyS2,1500000 console=tty1 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memoryquiet splash plymouth.ignore-serial-consoles single
	
ubuntu@ubuntu:~$

closed 2024-06-04 by douglasdock