Joshua-Riek/ubuntu-rockchip #819: wine32:i386 install

ageevp, 2024-05-20:
sudo dpkg -i '/home/petr/Downloads/wine32_9.0~repack-4build3_i386.deb' 
(Reading database ... 160259 files and directories currently installed.)
Preparing to unpack .../wine32_9.0~repack-4build3_i386.deb ...
Unpacking wine32:i386 (9.0~repack-4build3) over (8.0.1~repack-3ubuntu2) ...
dpkg: dependency problems prevent configuration of wine32:i386:
 wine32:i386 depends on libc6 (>= 2.38).
 wine32:i386 depends on libwine (= 9.0~repack-4build3).

dpkg: error processing package wine32:i386 (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 wine32:i386

aubreyrjones, 2024-06-16:
These are ARM boards, so I don't think that the x86 version of wine is going to work that easily.

ageevp, 2024-07-24:
> These are ARM boards, so I don't think that the x86 version of wine is going to work that easily.

Reposited wine is 64 bit only, But I need wine32.
I've install last version  by 
https://github.com/neofeo/BOX86-BOX64-WINEx86-TUTORIAL

closed 2025-01-06 by github-actions[bot]