Joshua-Riek/ubuntu-rockchip #822: [FR] Add support for various rk3566 retro handheld devics (Anbernic/Powkiddy RG353M/P/V X55 etc) + Add build documentation

aenertia, 2024-05-21:
These are all mostly similar to the OrangePi3B (8/32GB or None Emmc, and 2GB Ram) with various extra dtsi/dtb for their screens and controller inputs. Thankfully most of these are now in mainline. 

Ideally now that both the rk3588 and rk3566 have somewhat usable UEFI shims - moving to a standard UEFI style install would be grand, but that's a whole other ticket. 

I've been trying to figure out what your buid pipeline looks like so I can add these targets myself rather than taking the orangepi3b image and hacking at it. 

Any tips appreciated.

Joshua-Riek, 2024-05-22:
I'm trying to follow the upstream ubuntu process, but sadly its pretty much not documented and relies heavily on Launchpad. Because of that it will be hard to make your own modifications to the kernel and bootloader, everything else is kind of simple. 

I need to make adjustments to allow for easy kernel and bootloader developmnet.

hqnicolas, 2024-05-30:
@aenertia Talking about[ TODAY fixes](https://forum.armbian.com/topic/28895-efforts-to-develop-firmware-for-h96-max-v56-rk3566-8g64g/?do=findComment&comment=192628) for your problem
you can compile Radxa U-boot inside armbian build for kernel 6.1
and change the boot partition inside joshua  OrangePi3B image to your armbian image.
it will change the board boot method and the DTB files.

aenertia, 2024-05-31:
I have been using arkOS's ancient 4.19 BSP kernel and initrd and bootloader with the userspace ; the mainline uboot in 6.10 works with the rk3566 as used in rocknix ; I don't think the relevant DTSI's for the anbernic/powkiddy devices are in the 6.1 tree (they might be but I think they were not added to mainline until after whenever the ubuntu-rockchip 6.1 fork was made).

I was using the opi3b base. The main trick is going to be around the dtb. I think most of these boards are most similar to the evb2 dtb that's in there.

hqnicolas, 2024-05-31:
@aenertia 
>The main trick is going to be around the dtb.

**Some DTSi methods diff from mainline linux**

I think you can take the anbernic/powkiddy mainline linux as an example for:
USB [okay | disabled]
sdmmc* [okay | disabled]
VOP [okay | disabled]
UART [okay | disabled]
pmic enabling method
pinctrl RK_Pin names
RK_Pin names

and you can take Rockchip Kernel 6.1 DTS evb2 as an example for:
#include method
GPU enabling method
rkvdec enabling method
Wifi enabling method
sram enabling method
thermal_zones enabling method
reserved_memory enabling method

This is result in a Develop image where you can debug all other Peripherals

closed 2024-12-30 by github-actions[bot]