Joshua-Riek/ubuntu-rockchip #823: Blender 4.0.2 Segmentation Fault

khanumballz, 2024-05-21:
Ubuntu 24.04 Rockchip running on Orange Pi 5 Plus, from this repository

```
khan@khan-desktop:~$ blender
Writing: /tmp/blender.crash.txt
Segmentation fault (core dumped)
khan@khan-desktop:~$ cat /tmp/blender.crash.txt
# Blender 4.0.2, Unknown revision

# backtrace
blender(+0xa21c20) [0xaaaabb1a1c20]
blender(+0x7084f4) [0xaaaabae884f4]
linux-vdso.so.1(__kernel_rt_sigreturn+0) [0xffff7fefe7ec]
blender(+0xaa5634) [0xaaaabb225634]
blender(+0xa76760) [0xaaaabb1f6760]
blender(+0xa783dc) [0xaaaabb1f83dc]
blender(+0xa8dd28) [0xaaaabb20dd28]
blender(+0x6ad998) [0xaaaabae2d998]
/lib/aarch64-linux-gnu/libc.so.6(+0x284c4) [0xffff7a2d84c4]
/lib/aarch64-linux-gnu/libc.so.6(__libc_start_main+0x98) [0xffff7a2d8598]
blender(+0x6f86b0) [0xaaaabae786b0]

# Python backtrace
khan@khan-desktop:~$ 
```

Won't run blender. 4k video cannot play. 1080p video is extremely choppy. But aside from that, it's a smooth desktop experience. Well done!

khanumballz, 2024-05-31:
Current workaround:

1. Use Ubuntu Rockchip 22.04 only.
2. Download BoxBuddy (flatpak) and Distrobox
3. Install Ubuntu 18.04-docker
4. Install blender on dockerized ubuntu 18.
5. Install virgl-server on host machine.
6. Install blender 2.79 on boxbuddy-ubuntu 18 (sudo apt install blender will do)
7. Now run: ```virgl_test_server &``` on host machine (or tmux > virgl_test_server)
8. And then in dockerized ubuntu 18 run ```GALLIUM_DRIVER=virpipe blender```

You can then use Blender 2.79 with hardware acceleration.

Probably won't work with any newer version of blender. Cause of problem = Lack of Vulkan support on mali g610 drivers. Not sure if this will close this issue, or not.

PhilZProg, 2024-05-31:
Oh, I'm using 24.04. But still thank you for the workaround.

closed 2024-12-30 by github-actions[bot]