Joshua-Riek/ubuntu-rockchip #824: 24LTS bluetooth can not open，and wifi works

BaconSoong, 2024-05-22:
24LTS bluetooth can not open，and wifi works, please checkout.

BaconSoong, 2024-05-22:
[image]

BaconSoong, 2024-05-22:
root@i-desktop:/home/i# dmesg |grep -i bluetooth
[   13.580385] Bluetooth: Core ver 2.22
[   13.580407] NET: Registered PF_BLUETOOTH protocol family
[   13.580415] Bluetooth: HCI device and connection manager initialized
[   13.580426] Bluetooth: HCI socket layer initialized
[   13.580434] Bluetooth: L2CAP socket layer initialized
[   13.580446] Bluetooth: SCO socket layer initialized
[   17.608048] Bluetooth: HCI UART driver ver 2.3
[   17.608054] Bluetooth: HCI UART protocol H4 registered
[   17.608056] Bluetooth: HCI UART protocol BCSP registered
[   17.608058] Bluetooth: HCI UART protocol ATH3K registered
[   17.617214] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   17.623941] Bluetooth: HCI UART protocol Intel registered
[   17.633515] Bluetooth: HCI UART protocol Broadcom registered
[   17.639935] Bluetooth: HCI UART protocol QCA registered
[   17.639941] Bluetooth: HCI UART protocol AG6XX registered
[   17.642741] hci_uart_bcm serial0-0: Looking up vddio-supply property in node /serial@feb90000/bluetooth failed
[   18.012864] Bluetooth: hci0: BCM: chip id 130
[   18.013482] Bluetooth: hci0: BCM: features 0x0f
[   18.015866] Bluetooth: hci0: BCM4345C5
[   18.015872] Bluetooth: hci0: BCM4345C5 (003.006.006) build 0000
[   18.016077] Bluetooth: hci0: BCM: firmware Patch file not found, tried:
[   18.016092] Bluetooth: hci0: BCM: 'brcm/BCM4345C5.radxa,rock-5b.hcd'
[   18.016097] Bluetooth: hci0: BCM: 'brcm/BCM4345C5.hcd'
[   18.016101] Bluetooth: hci0: BCM: 'brcm/BCM.radxa,rock-5b.hcd'
[   18.016106] Bluetooth: hci0: BCM: 'brcm/BCM.hcd'
[   18.042132] Bluetooth: hci0: Opcode 0x2031 failed: -22
[   23.389313] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   23.389330] Bluetooth: BNEP filters: protocol multicast
[   23.389344] Bluetooth: BNEP socket layer initialized

Joshua-Riek, 2024-05-22:
In the future I will ignore all issues opened like this and will close them without comment, you've basically said Bluetooth does not work and given me no other information. 

What board, module, and provide full dmesg.

BaconSoong, 2024-05-22:
sorry, it's my mistake. the board is rock 5b and the bluetooth & wifi card is bcm43456 .
[rock5b_bcm43456_demsg.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/15409626/rock5b_bcm43456_demsg.txt)

BaconSoong, 2024-05-25:
hello, i copy https://github.com/armbian/firmware/tree/master/brcm to /lib/firmware/brcm to solve this problem.  now it works.  Thank you for your attention to this matter。

hqnicolas, 2024-05-29:
wifi works?
I have the same problem here with H96MAX RK3566
will take this as example

[   14.576093] [WLAN_RFKILL]: Enter rfkill_wlan_init
[   14.576968] [WLAN_RFKILL]: Enter rfkill_wlan_probe
[   14.577716] [WLAN_RFKILL]: can't find rockchip,grf property
[   14.578456] [WLAN_RFKILL]: wlan_platdata_parse_dt: wifi_chip_type = ap6256
[   14.579188] [WLAN_RFKILL]: wlan_platdata_parse_dt: enable wifi power control.
[   14.579915] [WLAN_RFKILL]: wlan_platdata_parse_dt: wifi power controled by gpio.
[   14.580647] [WLAN_RFKILL]: wlan_platdata_parse_dt: WIFI,host_wake_irq = 10, flags = 0.
[   14.581370] [WLAN_RFKILL]: wlan_platdata_parse_dt: The ref_wifi_clk not found !
[   14.582098] usb 3-1: new high-speed USB device number 2 using ehci-platform
[   14.582639] [WLAN_RFKILL]: rfkill_wlan_probe: init gpio
[   14.583561] [WLAN_RFKILL]: rfkill_set_wifi_bt_power: 1
[   14.584276] [WLAN_RFKILL]: Exit rfkill_wlan_probe

thanhtantran, 2024-05-30:
I have the same problem with Orange Pi 5 Pro board, first installation wifi work, but after update and upgrade, the wifi is down

here is the dmesg

```
[    6.958563] usbcore: registered new interface driver rndis_wlan
[    7.399240] [WLAN_RFKILL]: Enter rfkill_wlan_init
[    7.399930] [WLAN_RFKILL]: Enter rfkill_wlan_probe
[    7.400988] [WLAN_RFKILL]: wlan_platdata_parse_dt: wifi_chip_type = ap6256
[    7.401515] [WLAN_RFKILL]: wlan_platdata_parse_dt: enable wifi power control.
[    7.402040] [WLAN_RFKILL]: wlan_platdata_parse_dt: wifi power controled by gpio.
[    7.402578] [WLAN_RFKILL]: wlan_platdata_parse_dt: WIFI,poweren_gpio = 24 flags = 0.
[    7.403114] [WLAN_RFKILL]: wlan_platdata_parse_dt: WIFI,host_wake_irq = 0, flags = 0.
[    7.403645] [WLAN_RFKILL]: wlan_platdata_parse_dt: The ref_wifi_clk not found !
[    7.404177] [WLAN_RFKILL]: rfkill_wlan_probe: init gpio
[    7.405245] [WLAN_RFKILL]: Exit rfkill_wlan_probe
```

hqnicolas, 2024-05-30:
@thanhtantran can you share with us the version for the first install and the version that doesnt work?

Joshua-Riek, 2024-06-01:
Im closing this, please open a new issue with details and full dmesg logs.

closed 2024-06-01 by Joshua-Riek

hqnicolas, 2024-06-01:
@Joshua-Riek 
My wifi card is BRCM 4335
and I solve my wifi problem by changing these kernelconfigs

> CONFIG_BRCM_TRACING=y
> CONFIG_BRCMDBG=y
> CONFIG_B43LEGACY=m
> CONFIG_B43LEGACY_PCI_AUTOSELECT=y
> CONFIG_B43LEGACY_PCICORE_AUTOSELECT=y
> CONFIG_B43LEGACY_LEDS=y
> CONFIG_B43LEGACY_HWRNG=y
> CONFIG_B43LEGACY_DEBUG=y
> CONFIG_B43LEGACY_DMA=y
> CONFIG_B43LEGACY_PIO=y
> CONFIG_B43LEGACY_DMA_AND_PIO_MODE=y

thanhtantran, 2024-06-02:
> @thanhtantran can you share with us the version for the first install and the version that doesnt work?

my new issue opened here
https://github.com/Joshua-Riek/ubuntu-rockchip/issues/841