Joshua-Riek/ubuntu-rockchip #825: Unsupported platform after upgrade

Zaanza, 2024-05-23:
Hello, after experiencing some problems I tried to update my Orange Pi 5B (Ubuntu 22.04.4 LTS), I use :
```
sudo apt-get update 
sudo apt-get upgrade
```
And I got this error :
```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  python3-update-manager update-manager-core
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
2 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up initramfs-tools (0.140ubuntu13.4) ...
update-initramfs: deferring update (trigger activated)
Setting up flash-kernel (3.104ubuntu20) ...
flash-kernel: deferring update (trigger activated)
Processing triggers for initramfs-tools (0.140ubuntu13.4) ...
update-initramfs: Generating /boot/initrd.img-5.10.160-rockchip
Unsupported platform 'Orange Pi 5B'.
run-parts: /etc/initramfs/post-update.d//flash-kernel exited with return code 1
dpkg: error processing package initramfs-tools (--configure):
 installed initramfs-tools package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 initramfs-tools
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Joshua-Riek, 2024-05-23:
Id recommend to re-install your system as flash-kernel is no longer used and the kernel has been migrated. 

If you know what version you installed i may be able to come up with a quick hack to fix this, but know that the kernel is no longer supported.

Zaanza, 2024-05-23:
Thanks for the answer, is it possible to reinstall without flashing ? I have the eMMC version but I no longer have access to the flash button.

Joshua-Riek, 2024-05-25:
Boot from a sd card and follow the wiki:

https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Ubuntu-22.04-LTS#install-ubuntu-onto-an-emmc-from-linux

Zaanza, 2024-05-26:
Thank you very much, I will look at that.

closed 2024-05-26 by Zaanza