Joshua-Riek/ubuntu-rockchip #826: USB inputs don't work in Ubuntu 24.04 on Orange Pi5

douglasdock, 2024-05-25:
USB inputs don't work in Ubuntu 24.04 on Orange Pi5
When I put any storage device in any USB port it cannot mount.   The storage drives appear, but you cannot access them.  The keyboard and mouse work normally.

FreeWilly2022, 2024-05-25:
@douglasdock - I wanted you to get an answer, so I got on my Orange Pi 5 running Ubuntu 24.04 desktop, put a USB drive in the USB 3 and USB 2 ports, and it showed up on the desktop and I could open the drive. I then copied the files from the drive to my desktop. So basically I cannot reproduce your issue. Maybe you could check how you have the USB drive formatted, and google if its supported by Linux.

douglasdock, 2024-05-26:
Good morning, I still can't access the files on the USB:

[image]

[image]

[image]

If anyone can help me fix this error, I would be grateful.

FreeWilly2022, 2024-05-26:
@douglasdock - this does not appear to be a problem with the operating system. Either your USB drive is damaged or it is not formatted in a way that Linux can read it. Since this is an os github, you probably will get better help asking an ubuntu forum, or googling an answer. Here is a place you can start looking:

https://ubuntu.com/support/community-support

douglasdock, 2024-05-27:
Hi, thanks for the response.
Unfortunately there is no problem with the device, as they work on other IOS of the orange pi 5

Android 12
[image]

I will try other options and if I have any news I will report it, thank you very much for your attention and work

douglasdock, 2024-05-31:
Just to inform. I couldn't get the system to recognize the penddrive or SSD that were in NTFS, I converted the drives to fat32 and it recognized normally.
I looked for solutions on the internet and the following commands came to me:

sudo apt install nfs-kernel-server
sudo apt install nfs-common
sudo apt install ntfs-3g

But even after installing and updating it still didn't open, the solution was to convert to FAT

Joshua-Riek, 2024-05-31:
The usb ports work, try new usb devices.

closed 2024-05-31 by Joshua-Riek