Joshua-Riek/ubuntu-rockchip #828: Ask a compilation question

ohyeah521, 2024-05-26:
Hello, how can I define the kernel and compile it myself for the new version? The latest action is compiled using launchpad. What command should I use to compile the kernel and Ubuntu22.04 system?
Is the following command correct?
`sudo ./build.sh --board=rock-5b --release=jammy --project=server --kernel=rockchip-5.10`

Joshua-Riek, 2024-05-26:
I moved things to launchpad, so there is not a way to define the kernel. Sadly, this process is quite complex, but was necessary to follow the upstream ubuntu process. 

However, I have added a hack where you should be able to define a kernel branch to compile. 

https://github.com/Joshua-Riek/ubuntu-rockchip/blob/f65bb4ff7796ed2c4b7e2c5a6694ec03f27c52fe/config/releases/jammy.sh#L6-L7

ohyeah521, 2024-05-26:
Sorry, since I don’t know much about this project, how should I compile a custom kernel? Can you provide the steps and command code?

closed 2024-05-30 by Joshua-Riek

hqnicolas, 2024-05-31:
@ohyeah521 how mutch you will change the kernel?
in [my fork ](https://github.com/hqnicolas/ubuntu-rockchip) I was able to[ build a custom kernel config](https://github.com/hqnicolas/ubuntu-rockchip/actions/runs/9321238931)
using [the patch file ](https://github.com/hqnicolas/ubuntu-rockchip/blob/main/packages/u-boot-turing-rk3588/debian/patches/0006-add-Radxa-zero-3w-support.patch) 

in this case I need to keep on track how mutch I will change the patch file to edit the line counting.