Joshua-Riek/ubuntu-rockchip #829: Will adapt Radxa ROCK 5C？

zhemed, 2024-05-26:


Joshua-Riek, 2024-05-26:
Yup, working on it now.

majin2020, 2024-05-29:
> Yup, working on it now.

Great news :)

closed 2024-05-30 by Joshua-Riek

drsrinathsridhar, 2024-06-12:
Just wanted to confirm the 24.04 image works great on the Rock 5C. Microphones connected to the audio jack didn't work in the 5.xx kernel, but the 6.10 kernel fixes this.

Thanks for all you do @Joshua-Riek !

johnrichardrinehart, 2024-12-05:
> Just wanted to confirm the 24.04 image works great on the Rock 5C. Microphones connected to the audio jack didn't work in the 5.xx kernel, but the 6.10 kernel fixes this.
> 
> Thanks for all you do @Joshua-Riek !

Do you mean 6.1 kernel? I'm trying to get this working on 6.11 and I haven't seen anyone with a working kernel build beyond 6.1, yet.

drsrinathsridhar, 2024-12-05:
yes, 6.1.

johnrichardrinehart, 2024-12-06:
@Joshua-Riek You investigate getting this working on 6.11? I started and run into all manner of issues. Huge blocker around clock initialization super early on.

Joshua-Riek, 2024-12-07:
If it's a huge blocker, figure it out. It's quite easy to adapt a device tree for mainline linux. I'm pretty much done with user bugs and issues. For over a year, I've hated this project and forced myself to deal with this shit.

johnrichardrinehart, 2024-12-07:
> If it's a huge blocker, figure it out. It's quite easy to adapt a device tree for mainline linux. I'm pretty much done with user bugs and issues. For over a year, I've hated this project and forced myself to deal with this shit. 

I hear you. I wasn't trying to make a demand/expectation. I just wanted to know if I was wasting my time in the case where you already tackled the problems I'm seeing. 

I don't know your situation, but if you hate working on this then you might just archive it and call it a day. No shame in that. 

Thanks for getting back to me!

juergenpf, 2024-12-07:
I would also recommend to archive this project. This would be a clear statement, that there is no more support to expect and this is a dead end and people should look for alternatives or start their own project.

Joshua-Riek, 2024-12-07:
Just because I don't care anymore does not mean I'll be gone forever. This is my project, and i will do as i wish. I do not owe anyone an explanation for my actions.