Joshua-Riek/ubuntu-rockchip #831: Install image on partition instead entire device

pugant, 2024-05-27:
Good morning,
in reality this is not an issue, but more a request for information and any documentation.
For future updates that come out, I would like to setup my NVME device on Orange Pi 5 Plus with two partitions, so I can put the OS on one and reserve the other for data.
This way I will be able to update only the operating system partition with each release and not destroy the entire device with each update.

I tried to do this by partitioning my disk with two partitions:
- nvme0p1 (Operating System)
- nvme0p2 (Data)

I then flashed the image on the nvme0p1 partition with dd and rebooted, but what happens is that upon reboot I get the Orange splash screen and no system loads
I would like to know if this is possible and if so, what are the steps (with lots of detail) to be able to do it.

hakotani-o, 2024-05-28:
Hello.
I don't have an NVME DISK, so I can't confirm, but I was able to resize a USB-connected DISK.

Is it possible to boot ubuntu desktop from SD-CARD?

After booting,
There is an application called DISK in Utilities, so start it.

If NVME DISK is displayed, I think you can resize it.

I wish you the best.

pugant, 2024-05-28:
Hello,
I tried to resize the partitions successfully, directly on my nvme disk after installing the image using dd and I succeeded without problems.
Unfortunately however, when we have the next version of the image (for example version 3.0) thanks to the fantastic work of Joshua, if I want to update to the new version I will have to destroy the partition scheme and flash the image on the entire disk instead of on the specific partition .
If someone already knows how it is possible to do it and wants to provide step-by-step guidance, I believe they will make an important contribution to the correct management and updating of their SBC