Joshua-Riek/ubuntu-rockchip #834: Nanopc T6: Is it possible to boot from nvme which installed on E-Key slot?

4MT-HJM, 2024-05-28:
Hello,

I was trying to install system and boot from nvme ssd installed on E-Key M.2 slot, instead of M-Key, as I was planning to install some other pcie device on the 3.0x4 slot.

Firstly I boot the system from sd card, and I can see the nvme ssd device on E-key slot(with an adaption card) through lspci:
```
ubuntu@ubuntu:~$ lspci
0002:20:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
0002:21:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0003:30:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
0003:31:00.0 Non-Volatile memory controller: KIOXIA Corporation NVMe SSD (rev 01)
0004:40:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
0004:41:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
```

Also i can see nvme device in /dev:
```
ubuntu@ubuntu:~$ ls /dev
autofs           input         mmcblk0rpmb  tty0   tty41     vcs2
block            iomux         mmcblk1      tty1   tty42     vcs3
btrfs-control    kmsg          mmcblk1p1    tty10  tty43     vcs4
bus              kvm           mmcblk1p2    tty11  tty44     vcs5
cec0             log           mpp_service  tty12  tty45     vcs6
cec1             loop-control  mqueue       tty13  tty46     vcsa
char             loop0         mtd          tty14  tty47     vcsa1
console          loop1         mtd0         tty15  tty48     vcsa2
cpu_dma_latency  loop2         mtd0ro       tty16  tty49     vcsa3
crypto           loop3         mtdblock0    tty17  tty5      vcsa4
cuse             loop4         net          tty18  tty50     vcsa5
disk             loop5         ng0n1        tty19  tty51     vcsa6
dma_heap         loop6         null         tty2   tty52     vcsu
dri              loop7         nvme0        tty20  tty53     vcsu1
drm_dp_aux0      mali0         nvme0n1      tty21  tty54     vcsu2
fd               mapper        nvme0n1p1    tty22  tty55     vcsu3
full             mem           nvme0n1p2    tty23  tty56     vcsu4
fuse             mmcblk0       port         tty24  tty57     vcsu5
gpiochip0        mmcblk0boot0  ppp          tty25  tty58     vcsu6
gpiochip1        mmcblk0boot1  psaux        tty26  tty59     vendor_storage
gpiochip2        mmcblk0p1     ptmx         tty27  tty6      vfio
gpiochip3        mmcblk0p10    pts          tty28  tty60     vhci
gpiochip4        mmcblk0p11    ram0         tty29  tty61     vhost-net
gpiochip5        mmcblk0p12    random       tty3   tty62     vhost-vsock
hdmirx_hdcp      mmcblk0p13    rfkill       tty30  tty63     video-dec0
hugepages        mmcblk0p14    rga          tty31  tty7      video-enc0
hwrng            mmcblk0p15    rkspi-dev2   tty32  tty8      video0
i2c-0            mmcblk0p16    rtc          tty33  tty9      watchdog
i2c-10           mmcblk0p2     rtc0         tty34  ttyFIQ0   watchdog0
i2c-11           mmcblk0p3     shm          tty35  ubi_ctrl  zero
i2c-2            mmcblk0p4     snd          tty36  uhid      zfs
i2c-6            mmcblk0p5     stderr       tty37  uinput
i2c-7            mmcblk0p6     stdin        tty38  urandom
i2c-9            mmcblk0p7     stdout       tty39  v4l
iio:device0      mmcblk0p8     sw_sync      tty4   vcs
initctl          mmcblk0p9     tty          tty40  vcs1
```

I've used the script to install os into nvme and updated u-boot into SPI and EMMC, and after reboot seems u-boot only search for the M-key pcie:
```
pcie@fe150000: PCIe Linking... LTSSM is 0x0
pcie@fe150000: PCIe Linking... LTSSM is 0x0
pcie@fe150000: PCIe Linking... LTSSM is 0x0
pcie@fe150000: PCIe Linking... LTSSM is 0x0
pcie@fe150000: PCIe-0 Link Fail

Device 0: unknown device
```

I've tried to move the nvme ssd to M-key and os can boot successfully.

I did some more research in the u-boot code(but honestly I'm really new to arm board and u-boot...), and found it seems the driver for pci2x1 is already in Joshua-Riek/u-boot-rockchip/arch/arm/dts/rk3588.dtsi:
```
pcie2x1l0: pcie@fe170000 {
```
but not in arch/arm/dts/rk3588-nanopc-t6.dts

till here I got no idea how to go ahead... Any help would be greatly appreciated

Thanks in advance!

Joshua-Riek, 2024-05-28:
The nodes pcie2x1l0 and pcie2x1l2 are for the ethernet ports. While pcie2x1l1 would be the correct node. 

I threw this together but have no idea if it works, but id use it as a starting point.
```diff
diff --git a/arch/arm/dts/rk3588-nanopc-t6.dts b/arch/arm/dts/rk3588-nanopc-t6.dts
index ee7c7f005cc..9acaffb4900 100644
--- a/arch/arm/dts/rk3588-nanopc-t6.dts
+++ b/arch/arm/dts/rk3588-nanopc-t6.dts
@@ -78,6 +78,20 @@
                vin-supply = <&vcc5v0_sys>;
        };
 
+       vdd_mpcie_3v3: vdd-mpcie-3v3 {
+               compatible = "regulator-fixed";
+               regulator-name = "vdd_mpcie_3v3";
+               pinctrl-names = "default";
+               pinctrl-0 = <&pcie_m2_1_pwren>;
+               regulator-min-microvolt = <3300000>;
+               regulator-max-microvolt = <3300000>;
+               enable-active-high;
+               gpio = <&gpio4 RK_PC2 GPIO_ACTIVE_HIGH>;
+               regulator-boot-on;
+               regulator-always-on;
+               vin-supply = <&vcc5v0_sys>;
+       };
+
        led_sys: led-sys {
                u-boot,dm-pre-reloc;
                compatible = "regulator-fixed";
@@ -102,11 +116,23 @@
        status = "okay";
 };
 
+&pcie2x1l1 {
+       u-boot,dm-pre-reloc;
+       vpcie3v3-supply = <&vdd_mpcie_3v3>;
+       reset-gpios = <&gpio4 RK_PA2 GPIO_ACTIVE_HIGH>;
+       status = "okay";
+};
+
 &combphy0_ps {
        u-boot,dm-pre-reloc;
        status = "okay";
 };
 
+&combphy1_ps {
+       u-boot,dm-pre-reloc;
+       status = "okay";
+};
+
 &combphy2_psu {
        u-boot,dm-pre-reloc;
        status = "okay";
@@ -158,4 +184,13 @@
                        rockchip,pins = <4 RK_PA3 RK_FUNC_GPIO &pcfg_pull_none>;
                };
        };
+
+       pcie {
+               u-boot,dm-pre-reloc;
+               pcie_m2_1_pwren: pcie-m21-pwren {
+                       u-boot,dm-pre-reloc;
+                       rockchip,pins = <4 RK_PC2 RK_FUNC_GPIO &pcfg_pull_none>;
+               };
+       };
+
 };

```

4MT-HJM, 2024-06-11:
Thanks Joshua!
Tested and worked as expected, the only thing need to change is,
in "vdd_mpcie_3v3" need to add one line "u-boot,dm-pre-reloc;"

closed 2024-06-11 by Joshua-Riek

wmport, 2024-09-15:
it's a bad modification, it's a hardwired conflict. 

[   10.374044] pci_bus 0000:01: busn_res: can not insert [bus 01-ff] under [bus 00-0f] (conflicts with (null) [bus 00-0f])

PR #872 (2024-06-11): Add E-Key slot nvme boot support for nanopc t6 in u-boot
Resolves #834