Joshua-Riek/ubuntu-rockchip #835: Hardware acceleration not working any more

jackduckworth, 2024-05-28:
I installed Ubuntu 24.04 on an Orange Pi 5 and confirmed that hardware acceleration was working. I have just realised that it stopped working a few days ago.

(I cant remember what I was doing when it stopped, but nothing suspicious springs to mind.)

I've pasted various log excerpts and diagnostics below in the hope that someone can help diagnose it.

If I cant fix it I will probably re-install from scratch and
- check it works initially to rule out a hardware fault
- retrace my steps and see what caused it to fail.

chrome://gpu/ output
```
Graphics Feature Status
Canvas: Software only, hardware acceleration unavailable
Canvas out-of-process rasterization: Disabled
Direct Rendering Display Compositor: Disabled
Compositing: Software only. Hardware acceleration disabled
Multiple Raster Threads: Enabled
OpenGL: Disabled
Rasterization: Software only. Hardware acceleration disabled
Raw Draw: Disabled
Video Decode: Software only. Hardware acceleration disabled
Video Encode: Software only. Hardware acceleration disabled
Vulkan: Disabled
WebGL: Software only, hardware acceleration unavailable
WebGL2: Software only, hardware acceleration unavailable
WebGPU: Disabled
```

/var/log/syslog is full of this message, over-and-over...

```
2024-05-27T17:59:12.467405+01:00 orangepi5 mutter-x11-fram[2669]: (../../../gdk/x11/gdkdisplay-x11.c:2924):gdk_x11_display_get_visual_info_for_visual: runtime check failed: (nvisuals == 1)
2024-05-27T17:59:12.468541+01:00 orangepi5 mutter-x11-fram[2669]: message repeated 5 times: [ (../../../gdk/x11/gdkdisplay-x11.c:2924):gdk_x11_display_get_visual_info_for_visual: runtime check failed: (nvisuals == 1)]
```

glxinfo -B
```
name of display: :0
display: :0  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Mesa/X.org (0xffffffff)
    Device: llvmpipe (LLVM 14.0.6, 128 bits) (0xffffffff)
    Version: 23.0.0
    Accelerated: no
```

# System Details Report
```
## Hardware Information:
- **Hardware Model:**                              (null)
- **Memory:**                                      7.8 GiB
- **Processor:**                                   
- **Graphics 1:**                                  Unknown Graphics Controller
- **Graphics 2:**                                  Unknown Graphics Controller
- **Disk Capacity:**                               128.0 GB

## Software Information:
- **Firmware Version:**                            (null)
- **OS Name:**                                     Ubuntu 24.04 LTS
- **OS Build:**                                    (null)
- **OS Type:**                                     64-bit
- **GNOME Version:**                               46
- **Windowing System:**                            Wayland
- **Kernel Version:**                              Linux 6.1.0-1013-rockchip

```

jackduckworth, 2024-06-02:
Sorry for raising this and not giving you anything to go on.

I traced it back to something I shouldn't have done: after installing Jellyfin I followed the steps to turn on hardware acceleration (detailed on this page: https://jellyfin.org/docs/general/administration/hardware-acceleration/rockchip).

Without thinking I followed the link in "Install the ARM Mali OpenCL runtime (libmali) on host" and installed a binary from https://github.com/tsukumijima/libmali-rockchip/releases (which is only needed for HDR tone-mapping).

After that ... hardware acceleration was borked and various GUIs would not open. After uninstalling it everything is working again.

closed 2024-06-03 by Joshua-Riek

AkashRajpurohit, 2024-07-01:
@jackduckworth What did you do for HDR tone mapping in this case? How are you setting up hw acceleration on Orange Pi 5 with 24.04 image?

jackduckworth, 2024-07-01:
JR replied to a reddit thread: https://www.reddit.com/r/OrangePI/s/ADtrO1WlPM

Basically it's part of the 24.04 image now. You can install it on an older image with
_sudo apt-get install libmali-g610-x11_

All the details are here:
https://github.com/Joshua-Riek/ubuntu-rockchip/issues/879

AkashRajpurohit, 2024-07-01:
Thanks for sharing, I had gone through the linked GitHub issue as well and while the hardware acceleration works fine, the HDR tone mapping still fails for me and running the verification step provided in the jellyfin doc also fails.