Joshua-Riek/ubuntu-rockchip #839: No HDMI output with v.24.04 desktop installation on Orange Pi 3B

ddonevski, 2024-05-29:
Hi, while v.22.04 desktop installation boots from SD card on my Orange Pi 3B without problems, the v.24.04 gives no HDMI output. Did anyone else experience this?

ppmou, 2024-06-03:
I had the same problem.

hugorteixeira, 2024-06-04:
Me too, 24.04 was working fine before (eMMc), then I rebooted it and nothing. Tried with a ssd with 24.04 as well and nothing. About 2w is being used (says the usb c reader at the cable).

ddonevski, 2024-06-05:
Thanks for sharing your experiences. I couldn't afford to spend much time on this so I was wondering if anyone else had this issue since the image was available for about two weeks, an this issue wasn't reported. So now we know it is a real issue affecting at least some 3B boards. Doesn't seem to affect all 3B boards since some other isuues on 24.04 were reported by 3B users - obviously their display worked.

hugorteixeira, 2024-06-06:
My 3b ubuntu was installed in my emmc working 100%. I had to reboot it and nothing, blank. Tried with a new ubuntu image in the micro sd, nothing. Came here, wrote a post. Then I was playing with other images from Orange Pi itself, DietPi and whatnot and they all worked. But then, while testing, I forgot to insert the micro sd card once and turned the 3b on with the regular Ubuntu emmc. It just worked. I have no idea why. Maybe something with USB ports? Its very weird because the new micro sd card didn´t work that time too and I tried booting it many times.

ddonevski, 2024-06-07:
Yeah, these boards are picky about SD cards, they don't tolerate slow cards. But my experience with different images was that when provided a good SD card, images boot reliability. This one (Ubuntu Rockchip 24.04) is the only one that doesn't and this behavior was also consistent for me - it never booted. I never experienced inconsistent behavior such that some image boots occasionally.

agung483, 2024-06-10:
I have the same problem but for Rock 5b, i can see the login page again after turn off and turn on the monitor.

vinser, 2024-10-12:
> I have the same problem but for Rock 5b, i can see the login page again after turn off and turn on the monitor.

Yes, I have the same on my Orange Pi 5B when starting up the board from poweroff. There is no signal on HDMI. Switching monitor off/on helps. I have this issue after install 24.04 on EMMC from downloaded image. 22.04 had no this problem.

This problem was reported on [ask Ubuntu](https://askubuntu.com/questions/1512058/display-loses-signal-after-splash-screen-ubuntu-24-04).  They say it can be healed with curnel upgrade/degrade

ddonevski, 2024-10-12:
When I first opened this issue a few months ago, switching the display off and on didn't make any difference. However, I experienced this issue with release which was current (latest) at that time. With next release, the issue disappered and everything worked.

closed 2025-02-17 by github-actions[bot]