Joshua-Riek/ubuntu-rockchip #841: wifi lost after upgrade with 24.04 on orange pi 5 pro

thanhtantran, 2024-05-29:
Device: Orange Pi 5 Pro

Version: Ubuntu 24.04

Problem: First run with Ubuntu 24.04, i still have wifi on, and connect successfully 
but after run `sudo apt update && sudo apt upgrade`

the wifi is lost, cannot find in usb or pci list, nmtui neither

[image]

thanhtantran, 2024-05-30:
[mydmesg.log](https://github.com/Joshua-Riek/ubuntu-rockchip/files/15493730/mydmesg.log)

My DMESG log

hqnicolas, 2024-06-02:
Try to mark hold some packages on first run
like this:
> sudo apt-mark hold linux-image-6.1.0-1014-rockchip  linux-headers-6.1.0-1014-rockchip
you can try to figure out what package are upgraded and hold them

thanhtantran, 2024-06-03:
No, this is not the problem of kernel. I look around the issue and found this 
https://github.com/Joshua-Riek/ubuntu-rockchip/issues/674

it is exactly what i encoutered:
- cold boot (remove power plug) wifi + bluetooth work
- warm boot (entering `sudo reboot`) wifi + bluethoot down

ameer4, 2024-06-07:
I have wifi problem with usb wifi 6 dongle (mt7921)
Just using gsconnect to control mouse from my phone,  and moving the cursor for a few seconds it freeze (wifi disconnected)
Is this related to this bug? Or d I need to open a new one?

How do I temporarily fix it?

jinglong485, 2024-06-12:
I have the same problem. Also after reboot the Bluetooth mac address changes so previously connected keyboard/mouse cannot be recognized. I found the work around about bluetooth at #112 . but the wifi problem still exists.

Joshua-Riek, 2024-06-12:
The Orange Pi 5 Pro uses BCMDHD SDIO as the wireless driver so i honestly don't expect WiFi to work at all. This is because BCMDHD PCIE is enabled and used by multiple other devices. The BCMDHD SDIO and PCIE drivers can not be enabled at the same time. Orange Pi themselves use a different kernel just for the Orange Pi 5 Pro for this very reason. But it's way to much work to manage two extra kernels just for the Orange Pi 5 Pro.

I think a fix would be to blacklist BCMDHD PCIE and compile BCMDHD SDIO as a DKMS module.

tgutierrez, 2024-06-27:
@Joshua-Riek :

I have an Orange PI 5 Pro, and after a week of tinkering, I managed to get the WIFI to work after a reboot. However, I lost analog audio. Disclaimer: I'm a .net dev and I'm absolutely a noob in terms of Linux.

What I did was to modify the device tree:

From:

```
&i2c3 {
	status = "okay";

	es8388: es8388@11 {
		status = "okay";
		#sound-dai-cells = <0>;
		compatible = "everest,es8388", "everest,es8323";
		reg = <0x11>;
		clocks = <&mclkout_i2s2>;
		clock-names = "mclk";
		assigned-clocks = <&mclkout_i2s2>;
		assigned-clock-rates = <12288000>;
		pinctrl-names = "default";
		pinctrl-0 = <&i2s2m1_mclk>;
	};
};

```
to 
```
&i2c3 {
	status = "okay";

	es8388: es8388@10 {
		status = "okay";
		#sound-dai-cells = <0>;
		compatible = "everest,es8388", "everest,es8323";
		reg = <0x10>;
		clocks = <&mclkout_i2s2>;
		clock-names = "mclk";
		assigned-clocks = <&mclkout_i2s2>;
		assigned-clock-rates = <12288000>;
		pinctrl-names = "default";
		pinctrl-0 = <&i2s2m1_mclk>;
	};
};
```

I got that after comparing the orange pi device tree with the one on this distro. 

Of course  I get an error: `ES8323 3-0010: i2c recv Failed` , but at least wifi survives a reboot.

I also tried recompiling the kernel with 

```
diff --git a/debian.rockchip/config/config.common.ubuntu b/debian.rockchip/config/config.common.ubuntu
index 81ad34598c39..a18d5f4c1194 100644
--- a/debian.rockchip/config/config.common.ubuntu
+++ b/debian.rockchip/config/config.common.ubuntu
@@ -3127,10 +3127,10 @@ CONFIG_WIFI_BUILD_MODULE=y
 # CONFIG_WIFI_GENERATE_RANDOM_MAC_ADDR is not set
 CONFIG_BCMDHD=y
 CONFIG_AP6XXX=m
-# CONFIG_BCMDHD_SDIO is not set
-CONFIG_BCMDHD_PCIE=y
+CONFIG_BCMDHD_SDIO=y
+# CONFIG_BCMDHD_PCIE is not set
 CONFIG_BCMDHD_FW_PATH="/lib/firmware/ap6275p/fw_bcmdhd.bin"
-CONFIG_BCMDHD_NVRAM_PATH="/lib/firmware/ap6275p/nvram.txt"
+CONFIG_BCMDHD_NVRAM_PATH="/lib/firmware/rkwifi/nvram_ap6256.txt"
 # CONFIG_BCMDHD_STATIC_IF is not set
 CONFIG_WLAN_VENDOR_RSI=y
 # CONFIG_RSI_91X is not set
```
...but then I lost wifi completely. Hopefully there's a way to get both, but this is my first time doing this stuff, so not sure how to keep troubleshooting this. 

Thanks!

btw, thank you for your work!!

Joshua-Riek, 2024-07-14:
I started to make a dkms package to fix this issue, it's still a work in progress.

https://github.com/Joshua-Riek/bcmdhd-dkms

Joshua-Riek, 2024-07-15:
I'm starting to think there is a device tree issue with the configuration of SDIO.
```bash
[   16.378317] bcmdhd_sdio: loading out-of-tree module taints kernel.
[   16.388153] [dhd] _dhd_module_init: in Dongle Host Driver, version 101.10.591.52.27 (20240409-1)(20240411-2)(d83d8d7)
               /var/lib/dkms/bcmdhd-sdio/101.10.591.52.27-1/build/src compiled on Jul 15 2024 at 00:52:07
[   16.388184] [dhd] STATIC-MSG) dhd_static_buf_init : 101.10.361.36 (wlan=r892223-20231107-1)
[   16.388212] [dhd] STATIC-MSG) dhd_init_wlan_mem : prealloc ok for index 0: 1102848(1077K)
[   16.388225] dhd_wlan_init_gpio: unknown oob irqflags !
[   16.388227] [dhd] dhd_wlan_init_gpio: WL_HOST_WAKE=-1, oob_irq=-1, oob_irq_flags=0x0
[   16.388229] [dhd] dhd_wlan_init_gpio: WL_REG_ON=-1
[   16.388231] [dhd] dhd_wifi_platform_load: Enter
[   16.388234] [dhd] Power-up adapter 'DHD generic adapter'
[   16.388331] [dhd] dummy_probe: func->num=0x3; 
[   16.388334] [dhd] dummy_probe: class=0xff; vendor=0x2d0; device=0xffff
[   16.388391] [dhd] wifi_platform_set_power = 1, delay: 200 msec
[   16.388395] [dhd] ======== PULL WL_REG_ON(-1) HIGH! ========
[   16.595760] [dhd] wifi_platform_set_power = 1, sleep done: 200 msec
[   16.595777] [dhd] wifi_platform_bus_enumerate device present 1
[   16.595780] [dhd] ======== Card detection to detect SDIO card! ========
[   18.679311] [dhd] failed to power up DHD generic adapter, 0 retry left
[   18.695570] [dhd] wifi_platform_set_power = 0, delay: 10 msec
[   18.695583] [dhd] ======== PULL WL_REG_ON(-1) LOW! ========
[   18.706698] [dhd] wifi_platform_set_power = 0, sleep done: 10 msec
[   18.706713] [dhd] wifi_platform_bus_enumerate device present 0
[   18.706716] [dhd] ======== Card detection to remove SDIO card! ========
[   18.706719] [dhd] failed to power up DHD generic adapter, max retry reached**
[   18.706723] [dhd] unregister wifi platform drivers
[   18.706725] [dhd] wifi_platform_bus_enumerate device present 0
[   18.706727] [dhd] ======== Card detection to remove SDIO card! ========
[   18.706732] [dhd] _dhd_module_init: Failed to load the driver, try cnt 0
[   18.706735] [dhd] _dhd_module_init: Failed to load driver max retry reached**
[   18.706738] [dhd] STATIC-MSG) dhd_static_buf_exit : Enter
[   18.706808] [dhd] _dhd_module_init: Exit err=-19
```

jamstec0, 2024-07-18:
Is it possible to use brcmfmac for the orange pi 5 pro wifi driver?
Have you tried this?

Joshua-Riek, 2024-07-18:
I doubt it, but I'd love to be proven wrong. This BCMDHD driver is terrible.

jamstec0, 2024-07-25:
The root cause of this error appears to be an incomplete unbinding of the SDIO device. While a shutdown ensures all devices are unbound, a reboot doesn't always reset the previous MAC address and other settings, leading to this issue.

To fix this, run `echo 'mmc2:0001:1' | sudo tee /sys/bus/sdio/drivers/brcmfmac/unbind` before rebooting. Note that `mmc2:0001:1` might differ based on your system's configuration.

Joshua-Riek, 2024-07-25:
This is great! Thanks for finding this, I'll test it out after work! But now the root cause has been found, and we can start looking into the driver in depth to implement a proper fix.

jeremysalwen, 2024-08-01:
So I have an Orange Pi 5 Pro, and I installed the 24.04 server image.  The Wifi worked out of the box for me once I installed network-manager.  However, bluetooth is not showing up:
```
$ sudo service bluetooth start
$ sudo bluetoothctl
[bluetooth]# list
[bluetooth]#
```

Relevant dmesg logs from startup:
```
[    7.786212] pci_bus 0004:41: busn_res: [bus 41-4f] end is updated to 41
[    7.786234] pci 0004:40:00.0: BAR 8: assigned [mem 0xf4200000-0xf42fffff]
[    7.786241] pci 0004:40:00.0: BAR 6: assigned [mem 0xf4300000-0xf430ffff pref]
[    7.786248] pci 0004:40:00.0: BAR 7: assigned [io  0x100000-0x100fff]
[    7.786256] pci 0004:41:00.0: BAR 0: assigned [mem 0xf4200000-0xf4203fff 64bit]
[    7.786299] pci 0004:41:00.0: BAR 4: assigned [io  0x100000-0x1000ff]
[    7.786316] pci 0004:40:00.0: PCI bridge to [bus 41]
[    7.786322] pci 0004:40:00.0:   bridge window [io  0x100000-0x100fff]
[    7.786329] pci 0004:40:00.0:   bridge window [mem 0xf4200000-0xf42fffff]
[    7.787247] pcieport 0004:40:00.0: PME: Signaling with IRQ 117
[    7.790927] input: rockchip-es8388 Headset as /devices/platform/es8388-sound/sound/card2/input4
[    7.792220] Initializing XFRM netlink socket
[    7.792301] NET: Registered PF_PACKET protocol family
[    7.792492] [BT_RFKILL]: Enter rfkill_rk_init
[    7.792510] [WLAN_RFKILL]: Enter rfkill_wlan_init
[    7.793279] [WLAN_RFKILL]: Enter rfkill_wlan_probe
[    7.793318] [WLAN_RFKILL]: can't find rockchip,grf property
[    7.793334] [WLAN_RFKILL]: wlan_platdata_parse_dt: wifi_chip_type = ap6256
[    7.793349] [WLAN_RFKILL]: wlan_platdata_parse_dt: enable wifi power control.
[    7.793364] [WLAN_RFKILL]: wlan_platdata_parse_dt: wifi power controled by gpio.
[    7.793396] [WLAN_RFKILL]: wlan_platdata_parse_dt: WIFI,poweren_gpio = 24 flags = 0.
[    7.793432] [WLAN_RFKILL]: wlan_platdata_parse_dt: WIFI,host_wake_irq = 0, flags = 0.
[    7.793454] [WLAN_RFKILL]: wlan_platdata_parse_dt: The ref_wifi_clk not found !
[    7.793471] [WLAN_RFKILL]: rfkill_wlan_probe: init gpio
[    7.793486] [WLAN_RFKILL]: rfkill_set_wifi_bt_power: 1
[    7.793502] [WLAN_RFKILL]: Exit rfkill_wlan_probe
[    7.794519] [BT_RFKILL]: bluetooth_platdata_parse_dt: get property: uart_rts_gpios = 122.
[    7.794559] [BT_RFKILL]: bluetooth_platdata_parse_dt: get property: BT,reset_gpio = 29.
[    7.794585] [BT_RFKILL]: bluetooth_platdata_parse_dt: get property: BT,wake_gpio = 22.
[    7.794610] [BT_RFKILL]: bluetooth_platdata_parse_dt: get property: BT,wake_host_irq = 21.
[    7.794678] [BT_RFKILL]: Request irq for bt wakeup host
[    7.794764] [BT_RFKILL]: ** disable irq
[    7.794900] [BT_RFKILL]: bt_default device registered.
[    7.795012] Key type dns_resolver registered
[    7.795040] Key type ceph registered
[    7.795409] libceph: loaded (mon/osd proto 15/24)
```

Relevant dmesg logs after running `sudo service bluetooth start`:
```
[ 1108.726843] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[ 1108.726870] Bluetooth: BNEP filters: protocol multicast
[ 1108.726893] Bluetooth: BNEP socket layer initialized
```

Joshua-Riek, 2024-08-02:
I forgot to pre-install the rfkill package, so bluetooth will not work in the startup service. However I have released an update, so if you update your system and reboot it will be fixed.

jeremysalwen, 2024-08-02:
Thank you that fixed it for me!

jamstec0, 2024-08-08:
The Wi-Fi issue on the Orange Pi 5 Pro doesn’t seem to be a simple problem... The standard logs do not provide enough information, so I want to perform detailed debugging of `brcmfmac`, but I’m not very familiar with driver debugging and don’t know how to display the logs. I also can't find any documentation on `brcmfmac`. What settings should I configure?

defencedog, 2024-08-09:
@jamstec0 same here for me using OrangePi 3b v2.1 (AP6256 module) ... wifi works great BT doesn't!

closed 2025-02-17 by github-actions[bot]