Joshua-Riek/ubuntu-rockchip #848: Decklink card - kernel panic

vtv3000, 2024-05-30:
Hi i try to install decklink m.2 card on ubuntu 24.04 - driver install ok BUT not started. after install  write -"try reboot"

when i reboot i see mesage and if card plugin not start
when card unplug - boot ok

d50 0a4c7e20 ffff8000
ffff8000 08010224
08010224 ffff8000 0a983d70 ffff8000 08010338 ffff8000
0a983df0 ffff8000
080912f0 ffff8000 09b8df40 ffff8000 0a1cd3a0 ffff8000
d90
00a80000 ffff0001
0967c008 ffff8000 09b7c008 ffff8000 00000004 00000000
3db0
Bddo 00000001 04208040 fffea80a 00000000 09b7e8e0 ffff8000 092962d4 ffff8000
---[ end trace 0000000000000000 ]---
rk-pcie fe170000.pcie: PCIe Linking... LTSSM is 0x3
rk-pcie fe170000. pcie: PCIe Link Fail, LTSSM is 0x3, hw retries=1
rk-pcie fe170000.pcie: failed to initialize host
rkcifhw fdce0000.rkcif: Adding to tommu group 16
rkcifhw fdce0000.rkcif: No reserved memory region assign to CIF
platforn mtd_vendor_storage: deferred probe pending

dexterphillips, 2024-05-31:
We have found this as well. I have not been able to get a black magic card working on any rockchip board (tried friendlyelec, orange and radxa) using every combination of manufacture provided OS, and Joshua Ubuntu. I have also experimented with magewell and yuan cards with limited success.

**Adding to the information pool:** 
_Using the Joshua 5.10 kernel with 22.04_
Blackmagic - Kernel Panic.      Magewell (using jetson tx2 drivers) - Working     Yuan (using rockchip drivers) - Working

_Using the Joshua 6.1  Kernel with 24.04_ 
Blackmagic - Kernel Panic.      Magewell (using jetson tx2 drivers) - Kernel Panic   Yuan - No drivers available

So interestingly, something (probably to do with memory management) has changed between 5.10 and 6.1 which causes the magewell drivers to not work (even when compiled on the device). This may well be related to the black magic card not working as it seems like very similar symptoms in the kernel logs. I have attached kernel logs when using the magewell card in case someone cleverer than me can spot an obvious issue!!

FYI - I have also sent a rockchip board to magewell who are going to investigate it on their side as well. Will continue to update as the situation progresses.

Here are the attached kernel logs. Most useful is probably the one from when I installed the driver with the card installed. You can see where it tries to load the kernel module, and then dies. (Attachment number 2).
[Logs from failed install 2.zip](https://github.com/user-attachments/files/15511593/Logs.from.failed.install.2.zip)

dexterphillips, 2024-05-31:
This issue has been discussed previously:

https://forum.blackmagicdesign.com/viewtopic.php?f=12&t=190885
https://forum.blackmagicdesign.com/viewtopic.php?f=12&t=192592
https://forum.radxa.com/t/blackmagic-decklink-sdi-micro-in-rock5b/18872/6
https://github.com/Joshua-Riek/ubuntu-rockchip/discussions/413

As suggested in some of the above issues, VMallocing at boot does not make a difference. 
But the fact that we have successfully seen video on 5.10 from magewell and yuann proves to me that it is not a hardware / pcie lane limitation (as has been suggested) and is in fact a driver or OS issue. 

I have been in touch with blackmagic directly, but they were not interested in investigating the issue because not enough people are affected. So I would suggest that more people should email them to show interest in resolving the issue. 

Also - and I appreciate you're incredibly busy with everything else - but if josh would mind having a quick glance at the logs provided to see if it looks like it might be an OS issue or a driver issue?

I have a test bench set up with every combination of cards available, so if I can provide further information let me know!

Joshua-Riek, 2024-05-31:
Both the Rockchip 5.10 and 6.1 are Android kernel's, there is a high likelihood these drivers where not made with Android compatibility in mind. I can ensure its not the operating system at fault, its a kernel / driver issue. Honestly if you email the company responsible for the hardware / drivers they will likely do nothing as this is not mainline Linux, its an Android hacked-to-bits port. However digging into this myself would be to much work as this is an open source side project.

dexterphillips, 2024-05-31:
**New information**
The magewell card - which works with Joshua's 5.10, Ubuntu 22.04 build 
**Does work** with the FriendlyElec kernel 6.1.57, Ubuntu 22.04 build

I don't know how different their kernel is to the one josh uses. But I thought I'd share this finding anyway. 

The driver succesfully compiles, and I can see video through the card.

The friendly elec build described can be found: 

https://drive.usercontent.google.com/download?id=1Aws16hOG5UmYbtz0mJWbn1ZoRfsB6Ef2&export=download&authuser=0&confirm=t&uuid=33b68ea3-4f10-45bf-9602-3899c0f7b1dd&at=APZUnTW8iWVzh22vRFwOxKRQXd7C:1717173258435

https://wiki.friendlyelec.com/wiki/index.php/NanoPC-T6

vtv3000, 2024-05-31:
Can you test decklink on this kernel ?

dexterphillips, 2024-06-04:
Blackmagic released Desktop Video 14.0.1 yesterday.
Tried it out on my T6 on the following images:

FriendlyElec 6.1 22.04 - Drivers install (after some fiddling) - card starts to output for several seconds (according to gstreamer logs, don't have a cable to check if video actually comes out) - but then it fails. DMESG BlackmagicIO "Warning ran out of free frames" // Unable to handle kernel paging request at virtual address

Joshua 6.1 24.04 - Same Error

dexterphillips, 2024-06-04:
Don't have any h.fl cables with me (away from home on a long job with work). But the fact the output is broken is a good litmus test that the Blackmagic card doesn't work given previous attempts! 
I'm currently compiling armbian and will try all the cards I have on that. See if it's any different to Joshua's ubuntu builds. 

Shall feed back later.

vtv3000, 2024-06-06:
Yes it's interesting test !

dexterphillips, 2024-06-06:
Interesting Result: 
Taking the friendly elec 6.1 22.04 build
And then hacking in an upgrade to ubuntu 24 (with joshua ppa packages)...

We have working magewell drivers.

So that's cool.

dexterphillips, 2024-06-06:
So potentially from this we learn:
Joshua ubuntu 24 bits are not casuing the issue 
There is some difference in the 6.1 friendly kernel, and the joshua one that causes it issues.
(They are both based off the same upstream androidy hacky kernel I believe - the radaxa one?).

vtv3000, 2024-06-06:
On orange pi official download page I see Ubuntu 6.1 . May be this kernel better ?

dexterphillips, 2024-06-08:
New update: 

We've managed to take Joshua rockchip 6.1 ubuntu 24 and swap out the kernel with his new 5.10 kernel.

This gives us working magewell and working yuann capture cards, with ubuntu 24 on rockchip devices. Success!

This is the best compatability with capture devices we've had so far. So I think that's conclusively pointing to the 6.1 Josh kernel causing issues for the capture drivers in their current form. Magewell are still currently looking at drivers for the 6.1 kernel, so we'll see what they say... 

But for anyone who wants capture cards with rockchip... 

1 - Get magewell eco cards
2 - For ubuntu 22, use joshua 5.10, 22. For ubuntu 24, use joshua 6.1 24 and downgrade the kernel to the joshua 5.10.
3 - To install magewell drivers, use the Jetson TX2 ones on their website. Make sure kernel headers, make and gcc are installed. Then run the driver install script.

Thanks for everything Josh! And big shoutout to magewell for being infinitely more responsive and helpful than any other capture card manufacturer.

Happy capture carding everyone!

vtv3000, 2024-06-08:
Very good job ! Thx.

Bot for my project still not progress:(

I need 6.1 kernel

vtv3000, 2024-12-08:
Hi I waiting new release of Ubuntu and drivers from Blackmagic . And after new release I will check . But now not progress .. 

Kernel panic .

closed 2025-02-10 by github-actions[bot]