Joshua-Riek/ubuntu-rockchip #850: [24.04] Target board RADXA Rock 5B is unbootable

eshcheglov, 2024-05-31:
Hi,

I tried to run Ubuntu 24.04 on my **Radxa 5B** using the following steps:

* Downloaded the image ubuntu-24.04-preinstalled-server-arm64-rock-5b.img.xz
*  Used Etcher with verification to flash the image onto an SD card
* Inserted the card into the board and powered it on
* Through the serial console, I see that during the Linux kernel boot, the board keeps rebooting and goes into an endless reboot loop

I suspect it is a problem with the DT or U-boot, but I can't find clear instructions on how to properly flash the board. I have an SD card with the standard Armbian distribution which boots correctly, but I don't know where to find the necessary U-boot binary files that I need to use for flashing.

Serial console log: [radxa_5b_reboot.log](https://github.com/user-attachments/files/15512065/radxa_5b_reboot.log)

Thank you!

watou, 2024-05-31:
Be sure, no matter what else, you are using dumb 5V 4A non-PD power supply. Working with other OS is not reason to discount requirement.

eshcheglov, 2024-05-31:
Thank you!

I replaced the power supply with a standard one from Raspberry Pi (P0320 AC ADAPTOR) and the system booted up. It seems my previous power supply was PD. The issue is that the OS kernel handles PD negotiation, and sometimes it does this too late, leading to reboots.

To people from the future who find this ticket: a list of working/non-working chargers is on the official Radxa website: https://wiki.radxa.com/Rock5/5b/power_supply

closed 2024-05-31 by eshcheglov