Joshua-Riek/ubuntu-rockchip #852: Wifi not working on Radxa Zero 3 with new Ubuntu 22.04 image

RSchwan, 2024-05-31:
Hi Joshua,

I've been using the "old" Ubuntu 22.04 image on my Radxa Zero 3 and I got everything working using cloud-init. I now flashed the "new" Ubuntu 22.04 image on the same Radxa Zero 3 using exactly the same network-config, but wifi is not correctly configured anymore, i.e., it doesn't connect with the access point. I've tried a bunch of things without much luck, and since I don't have internet access, I also can't install any additional networking/tooling packages. Ideally, I would like to use the newest image, and I'm also willing to help to fix this issue, but I'm not entirely sure where to start. :)

Joshua-Riek, 2024-05-31:
I recently packaged a dkms package for aic8800, you can try to enter the below commands:
```bash
sudo apt update
sudo apt install -y dkms aic8800-sdio-dkms```

RSchwan, 2024-06-02:
If I run the commands, I get `Unable to locate package aic8800-sdio-dkms`. I assume this is due to not having an internet connection. In a sense, it's a chicken-and-egg problem, since I'm kind of locked into the system without an internet connection.

Is there a way to build an image which already includes this driver?

RSchwan, 2024-06-02:
I should maybe also mention that I'm running the `ubuntu-22.04(.3)-preinstalled-server-arm64-radxa-zero3` images. More specifically, wifi is working with v1.33 and stopped working with v2.1.0.

Joshua-Riek, 2024-06-02:
Hmmm i will look at it, i've been hyper-focusing on Radxa kernel stuff today, I'm sure the DKMS will fix the issue, i just need to double check if there will be any drivers clashing as aic8800 is already enabled in the kernel. If there is no clashing, this should be a very easy fix.

Joshua-Riek, 2024-06-02:
Ok, i looked into this and blacklisted the aic8800 kernel driver and replaced it by the dkms package which works perfectly. When the nightly GitHub build (linked below) starts, this problem should be fixed for both Ubuntu 22.04 an 24.04.

https://github.com/Joshua-Riek/ubuntu-rockchip/actions/workflows/nightly.yml

RSchwan, 2024-06-03:
Ok, after playing round a bit more I managed to get the wifi working.

Looking at the journalctl logs, I got the following errors:
[image]
which I fixed by installing the `wpasupplicant` package. It seems to me this is a required package, since netplan configures for it.

Then I ran into some driver issues:
[image]
Revering your changes and using the old aic8800 kernel drivers fixed this issue and the wifi finally connected to the access point.

[Here](https://github.com/Joshua-Riek/ubuntu-rockchip/compare/main...RSchwan:ubuntu-rockchip:main#diff-8e82eec08c9cd625bf0bae79025211bd41604ce938d443eb3f188949f2f92a5d) the changes I made to the build script. I'm not sure what's the issue with the new drivers, but they didn't work for me.

Joshua-Riek, 2024-06-04:
I updated the ubuntu-rockchip meta package to include wpasupplicant, i confirmed that the latest nightly build has this issue resolved.

https://github.com/Joshua-Riek/ubuntu-rockchip-meta/releases/tag/22.04.2
https://github.com/Joshua-Riek/ubuntu-rockchip-meta/releases/tag/24.04.5

closed 2024-06-04 by Joshua-Riek