Joshua-Riek/ubuntu-rockchip #854: 22.04 OP5+ Ubuntu server image: Samba / avahi install fails

icojb25, 2024-06-01:
I get the following on my OP5+ with the 22.04 image when trying to install `avahi-daemon`

```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
avahi-daemon is already the newest version (0.8-5ubuntu5.2).
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up flash-kernel (3.104ubuntu16+21) ...
flash-kernel: deferring update (trigger activated)
Processing triggers for initramfs-tools (0.140ubuntu13.4) ...
update-initramfs: Generating /boot/initrd.img-5.10.0-1005-rockchip
flash-kernel: installing version 5.10.0-1005-rockchip
mv: cannot move '/tmp/flash-kernel.F6zaIu3J/vmlinuz' to '/boot/firmware/vmlinuz'
: No such file or directory
run-parts: /etc/initramfs/post-update.d//flash-kernel exited with return code 1
dpkg: error processing package initramfs-tools (--configure):
 installed initramfs-tools package post-installation script subprocess returned 
error exit status 1
Errors were encountered while processing:
 initramfs-tools
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Any ideas?

Joshua-Riek, 2024-06-01:
You installed the flash-kernel package for some reason, please remove it with the below command:
```bash
sudo apt-get purge -y flash-kernel
```

icojb25, 2024-06-02:
Thanks a lot. That worked. 

Any idea how that would have been installed? I am almost 100% sure i did not install it. I have (had) a clean install of 22.04 server with only Casa OS over the top.  I believe I installed avahi myself afterwards.

In any event, thanks a lot.

Joshua-Riek, 2024-06-02:
I cant say there are a lot of factors. But I will keep a general eye out as if a user updates their system after a release and this package is installed I would say that is an issue.

closed 2024-06-02 by Joshua-Riek

Joshua-Riek, 2024-06-06:
I found the issue https://github.com/Joshua-Riek/linux-rockchip/commit/d245a216eda176f505412ed3a73d298e80294f73, uploading a new kernel now.

icojb25, 2024-06-06:
Thanks a lot for the fix