Joshua-Riek/ubuntu-rockchip #857: /boot/exilinux/extlinux.conf cgroup_enable=memoryquiet in append line

hakotani-o, 2024-06-02:
Hello.
Thank you for great work.

I download from https://joshua-riek.github.io/ubuntu-rockchip-download/boards/orangepi-5.html.
ubuntu-24.04-preinstalled-desktop-arm64-orangepi-5.img.xz

/boot/extlinux/extlinux.conf have typo in append line.
Dose cgroup_enable=memoryquiet need space ?  

append root=UUID=e3f26f82-39fc-44f2-ad85-3b4762116d1c rootwait rw console=ttyS2,1500000 console=tty1 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memoryquiet splash plymouth.ignore-serial-consoles

Thank you.

Joshua-Riek, 2024-06-02:
Yeah its a typo, I fixed it here: https://github.com/Joshua-Riek/livecd-rootfs/commit/6efb5e07b68cfc41a9c4268c81683d02752e6a3f

closed 2024-06-02 by Joshua-Riek

hakotani-o, 2024-06-02:
Thank you for your time.
I look forward to working with you.