Joshua-Riek/ubuntu-rockchip #860: build error in /scripts/config-image.sh

Manu7788, 2024-06-04:
i tried to build the project on an vistual linux machine with ubuntu 22.04.04. 
working well until it reaches the `/script/config-image.sh`

```
mount: rootfs/dev: mount point does not exist.
Error: in ./scripts/config-image.sh on line 86
Error: in ./build.sh on line 229
```

any ideas how to fix this problem?
br,
manuel

Joshua-Riek, 2024-06-04:
Builds fine for me. Id recommend forking the repo and build with github actions.

closed 2024-06-04 by Joshua-Riek

Manu7788, 2024-06-04:
its just building fine now after i used your recommended packages before compile:

```
sudo apt-get install -y build-essential gcc-aarch64-linux-gnu bison \
qemu-user-static qemu-system-arm qemu-efi u-boot-tools binfmt-support \
debootstrap flex libssl-dev bc rsync kmod cpio xz-utils fakeroot parted \
udev dosfstools uuid-runtime git-lfs device-tree-compiler python2 python3 \
python-is-python3 fdisk bc debhelper python3-pyelftools python3-setuptools \
python3-distutils python3-pkg-resources swig libfdt-dev libpython3-dev
```

MattiaGrigoli, 2024-10-05:
it still doesn't work after installing all the packages, i'm getting the same error

wanyingwe, 2024-10-10:
@MattiaGrigoli I have the same problem, Have you fix it ?

MattiaGrigoli, 2024-10-10:
> @MattiaGrigoli I have the same problem, Have you fix it ?

no, still working on it.

wanyingwe, 2024-10-12:
> > @MattiaGrigoli I have the same problem, Have you fix it ?
> 
> no, still working on it.

I re-cloned the latest version, and it works