Joshua-Riek/ubuntu-rockchip #861: Rock 5A: No HW Acceleration on Chromium

MyVizDrake, 2024-06-04:
Had noticed the CPU seemed more loaded when using chromium and did some digging ..

From Chromium about:gpu
```Graphics Feature Status
Canvas: Software only, hardware acceleration unavailable
Canvas out-of-process rasterization: Disabled
Direct Rendering Display Compositor: Disabled
Compositing: Software only. Hardware acceleration disabled
Multiple Raster Threads: Enabled
OpenGL: Unavailable
Rasterization: Unavailable
Raw Draw: Disabled
Video Decode: Unavailable
Video Encode: Software only. Hardware acceleration disabled
Vulkan: Disabled
WebGL: Unavailable
WebGL2: Unavailable
WebGPU: Disabled
Driver Bug Workarounds
clear_uniforms_before_first_program_use
count_all_in_varyings_packing
enable_webgl_timer_query_extensions
exit_on_context_lost
disabled_extension_GL_KHR_blend_equation_advanced
disabled_extension_GL_KHR_blend_equation_advanced_coherent
Problems Detected
WebGPU has been disabled via blocklist or the command line.
Disabled Features: webgpu
Accelerated video encode has been disabled, either via blocklist, about:flags or the command line.
Disabled Features: video_encode
Gpu compositing has been disabled, either via blocklist, about:flags or the command line. The browser will fall back to software compositing and hardware acceleration will be unavailable.
Disabled Features: gpu_compositing
GL driver is software rendered. GPU acceleration is disabled: [59302](http://crbug.com/59302), [315217](http://crbug.com/315217), [1155974](http://crbug.com/1155974)
Disabled Features: accelerated_2d_canvas, accelerated_webgl, accelerated_video_decode, accelerated_video_encode, gpu_rasterization, accelerated_webgl2, android_surface_control, accelerated_gl, vulkan, canvas_oop_rasterization, accelerated_webgpu, skia_graphite
Clear uniforms before first program use on all platforms: [124764](http://crbug.com/124764), [349137](http://crbug.com/349137)
Applied Workarounds: clear_uniforms_before_first_program_use
Mesa drivers in Linux handle varyings without static use incorrectly: [333885](http://crbug.com/333885)
Applied Workarounds: count_all_in_varyings_packing
Disable KHR_blend_equation_advanced until cc shaders are updated: [661715](http://crbug.com/661715)
Applied Workarounds: disable(GL_KHR_blend_equation_advanced), disable(GL_KHR_blend_equation_advanced_coherent)
Expose WebGL's disjoint_timer_query extensions on platforms with site isolation: [808744](http://crbug.com/808744), [870491](http://crbug.com/870491)
Applied Workarounds: enable_webgl_timer_query_extensions
Some drivers can't recover after OUT_OF_MEM and context lost: [893177](http://crbug.com/893177)
Applied Workarounds: exit_on_context_lost
DAWN Info

<CPU> Vulkan backend - SwiftShader Device (LLVM 10.0.0)
[WebGPU Status]
Blocklisted
[Default Toggle Names]
lazy_clear_resource_on_first_use: https://crbug.com/dawn/145: Clears resource to zero on first usage. This initializes the resource so that no dirty bits from recycled memory is present in the new resource.
use_temporary_buffer_in_texture_to_texture_copy: https://crbug.com/dawn/42: Split texture-to-texture copy into two copies: copy from source texture into a temporary buffer, and copy from the temporary buffer into the destination texture when copying between compressed textures that don't have block-aligned sizes. This workaround is enabled by default on all Vulkan drivers to solve an issue in the Vulkan SPEC about the texture-to-texture copies with compressed formats. See #1005 (https://github.com/KhronosGroup/Vulkan-Docs/issues/1005) for more details.
vulkan_use_d32s8: https://crbug.com/dawn/286: Vulkan mandates support of either D32_FLOAT_S8 or D24_UNORM_S8. When available the backend will use D32S8 (toggle to on) but setting the toggle to off will make it use the D24S8 format when possible.
vulkan_use_s8: https://crbug.com/dawn/666: Vulkan has a pure stencil8 format but it is not universally available. When this toggle is on, the backend will use S8 for the stencil8 format, otherwise it will fallback to D32S8 or D24S8.
disallow_unsafe_apis: http://crbug.com/1138528: Produces validation errors on API entry points or parameter combinations that aren't considered secure yet.
use_placeholder_fragment_in_vertex_only_pipeline: https://crbug.com/dawn/136: Use a placeholder empty fragment shader in vertex only render pipeline. This toggle must be enabled for OpenGL ES backend, the Vulkan Backend, and serves as a workaround by default enabled on some Metal devices with Intel GPU to ensure the depth result is correct.
use_vulkan_zero_initialize_workgroup_memory_extension: https://crbug.com/dawn/1302: Initialize workgroup memory with OpConstantNull on Vulkan when the Vulkan extension VK_KHR_zero_initialize_workgroup_memory is supported.
[WebGPU Forced Toggles - enabled]
disallow_spirv: https://crbug.com/1214923: Disallow usage of SPIR-V completely so that only WGSL is used for shader modules. This is useful to prevent a Chromium renderer process from successfully sending SPIR-V code to be compiled in the GPU process.
[Supported Features]
texture-compression-bc
texture-compression-etc2
texture-compression-astc
timestamp-query
timestamp-query-inside-passes
depth-clip-control
depth32float-stencil8
indirect-first-instance
rg11b10ufloat-renderable
bgra8unorm-storage
[... 150 lines cut ...]
SDR white level in nits
203
HDR relative maximum luminance
1
Bits per color component
8
Bits per pixel
24
Refresh Rate in Hz
59
Video Acceleration Information
Decoding
Encoding
Vulkan Information
Device Performance Information
Log Messages
[4150:4150:0604/154328.530434:WARNING:sandbox_linux.cc(393)] : InitializeSandbox() called with multiple threads in process gpu-process.```

This isn't what I recall seeing previously with this SBC and JR Ubuntu. Maybe something broke on 1015?

![neofetch](https://github.com/Joshua-Riek/ubuntu-rockchip/assets/88993550/e5db1a94-d022-4508-8b0d-5e29971b2ef6)

![Chromium About](https://github.com/Joshua-Riek/ubuntu-rockchip/assets/88993550/1249415e-e44a-4b4e-b2dc-608720b5b114)

Willing to consider 'user error' but documenting this just in case it isn't just me. :-)

Joshua-Riek, 2024-06-04:
I installed a fresh system and updated it to the latest with apt. I see no problems with chrome or video playback.

[image]

MyVizDrake, 2024-06-04:
Thanks for the quick check ... lemme reinstall ...

MyVizDrake, 2024-06-04:
Is there a way to reimage an SD card and then transfer that to the installed eMMC? Asking for a friend ... :-)

MyVizDrake, 2024-06-05:
Reinstalled w/ kernel tag 1013. HW acceleration is confirmed as present in Chrome.  apt update / upgrade. reboot. Confirm kernel tag 1015 in use. add display-10fhd  overlay. Reboot.   HW acceleration is no longer present. If necessary I can attempt to duplicate via SD as it is more easily interchanged right now :-)

Joshua-Riek, 2024-06-05:
Uhhhhhhhhhhh what do you mean HW acceleration is no longer presient? is the GPU not working? is MPP not working??

MyVizDrake, 2024-06-05:
Same situation as the original about:gpu ouput as provided in the original note. If you need additional logs or info I will be happy to provide them.  if I need to reinstall to SD ... provide uname -r and screenshots after install, after apt update / upgrade reboot and after enabling the display overlay and reboot I can do it!

Joshua-Riek, 2024-06-05:
can you show a screen shot of the about gpu page on chrome?

Joshua-Riek, 2024-06-05:
Right, so you download the v2.1.0 image and run `sudo apt-get update && sudo apt-get upgrade` then reboot?

MyVizDrake, 2024-06-05:
yes and it was still OK .. Then I enabled the rock-5a-radxa-display-10fhd.dtbo via /etc/defaults/u-boot followed by sudo u-boot-update and then reboot

Joshua-Riek, 2024-06-05:
Hmmm that is very strange, maybe there is a bug in the device tree overlay. I wont be able to test this on my end tonight, but i should be able to investigate tomorrow.

MyVizDrake, 2024-06-05:
Appreciate it! Just trying to make sure I am not going crazy as well and ensuring some weird regression hadn't crept in. I suspect something in the overlay. Will be interesting to see if you see the same thing with your 10hd or 8hd (forget which one you have).  Enjoy the rest of your evening!

MyVizDrake, 2024-06-05:
I can confirm that removing the overlay and rebooting returned things to normal
[image]

Joshua-Riek, 2024-06-06:
I was able to reproduce this on the ROCK 5B, i'm not sure why this issue happens at the moment. It seems that only Chromium is affected.

MyVizDrake, 2024-07-16:
I trust this has proven to be a tough nut to crack, which I understand.  Especially since it only appears to be with Chromium.  

now that I have the whole Panel PC assembled in it's custom 3D printed case / enclosure,  I have fully updated the image and there has been no change. Not that I was expecting any, mind you. Just a data point to show that any Chromium or other HW acceleration package updates since then did not resolve it.

@RadxaStephen does the Radxa team have any investigative ideas ?

Joshua-Riek, 2024-07-17:
When using X11 GPU acceleration will work as expected, but when Wayland is being used it seems to break.

MyVizDrake, 2024-07-17:
Well that's annoying :-)

Weird how Wayland without the display is OK (default install IIRC) while  Wayland with the display is not OK.  Issue is probably at the touch driver level instead of the Radxa device overlay?  Or is the entire HW acceleration stack a cobbled web of backports and hacks?

Joshua-Riek, 2024-07-17:
I suspect this is actually a problem with mesa since GPU acceleration stops working. However Panfork is dead, so i'm not sure what can really be done here besides trying with Panthor, but I currently do not support Panthor.

MyVizDrake, 2024-07-23:
Any value in me trying 22.04 based on [this](https://discord.com/channels/855634073376260096/984971406544621569/1257614820300165160) link on the Radxa Discord?

MyVizDrake, 2024-07-30:
well THIS is interesting .. doesn't explain the ffmpeg / mpv lack of HW acceleration but ... if I right click on the Chromium icon and select "Launch using Discrete Graphics card" ... it works ?

[image]

[image]

Contents of the about:gpu are here
```Graphics Feature Status
Canvas: Hardware accelerated
Canvas out-of-process rasterization: Enabled
Direct Rendering Display Compositor: Disabled
Compositing: Hardware accelerated
Multiple Raster Threads: Enabled
OpenGL: Enabled
Rasterization: Hardware accelerated
Raw Draw: Disabled
Video Decode: Hardware accelerated
Video Encode: Software only. Hardware acceleration disabled
Vulkan: Disabled
WebGL: Hardware accelerated
WebGL2: Hardware accelerated
WebGPU: Disabled
Driver Bug Workarounds
clear_uniforms_before_first_program_use
count_all_in_varyings_packing
enable_webgl_timer_query_extensions
exit_on_context_lost
disabled_extension_GL_KHR_blend_equation_advanced
disabled_extension_GL_KHR_blend_equation_advanced_coherent
Problems Detected
WebGPU has been disabled via blocklist or the command line.
Disabled Features: webgpu
Accelerated video encode has been disabled, either via blocklist, about:flags or the command line.
Disabled Features: video_encode
Clear uniforms before first program use on all platforms: [124764](http://crbug.com/124764), [349137](http://crbug.com/349137)
Applied Workarounds: clear_uniforms_before_first_program_use
Mesa drivers in Linux handle varyings without static use incorrectly: [333885](http://crbug.com/333885)
Applied Workarounds: count_all_in_varyings_packing
Disable KHR_blend_equation_advanced until cc shaders are updated: [661715](http://crbug.com/661715)
Applied Workarounds: disable(GL_KHR_blend_equation_advanced), disable(GL_KHR_blend_equation_advanced_coherent)
Expose WebGL's disjoint_timer_query extensions on platforms with site isolation: [808744](http://crbug.com/808744), [870491](http://crbug.com/870491)
Applied Workarounds: enable_webgl_timer_query_extensions
Some drivers can't recover after OUT_OF_MEM and context lost: [893177](http://crbug.com/893177)
Applied Workarounds: exit_on_context_lost
DAWN Info

<CPU> Vulkan backend - SwiftShader Device (LLVM 10.0.0)
[WebGPU Status]
Blocklisted
[Default Toggle Names]
lazy_clear_resource_on_first_use: https://crbug.com/dawn/145: Clears resource to zero on first usage. This initializes the resource so that no dirty bits from recycled memory is present in the new resource.
use_temporary_buffer_in_texture_to_texture_copy: https://crbug.com/dawn/42: Split texture-to-texture copy into two copies: copy from source texture into a temporary buffer, and copy from the temporary buffer into the destination texture when copying between compressed textures that don't have block-aligned sizes. This workaround is enabled by default on all Vulkan drivers to solve an issue in the Vulkan SPEC about the texture-to-texture copies with compressed formats. See #1005 (https://github.com/KhronosGroup/Vulkan-Docs/issues/1005) for more details.
vulkan_use_d32s8: https://crbug.com/dawn/286: Vulkan mandates support of either D32_FLOAT_S8 or D24_UNORM_S8. When available the backend will use D32S8 (toggle to on) but setting the toggle to off will make it use the D24S8 format when possible.
vulkan_use_s8: https://crbug.com/dawn/666: Vulkan has a pure stencil8 format but it is not universally available. When this toggle is on, the backend will use S8 for the stencil8 format, otherwise it will fallback to D32S8 or D24S8.
disallow_unsafe_apis: http://crbug.com/1138528: Produces validation errors on API entry points or parameter combinations that aren't considered secure yet.
use_placeholder_fragment_in_vertex_only_pipeline: https://crbug.com/dawn/136: Use a placeholder empty fragment shader in vertex only render pipeline. This toggle must be enabled for OpenGL ES backend, the Vulkan Backend, and serves as a workaround by default enabled on some Metal devices with Intel GPU to ensure the depth result is correct.
use_vulkan_zero_initialize_workgroup_memory_extension: https://crbug.com/dawn/1302: Initialize workgroup memory with OpConstantNull on Vulkan when the Vulkan extension VK_KHR_zero_initialize_workgroup_memory is supported.
[WebGPU Forced Toggles - enabled]
disallow_spirv: https://crbug.com/1214923: Disallow usage of SPIR-V completely so that only WGSL is used for shader modules. This is useful to prevent a Chromium renderer process from successfully sending SPIR-V code to be compiled in the GPU process.
[Supported Features]
texture-compression-bc
texture-compression-etc2
texture-compression-astc
timestamp-query
timestamp-query-inside-passes
depth-clip-control
depth32float-stencil8
indirect-first-instance
rg11b10ufloat-renderable
bgra8unorm-storage
[... 163 lines cut ...]
48x48 to 7680x4320 pixels
Decode vp9 profile0
48x48 to 7680x4320 pixels
Encoding
Encode h264 baseline
0x0 to 1920x1080 pixels, and/or 30.000 fps.
Encode h264 main
0x0 to 1920x1080 pixels, and/or 30.000 fps.
Encode h264 high
0x0 to 1920x1080 pixels, and/or 30.000 fps.
Encode vp8
0x0 to 1920x1080 pixels, and/or 30.000 fps.
Vulkan Information
Device Performance Information
Log Messages
[3432:3432:0730/161409.107274:ERROR:gl_display.cc(520)] : EGL Driver message (Error) eglQueryDeviceAttribEXT: eglQueryDeviceAttribEXT
[3432:3432:0730/161409.107402:ERROR:gl_angle_util_vulkan.cc(189)] : Failed to retrieve vkGetInstanceProcAddr
[... cut ...]
[3432:3691:0730/161414.012370:ERROR:v4l2_video_encode_accelerator.cc(1026)] : Failure in termination
[3432:3432:0730/161417.882981:ERROR:gl_surface_presentation_helper.cc(260)] : GetVSyncParametersIfAvailable() failed for 1 times!
[3432:3432:0730/161435.676868:ERROR:gl_surface_presentation_helper.cc(260)] : GetVSyncParametersIfAvailable() failed for 2 times!
[3432:3432:0730/161435.706368:ERROR:gl_surface_presentation_helper.cc(260)] : GetVSyncParametersIfAvailable() failed for 3 times!```

htop looks like it might actually be working !?

MyVizDrake, 2024-07-30:
[image]

Joshua-Riek, 2024-07-31:
Good find, I hope to look a bit more into this over the weekend. I've seen a number of tablet / laptop RK3588 devices that are coming up and this may be a large issue as they use Display Panels.

MyVizDrake, 2024-07-31:
The same "Launch using Discrete Graphics Card" option exists even for MPV but for the life of me I cannot drag / drop the mp4 file into the MPV window and get it to do anything so I cannot test MPV acceleration from within the Desktop. MPV from CLI remains unaccelerated.  I believe ffplay remains unaccelerated as well.  Will keep poking around

Joshua-Riek, 2024-08-02:
I talked to someone in the Orange Pi Discord, and they were saying that using the Orange Pi 5 Plus and Orange Pi display planel do not break hardware acceleration.

MyVizDrake, 2024-08-04:
Thanks. Going to try Samuel’s (Spooky) Arch via SD as an experiment. Running right now with your kernel. Just need to figure out how to activate the display. It’s been so long I forgot how I did it before … display and touch was working with 5.10 Radxa BSP but no WLAN / BT. His Arch with your kernel right now is running with WLAN / BT but no display. Will let you know if anything interesting comes up with this test. If it has the same problem will try another kernel. Your 24.04 remains on the eMMC.

MyVizDrake, 2024-08-05:
Spooky from the Radxa forum helped me with enabling the overlay and interestingly enough, Chromium [127.0.6533.88 (Official Build) Arch Linux ARM (64-bit)] and it seems to be working fine.  It is running your 6.1 kernel

[image]

Not sure what it means, but ... yout Ubuntu remains on the eMMC for now and Arch is on the SD card.  If you'd like me to provide any information from either system let me know!

Joshua-Riek, 2024-08-05:
Im Spooky on the Radxa forum lol

MyVizDrake, 2024-08-05:
ha! i knew that !!  i meant Samuel !

Joshua-Riek, 2024-08-05:
So GPU acceleration works with the display panel and panfork on Arch?

MyVizDrake, 2024-08-05:
it appears so!  What can i get to you to show what his Arch is using ? let me know the logs or command outputs you'd like to see and I'll get them for you!

Joshua-Riek, 2024-08-05:
Can you install glmark2 and start the program? It will show which driver is being used.

MyVizDrake, 2024-08-05:
[image]

although to be honest now using it there is some significant laginess in moving windows around the desktop ....

Joshua-Riek, 2024-08-05:
Yeah this looks like Panthor

54601, 2025-01-05:
I'm experiencing the same issue with hardware acceleration in chrome. I used the latest version (ubuntu-24.04-preinstalled-desktop-arm64-orangepi-5-plus.img) and upgrated in gui. However, right click and launch using discrete gpu doesn't solve the problem for me. Is there any fix?

54601, 2025-01-05:
> I'm experiencing the same issue with hardware acceleration in chrome. I used the latest version (ubuntu-24.04-preinstalled-desktop-arm64-orangepi-5-plus.img) and upgrated in gui. However, right click and launch using discrete gpu doesn't solve the problem for me. Is there any fix?

Seems that's related with remote desktop. When connected locally, chrome was able to use gpu accelaration even when opened directly.