Joshua-Riek/ubuntu-rockchip #862: Display not working with Ubuntu22.04-LTS

kalyanikasam, 2024-06-06:
Hi, i have tried Ubuntu22.04-LTS today followed the instructions as given in https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Ubuntu-22.04-LTS to enable 8inch radxa display, display is not working, even backlight is not there.
 path to *.dtbo file is not correct in the above wiki page, i have changed it still it is not working. Am i missing anything, with same display connections i have changed SD card that is loaded with ubuntu22.04 old version by JR, added overlay file in ubuntuEnv.txt and it is working without touch functionality.
[image]

kalyanikasam, 2024-06-06:
Reposting the u-boot file 
[image]

MyVizDrake, 2024-07-16:
@kalyanikasam try getting rid of the comment symbol on your FDT_OVERLAYS line.

mine
```
## /etc/default/u-boot - configuration file for u-boot-update(8)

#U_BOOT_UPDATE="true"

#U_BOOT_ALTERNATIVES="default recovery"
#U_BOOT_DEFAULT="l0"
#U_BOOT_PROMPT="1"
#U_BOOT_ENTRIES="all"
#U_BOOT_MENU_LABEL="Debian GNU/Linux"
#U_BOOT_PARAMETERS="ro earlycon"
#U_BOOT_ROOT=""
#U_BOOT_TIMEOUT="50"
#U_BOOT_FDT=""
#U_BOOT_FDT_DIR="/lib/firmware/"
U_BOOT_FDT_OVERLAYS="device-tree/rockchip/overlay/rock-5a-radxa-display-10fhd.dtbo"
#U_BOOT_FDT_OVERLAYS_DIR="/lib/firmware/"
#U_BOOT_SYNC_DTBS="false"
```
After saving the changes, don't forget to 'sudo u-boot-update'. I always forget that part. :-)

Hope this helps!

kalyanikasam, 2024-07-23:
@MyVizDrake Thank you for the response. But it did not work. Did it work for you?
My display is totally off even the back light is not coming. I have removed the #, saved changes and 
run sudo u-boot-update.

Joshua-Riek, 2024-07-23:
You've not said what display, device, or kernel. Please provide more information.

MyVizDrake, 2024-07-23:
@kalyanikasam yes that is my exact file for my Rock5A and Radxa Display 10FHD. The display and touch work great. Separate issue in that them both working great breaks HW acceleration …. Separate issue :-)

kalyanikasam, 2024-07-25:
@MyVizDrake I have done the same but no luck, display is not even switching on. I am using Rock5B with 8 inch Radxa Display. The same display is functional with JR ubuntu 22.04 earlier image with out touch functionality. With same hardware setup and JR ubuntu 22.04 LTS , display is not working.
In your file above, you are not proving complete file location of *.dtb file, meaning the line #U_BOOT_FDT_OVERLAYS_DIR="/lib/firmware/" is still commented. How does it work if you are not providing complete location address. I tried but didnt work that way also.
Touch functionality is a priority in my project. Hope it gets resolved asap.

Joshua-Riek, 2024-07-25:
Don't provide the full path of the drive tree overlay. That may be your problem.

Defining the full path would break whenever there is a kernel update, so logic was added to avoid this problem. A proper example of adding the overlay would be `U_BOOT_FDT_OVERLAYS="device-tree/rockchip/overlay/rock-5a-radxa-display-10fhd.dtbo`. More info is on the wiki.

I would follow the example by @MyVizDrake and make sure to run the`sudo u-boot-update` command. If this does not work, please post the contents of `/boot/extlinux/extlinux.conf`.

kalyanikasam, 2024-07-29:
That was the issue i was giving full path. Touch screen is working now with touch functionality. @MyVizDrake Is the touch functionality breaking H/w acceleration? I am trying to run obs for video streaming, both touch functionality and H/w acceleration both are important in my case. I am able to install OBS. Testing with video recording and streaming. i will update about Hardware acceleration in a while. Thanks for you guidance @Joshua-Riek and @MyVizDrake

Joshua-Riek, 2024-07-29:
I'm not entirely sure why hardware acceleration breaks here. It's a very strange problem. I've not been able to pinpoint the problem.

MyVizDrake, 2024-07-29:
@kalyanikasam it would be interesting to know if HW acceleration is broken on 22.04 as well.... my issue that Joshua is referring to is on 24.04 and I have not gone back to test if the same issue exists or not on 22.04.  I would expect the issue to remain, but .... in my case, Chromium about://gpu showed HW acceleration not available. Was also able to confirm MPV and ffmpeg were not accelerated either.

Joshua-Riek, 2024-07-29:
I recall testing Ubuntu 22.04, it should have the same issue.

Since the GPU and VPU do not work correctly, it really confuses me.

kalyanikasam, 2024-08-01:
Hi, I have tested the Ubuntu 22.04, I am attaching logs for your information. I could see malig610 as renderer with glxinfo -B, But with chrome GPU Video decode is with H/w but for Video Encode H/w acc is disabled. 
[Gpu info.odt](https://github.com/user-attachments/files/16451481/Gpu.info.odt)
[chrome gpu1.odt](https://github.com/user-attachments/files/16451483/chrome.gpu1.odt)

kalyanikasam, 2024-08-01:
I have connected 7inch Waveshare touch screen which has HDMI interface and USB-c (Touch screen side) to USB-A on Rock5B side for touch functionality. Initially the display is detected but with out touch functionality. I have added waveshare dtb file to the overlay, updated uboot after reboot system is running in rescue mode. If i still run the system in default mode by pressing ctrl+D, touch is working normally. 
[image]

closed 2024-08-06 by kalyanikasam