Joshua-Riek/ubuntu-rockchip #864: Orange Pi 5 Plus, Ubuntu 24.04 desktop, ov13855 camera doess't work

PhilZProg, 2024-06-06:
So I got official orange pi  cam adapter with a sinle camera on it which is ov13855. I've enabled orangepi-5-plus-ov13855.dtbo, installed cheese app and no device shows up there as well as in any other apps that use camera. USB camera works fine though. 
[opi5+-cam-log.log](https://github.com/user-attachments/files/15629207/opi5%2B-cam-log.log)

Joshua-Riek, 2024-06-06:
I removed the Rockchip camera engine because I did not want to distribute unknown binary files to users. I need to find the source code for the Rockchip 5.10 and 6.1 SDK releases and package it properly, however I no longer have access. Also Cheese will not work with the Rockchip cameras, gstreamer be used, however cheese can be patched iirc.

PhilZProg, 2024-06-07:
> I removed the Rockchip camera engine because I did not want to distribute unknown binary files to users.
Okay, so in order to get it working I need to install Rockchip camera engine. Is that right? Also would you mind sharing those binary files?

>  I need to find the source code for the Rockchip 5.10 and 6.1 SDK releases
Well if I will find it, I'll leave a link here. 

> Also Cheese will not work with the Rockchip cameras, gstreamer be used, however cheese can be patched iirc.
Understood, thank you.

0312birdzhang, 2024-06-12:
Can you try this? https://github.com/Joshua-Riek/ubuntu-rockchip/issues/254#issuecomment-1858937548

Joshua-Riek, 2024-06-23:
I've been working on the camera engine for the past few days and it should be working now. Please update your system with `sudo apt-get update` and install the camera engine package `sudo apt-get install camera-engine-rkaiq-rk3588`. You will then need to reboot your device.

closed 2024-06-27 by Joshua-Riek

PhilZProg, 2024-07-08:
Thank you for your amazing work on this! It works. Did you manage to find source files and compile, as you wanted?