Joshua-Riek/ubuntu-rockchip #867: firefly roc-rk3588-pc can't install ubuntu.

joehyunsuk, 2024-06-08:
I want use [ROC-RK3588S-PC] device.

I was downloaded [ubuntu-24.04-preinstalled-desktop-arm64-roc-rk3588s-pc.img.xz] file from release.

and write image to SD cards [sand disk and samsung ] with [balenaEtcher].

insert sd card and boot board but it did'nt show ubuntu install screen.

i wirte image with xz file and unzip file, but both 2 types are fail.

Joshua-Riek, 2024-06-11:
I don't have this board so there is little I can do. Can you attach a uart console and post the logs?

SharerMax, 2024-06-17:
> I want use [ROC-RK3588S-PC] device.
>
> I was downloaded [ubuntu-24.04-preinstalled-desktop-arm64-roc-rk3588s-pc.img.xz] file from release.
>
> and write image to SD cards [sand disk and samsung ] with [...]

you can try 2.2.1. may be fixed by #811

closed 2025-02-10 by github-actions[bot]