Joshua-Riek/ubuntu-rockchip #873: Not sure if booted - OPi5

cmdshft, 2024-06-11:
I have an Orange Pi 5 that I am trying to boot with this image: https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v2.2.0/ubuntu-22.04-preinstalled-desktop-arm64-orangepi-5.img.xz

When I start it up, I get two blinks of the green activity light repeatedly. I am running from the SD card (which is a class 10 as recommended) and I have an unformatted NVMe (128GB) SSD physically installed as well.

The network port blinks in an expected way for seeing data transfer and I see network activity on my gateway for the device. When I try to login to it via SSH, I get a `connection refused` error.

`cmdshft >> ssh ubuntu@10.0.0.119`
`ssh: connect to host 10.0.0.119 port 22: Connection refused`

I'm not really sure what the issue is.

Joshua-Riek, 2024-06-11:
Is the LED blinking? Have you connected HDMI? Serial console?

cmdshft, 2024-06-11:
I decided to try the 24.04 version and it seems to be working. I am completely headless, but I was able to get in via SSH now.

Joshua-Riek, 2024-06-11:
I tested the 22.04 image and it works fine for me, maybe you had a bad install or did not wait long enough for cloud-init to configure.

closed 2024-06-11 by Joshua-Riek

marijnbent, 2024-06-20:
> I decided to try the 24.04 version and it seems to be working. I am completely headless, but I was able to get in via SSH now.

how do you connect to headless to the ssh?

o6uoq, 2025-06-25:
@marijnbent if you're using Turing Pi 2, simply SSH to the IP of the node you've flashed and turned on.