Joshua-Riek/ubuntu-rockchip #874: Radxa rock 5 itx not recognising SATA HDD after upgrade to kernel 6.1.0-1016

stevepringle, 2024-06-12:
My Radxa rock 5 itx is not recognising my 2TB SATA HDD after an upgrade to kernel 6.1.0-1016. Reverting back to the 6.1.0-1014 kernel fixes the problem.

stevepringle, 2024-06-13:
> That is very strange, i don't see any changes made that should affect SATA for the Rock 5 ITX. I know there is some talk about this issue in the Radxa Discord, users report that when using the Rockchip 6.1 kernel, SATA drives can sometimes not work. However it seems the Rockchip 5.10 kernel works just fine, so I would recommend using Ubuntu 22.04 for now until this problem is resolved.
> Kernel changelog:

Thanks Joshua,I think for now I’ll probably stick with the 6.1 kernel version that seems to work OK. I wonder if it’s a power supply issue to the disk as I’m using the one of the 4 pin power connectors on the motherboard to power the hard disk, perhaps it can’t supply enough current. I might try connecting the disk straight to the ATX power supply instead.Regards,SteveSent from my iPad

stevepringle, 2024-06-18:
Hi Joshua,In the end, I did as you suggested and installed 22.04 which seems to have fixed the problem. Many thanks for your help and advice and keep up the good work, it’s much appreciated :-)Regards,Steve PringleSent from my iPad

vishap, 2025-01-06:
I still have this issue.
```
Linux rock-5-itx 6.1.75-vendor-rk35xx #1 SMP Tue Nov 12 08:48:32 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux
```

ypopovych, 2025-02-20:
I think it could be fixed by this: https://lore.kernel.org/all/20240906082511.2963890-6-heiko@sntech.de/

This patch already accepted into mainline kernel v6.13.

muravjov, 2025-05-20:
@Joshua-Riek's kernels are just great (I use them for three rk3588 boards, including radxa 5 itx)

But when I started to test my SATA disks, I got issues.
**tldr;** both kernels from the images, 6.1.0 (native for 24.04) and 5.10.0 (native for 22.04), don't work properly.
I've built latest stable kernel, v6.14.5, not vendor one, and it works without issues.

My tests:
- kernel 6.1.0 (native for 24.04) works worse than 5.10.0 (native for 22.04)
- both used to fail:
  - at PCI stage - 0001:11 is not recongized at all or has class -1 = 0xffffff (PCI stage)
    pci 0001:11:00.0: [1b21:1164] type 00 class 0xffffff
    if fails here => do not see further

    ```bash
    $ ssh sa@172.30.5.143 dmesg | grep -i 'PCI.*0xffffff'
    [   35.616854] pci 0001:11:00.0: [1b21:1164] type 00 class 0xffffff
    $ lspci -nn
    ...
    0001:11:00.0 Unassigned class [ffff]: ASMedia Technology Inc. Device [1b21:1164] (rev 02)
    ...
    ```

  - AHCI errors like
      ahci 0001:11:00.0: controller reset failed (0xffffffff)
      or
      ahci can't change power state from D3cold to D0 (config space inaccessible)
      or
      ahci: probe of 0000:01:00.0 failed with error -5
    means disks had enough time to init after powering up, so not ready
    and kernel decided they are broken
    Again, if fails here => do not see further (ahci subsystem is not inited properly)
    Workaround:
    - hot plug: connect power cable FIRST, then data cable
    - before booting: make a timeout for u-boot menu before going for a default
      menu item (what kernel to boot)

@ypopovych 
Btw, radxa original b3 works as expected, no issues, disks can be attached before boot or hot plug.
Its kernel is a vendor one, based on 6.1, and so doesn't contain that patch (and still works). 

I built latest stable kernel (at the time of building) with armbian [build](https://github.com/armbian/build):
```bash
$ ./compile.sh BOARD=rock-5-itx \
  BRANCH=edge KERNEL_GIT=shallow \
  KERNELBRANCH="tag:v6.14.5" \
  BUILD_DESKTOP=no BUILD_MINIMAL=yes RELEASE=noble \
  kernel
```
debs may be found in ./output direcory.

Along installing those debs into a working ubuntu, you need to adjust fdtdir like

```
$ sudo ln -s /usr/lib/linux-image-6.14.5-edge-rockchip64 /lib/firmware/6.14.5-edge-rockchip64

$ ls -lah linux-image-edge-rockchip64_25.05.0-trunk_arm64__6.14.5-Saeae-Db228-P4b40-Ce03dH3d80-HK01ba-Vc222-B9bbb-R448a.deb
$ sudo dpkg -i linux-image-edge-rockchip64_25.05.0-trunk_arm64__6.14.5-Saeae-Db228-P4b40-Ce03dH3d80-HK01ba-Vc222-B9bbb-R448a.deb

# good, kernel added to the boot menu properly
$ grep fdt /boot/extlinux/extlinux.conf
	fdtdir /lib/firmware/6.14.5-edge-rockchip64/
	fdtdir /lib/firmware/6.14.5-edge-rockchip64/
	fdtdir /lib/firmware/6.1.0-1025-rockchip/device-tree/
	fdtdir /lib/firmware/6.1.0-1025-rockchip/device-tree/
	fdtdir /lib/firmware/5.10.0-1012-rockchip/device-tree/
	fdtdir /lib/firmware/5.10.0-1012-rockchip/device-tree/

# adjust default to boot, if needed
$ sudo vim /etc/default/u-boot
# regen the menu again
$ sudo u-boot-update

# reboot and test the new kernel
$ sudo shutdown now
```