Joshua-Riek/ubuntu-rockchip #878: [RK3566] Correct DT Overlay for PWM in Radxa 3W

eshcheglov, 2024-06-14:
Hello, Mr. Joshua,

I have been trying to understand how the Device Tree Overlay for PWM works on the Radxa 3W board in recent days.

I know it works correctly in the official Radxa distribution called [radxa-zero3_debian_bullseye_xfce_b6.img.xz](https://github.com/radxa-build/radxa-zero3/releases). For this, I simply need to enable the DT Overlay named `rk3568-pwm9-m0.dtbo` and a directory named `pwmchip9` appears in `/sys/class/pwm`.

I see that in your distribution, all PWM DTO's for the rk3568 chip are disabled (just not present) 
(I checked here: https://github.com/Joshua-Riek/linux-rockchip/blob/noble/arch/arm64/boot/dts/rockchip/overlay/Makefile).

I have tried the following approaches:

1) I took the [official DTS from Radxa](https://github.com/radxa-pkg/radxa-overlays), compiled it into .dtbo, and enabled it. 
As a result, the PWM works, but it has an incorrect name (`pwmchip0` instead of `pwmchip9`). If I enable `PWM9` and `PWM8` simultaneously, they appear with incorrect names `pwmchip0` and `pwmchip1`, respectively. 
However, the PWM functions themselves work correctly.

2) I tried taking the `.dtbo` from the Radxa distribution and connecting it to your distribution. The result is the same.

3) I tried decompiling the DTBO into DTS in the Radxa distribution and then compiling it back from that DTS into DTBO in your distribution. The result is the same.

I also noticed that in all cases, the binary diff does not see any differences between these DTBOs. So, I can assume that the problem lies elsewhere.

I would like to make a merge request to your kernel that adds all PWMs for the Radxa 3W, but until I resolve the issue with incorrect naming, I cannot do so.

Do you have any ideas about what might be causing the incorrect directory naming?

Thank you.

eshcheglov, 2024-06-14:
**Steps to reproduce:**

- Install fresh `24.04` distro
- Check that pwm is emtpy:
```bash
# ls /sys/class/pwm/
```
- Clone and compile Radxa DT Overlays:
```bash
git clone https://github.com/radxa-pkg/radxa-overlays.git
cd radxa-overlays
make build-dtbo -j4
```

- Install compiled DTBO for PWM9:

```bash
cp ./arch/arm64/boot/dts/rockchip/overlays/rk3568-pwm9-m0.dtbo /usr/lib/firmware/6.1.0-1016-rockchip/device-tree/rockchip/overlay/
```

- Add overlay to U-boot: put line `U_BOOT_FDT_OVERLAYS="device-tree/rockchip/overlay/rk3568-pwm9-m0.dtbo"` to `/etc/default/u-boot`

```
u-boot-update
```

Check new config: 
```bash
$  grep pwm9 /boot/extlinux/extlinux.conf
	fdtoverlays  /lib/firmware/6.1.0-1016-rockchip/device-tree/rockchip/overlay/rk3568-pwm9-m0.dtbo
```

- Reboot device

- Check again the PWM's:

```
$ ls /sys/class/pwm/
pwmchip0
```

directory is named `pwmchip0`, but it must be named `pwmchip9`

```
$ uname -a
Linux ubuntu 6.1.0-1016-rockchip #16-Ubuntu SMP Thu Jun  6 00:11:54 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux

```

RadxaStephen, 2024-06-14:
I think the aliases name is needed in device tree.

Like this.

https://github.com/radxa/kernel/blob/linux-5.10-gen-rkr4.1/arch/arm64/boot/dts/rockchip/rk3568.dtsi#L57

eshcheglov, 2024-06-17:
> I think the aliases name is needed in device tree.
> Like this. https://github.com/radxa/kernel/blob/linux-5.10-gen-rkr4.1/arch/arm64/boot/dts/rockchip/rk3568.dtsi#L57

The [corresponding file](https://github.com/Joshua-Riek/linux-rockchip/blob/noble/arch/arm64/boot/dts/rockchip/rk3568.dtsi) in the linux-rockchip kernel by Joshua looks pretty limited (I'd rather say "cut down"). I think it was done for some reasons. Maybe because of the lack of QA possibilities for Radxa Rock3 boards?..

Okay, I'll test it, thanks!

eshcheglov, 2024-06-17:
No result still.

Tried this:

```bash
git clone --depth 1 https://github.com/Joshua-Riek/linux-rockchip.git
cd linux-rockchip/arch/arm64/boot/dts/rockchip
vim rk356x.dtsi  
```

Lines added to `aliases` section:
```
<...>
                spi3 = &spi3;
                pwm0 = &pwm0;
                pwm1 = &pwm1;
                pwm2 = &pwm2;
                pwm3 = &pwm3;
                pwm4 = &pwm4;
                pwm5 = &pwm5;
                pwm6 = &pwm6;
                pwm7 = &pwm7;
                pwm8 = &pwm8;
                pwm9 = &pwm9;
                pwm10 = &pwm10;
                pwm11 = &pwm11;
                pwm12 = &pwm12;
                pwm13 = &pwm13;
                pwm14 = &pwm14;
                pwm15 = &pwm15;
        };
<...>
```

```
cpp -I ~/linux-rockchip/include/ -nostdinc -undef -x assembler-with-cpp rk3566-radxa-zero3.dts rk3566-radxa-zero3.dts.preprocessed
dtc -@ -I dts rk3566-radxa-zero3.dts.preprocessed -O dtb -o /tmp/rockchip.dtb
cp /tmp/rockchip.dtb /usr/lib/firmware/6.1.0-1016-rockchip/device-tree/rockchip/rk3566-radxa-zero3.dtb
reboot
```

I saw my changes in the `rk3566-radxa-zero3.dts.preprocessed` file, so I assume that I changed the correct line in the correct file

closed 2024-12-16 by github-actions[bot]

ioncubekh, 2025-03-23:
@eshcheglov @RadxaStephen Any update on this issue. I am also trying to activate PWM with no success. Details here

> https://forum.radxa.com/t/zero-3e-pwm-gpio-under-kernel-6-1-ubuntu/26423

eshcheglov, 2025-04-12:
Hi, @ioncubekh 

I was able to activate the PWM, as described in my opening message.
The PWM works well, but the issue with numbering is still relevant.

Feel free to contact me via email if you need help with this topic: eugenyshcheglov [at] gmail [dot] com."