Joshua-Riek/ubuntu-rockchip #879: Add OpenCL support to the images

HeyMeco, 2024-06-14:
For the current Noble based images with 6.1 Kernel we need Mali DDK g18p0-01eac0. (5.10 should support the latest DDK)

The fix I did was to install this releases deb since it does all the configuration: https://github.com/tsukumijima/libmali-rockchip/releases/tag/v1.9-1-2d267b0 

[libmali-valhall-g610-g13p0-x11-wayland-gbm_1.9-1_arm64.deb](https://github.com/tsukumijima/libmali-rockchip/releases/download/v1.9-1-2d267b0/libmali-valhall-g610-g13p0-x11-wayland-gbm_1.9-1_arm64.deb)

---

We already have the mali firmware included but in the future it would be better to also have the OpenCL ICD loader installed and libmali added to ICD during the build process so it works out of the box.

---

To get the right driver DDK version, such as "g21p0-01eac0" you have to look in the file kernel/drivers/gpu/arm/bifrost/Kbuild:
"MALI_RELEASE_NAME ?= '"g21p0-01eac0"'".

Context: This is needed for HDR Tone-Mapping in upstream Jellyfin which supports RK3588 VPU HW acceleration

Joshua-Riek, 2024-06-17:
g21p0-01eac0 does not support wayland so I reverted support for this blob in the kernel, so g18p0-01eac0 is included by default. I'm confused on what the problem is and what exactly is being requested to change.

HeyMeco, 2024-06-17:
@Joshua-Riek the firmware for g18p0… is included yes but for example if you install clinfo you can see that the OpenCL aspect is missing in the image.

HeyMeco, 2024-06-17:
Maybe this page with the section for Orange Pi 5 (but applies to all) does a better job explaining what needs to be done. Besides having the driver/firmware in the image. https://llm.mlc.ai/docs/install/gpu.html#orange-pi-5-rk3588-based-sbc

Joshua-Riek, 2024-06-18:
I see, it's been a while since I messed with the Mali blobs, but i worry installing this may break hardware acceleration with panfork. This will need some testing then it must be built and packaged on Launchpad, i have the Mali blob managed by another package so this will need some hacking.

HeyMeco, 2024-06-18:
opencl & the deb do not interfere with panfork as the solution I went with for example is just a mirror of jeffycn’s rockchip mali blob and only builds  a „one click install for it“. It’s action is also open source so maybe it would make sense to take it to launchpad and just use the deb from before january for anything that needs g18 and other kernels can use the most recent one

Joshua-Riek, 2024-06-18:
Got it, from taking a quick look it should be easy to add to Launchpad, i'm testing the build process now. But I will need to add a hack to remove the **mali_csffw.bin** blob as it's handled by another package.

Joshua-Riek, 2024-06-18:
Oh yikes, the repo is over 8GB, this will cause problems with Launchpad as the limit is 8GB.

HeyMeco, 2024-06-18:
> Oh yikes, the repo is over 8GB, this will cause problems with Launchpad as the limit is 8GB.

Why do you not just make the build process do:
Download and install libmali-g610.so
``` 
cd /usr/lib && sudo wget https://github.com/JeffyCN/mirrors/raw/libmali/lib/aarch64-linux-gnu/libmali-valhall-g610-g6p0-x11-wayland-gbm.so
```

(Check if file mali_csffw.bin exist under path /lib/firmware, if not download it with command:)

```
cd /lib/firmware && sudo wget https://github.com/JeffyCN/mirrors/raw/libmali/firmware/g610/mali_csffw.bin
```

Download OpenCL ICD loader and manually add libmali to ICD
```
sudo apt update
sudo apt install mesa-opencl-icd
sudo mkdir -p /etc/OpenCL/vendors
echo "/usr/lib/libmali-valhall-g610-g6p0-x11-wayland-gbm.so" | sudo tee /etc/OpenCL/vendors/mali.icd
```
Download and install libOpenCL
`sudo apt install ocl-icd-opencl-dev`
Download and install dependencies for Mali OpenCL
```
sudo apt install libxcb-dri2-0 libxcb-dri3-0 libwayland-client0 libwayland-server0 libx11-xcb1 
```
Download and install clinfo to check if OpenCL successfully installed
`sudo apt install clinfo`

since you already have the firmware and do not need the whole repo to add OpenCL

Joshua-Riek, 2024-06-18:
This is insightful, thank you very much. I can create a minimal package with the `libmali-valhall-g610-g6p0-x11-wayland-gbm.so` file and add the path to `/etc/OpenCL/vendors/mali.icd` when the package is installed.

Joshua-Riek, 2024-06-18:
I will likely add this to the mali-g610-firmware package on Launchpad.
https://launchpad.net/~jjriek/+archive/ubuntu/panfork-mesa/+packages

Joshua-Riek, 2024-06-18:
I see `libmali-valhall-g610-g6p0-x11-wayland-gbm.so` and `libmali-valhall-g610-g13p0-x11-wayland-gbm.so` any idea which one works better / is newer?

HeyMeco, 2024-06-18:
I only know g13p0 is the right one in this case. The versioning scheme is… something.

Joshua-Riek, 2024-06-18:
It looks like I don't need to package anything, it already exists, simply download the `libmali-g610-x11` and OpenCL should be working.

HeyMeco, 2024-06-18:
For testing you can use `geekbench6 --gpu ` which does OpenCL RK3588 should score 3500 +-100. Example: https://browser.geekbench.com/v6/compute/2329271

HeyMeco, 2024-06-18:
@Joshua-Riek the issue is that there is two different DDK versions and the ones since 2024 are based on version 21 and are NOT compatible with g18 which is used by Kernel 6.1. /  5.10 can use the most recent

Joshua-Riek, 2024-06-18:
Got it, the libmali-g610-x11 package has an older blob that seems to be compatible with g18. I'm running the geekbench test now to confirm it's working as expected.

Joshua-Riek, 2024-06-18:
Yup, working as expected, see the results below.
https://browser.geekbench.com/v6/compute/2349029

I will include these blobs by default for future builds as implemented in https://github.com/Joshua-Riek/ubuntu-rockchip/commit/b8be8d65220469a0285a245dfd65191fc2edee3c so OpenCL will work out of the box.

HeyMeco, 2024-06-18:
I think I even found **the** maybe just **a** difference between g6p0 and g13p0. One is OpenCL 2.1 the other 3.0

Joshua-Riek, 2024-06-19:
Good find, i would assume OpenCL 3.0 is newer, so using the g13p0 blob would be best.

Joshua-Riek, 2024-06-19:
With the latest nightly build OpenCL should be working out of the box.

https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/9573838480

I'm testing the images now to confirm.

KylinDemons, 2024-06-19:
May I ask, does this nightly build which you mean that opencl included support for noble kernel 6.1 version？

Joshua-Riek, 2024-06-19:
Yup, if you have a currently installed system you can simply enter `sudo apt-get install libmali-g610-x11` to install the mesa blobs required for OpenCL. 

As for testing, I can confirm the latest nightly builds of Ubuntu 22.04 and 24.04. Both now work with OpenCL out of the box.

KylinDemons, 2024-06-19:
> Yup, if you have a currently installed system you can simply enter `sudo apt-get install libmali-g610-x11` to install the mesa blobs required for OpenCL. 
> 
> As for testing, I can confirm the latest nightly builds of Ubuntu 22.04 and 24.04. Both now work with OpenCL out of the box.

Thank you. I appreciate your work！

closed 2024-06-19 by Joshua-Riek

KylinDemons, 2024-07-06:
@HeyMeco Do you get jellyfin HDR-tonemap worked? I installed the last nightly build, and checked the output of the clinfo, all seems right, but when I play a HDR video through jellyfin 10.9.7, the color isn't right.

HeyMeco, 2024-07-06:
> @HeyMeco Do you get jellyfin HDR-tonemap worked? I installed the last nightly build, and checked the output of the clinfo, all seems right, but when I play a HDR video through jellyfin 10.9.7, the color isn't right. 

@KylinDemons did you enable hdr tonemappig in the transcoding settings?

KylinDemons, 2024-07-06:
@HeyMeco Of course I did, I tested a HDR10 video, tonemapping worked fine, but DoVi videos failed, it's wired.

HeyMeco, 2024-07-06:
@KylinDemons 
> DoVi videos failed

P5 doesn't work yet (if ever). P8 or P7 (any that fallback to HDR10) work.

KylinDemons, 2024-07-06:
> @KylinDemons
> 
> > DoVi videos failed
> 
> P5 doesn't work yet (if ever). P8 or P7 (any that fallback to HDR10) work.

Thanks a lot!! I checked the media info, it is indeed DV Profile 5. Now I think I should directly play it through potplayer instead of jellyfin.

AkashRajpurohit, 2024-07-09:
Hey @KylinDemons Can you share how did you get the HDR tone mapping to work correctly? I am on Ubuntu Rockchip `24.04` kernel `6.1.0-1019-rockchip` and clinfo does show that it seems to be working fine but when I test with the command mentioned in the jellyfin doc: `sudo docker exec -it jellyfin /usr/lib/jellyfin-ffmpeg/ffmpeg -v debug -init_hw_device rkmpp=rk -init_hw_device opencl=ocl@rk` I see this error:

```
ffmpeg version 6.0.1-Jellyfin Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 12 (Debian 12.2.0-14)
  configuration: --prefix=/usr/lib/jellyfin-ffmpeg --target-os=linux --extra-version=Jellyfin --disable-doc --disable-ffplay --disable-ptx-compression --disable-static --disable-libxcb --disable-sdl2 --disable-xlib --enable-lto --enable-gpl --enable-version3 --enable-shared --enable-gmp --enable-gnutls --enable-chromaprint --enable-opencl --enable-libdrm --enable-libass --enable-libfreetype --enable-libfribidi --enable-libfontconfig --enable-libbluray --enable-libmp3lame --enable-libopus --enable-libtheora --enable-libvorbis --enable-libopenmpt --enable-libdav1d --enable-libsvtav1 --enable-libwebp --enable-libvpx --enable-libx264 --enable-libx265 --enable-libzvbi --enable-libzimg --enable-libfdk-aac --arch=arm64 --cross-prefix=/usr/bin/aarch64-linux-gnu- --toolchain=hardened --enable-cross-compile --enable-rkmpp --enable-rkrga
  libavutil      58.  2.100 / 58.  2.100
  libavcodec     60.  3.100 / 60.  3.100
  libavformat    60.  3.100 / 60.  3.100
  libavdevice    60.  1.100 / 60.  1.100
  libavfilter     9.  3.100 /  9.  3.100
  libswscale      7.  1.100 /  7.  1.100
  libswresample   4. 10.100 /  4. 10.100
  libpostproc    57.  1.100 / 57.  1.100
Splitting the commandline.
Reading option '-v' ... matched as option 'v' (set logging level) with argument 'debug'.
Reading option '-init_hw_device' ... matched as option 'init_hw_device' (initialise hardware device) with argument 'rkmpp=rk'.
Reading option '-init_hw_device' ... matched as option 'init_hw_device' (initialise hardware device) with argument 'opencl=ocl@rk'.
Finished splitting the commandline.
Parsing a group of options: global .
Applying option v (set logging level) with argument debug.
Applying option init_hw_device (initialise hardware device) with argument rkmpp=rk.
Applying option init_hw_device (initialise hardware device) with argument opencl=ocl@rk.
[AVHWDeviceContext @ 0xaaaac8ea2f00] Failed to get number of OpenCL platforms: -1001.
Device creation failed: -19.
Failed to set value 'opencl=ocl@rk' for option 'init_hw_device': No such device
Error parsing global options: No such device
```

It used to work fine before when I was on Orange Pi version of Ubuntu 22.04 and now currently trying to figure out what could be possibly going wrong with Ubuntu Rockchip 24.04

KylinDemons, 2024-07-09:
First make sure your clinfo output on the host is fine, if not, try install the lastest nightly build in Github action or just run:
```
sudo apt-get install libmali-g610-x11
```
second, make sure you follow the [jellyfin documentation](https://jellyfin.org/docs/general/administration/hardware-acceleration/rockchip#configure-with-linux-virtualization) to install the right container. If still not work, try to follow [this](https://github.com/jellyfin/jellyfin/issues/12174#issuecomment-2190274438) to run container as root.