Joshua-Riek/ubuntu-rockchip #880: How to enable Ralink RT5370 wireless adapter in rt2800usb kernel module?

alextrezvy, 2024-06-14:
This adapter in Orange Pi 5 was disabled for whatever reasons:
`modinfo rt2800usb | grep '5370'`

vim /boot/config-5.10.160-rockchip
# CONFIG_RT2800USB_RT53XX is not set

Hence, suddenly I can't use my adapter that worked for 10+ years.
It should be possible to compile "rt2800usb" module from sources, but I don't know how to do it without compiling the whole kernel.
And compiling the whole kernel seems overkill and errorprone.
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.4 LTS
Release:	22.04
Codename:	jammy
```

alextrezvy, 2024-06-15:
Same issue in 24.04.

Joshua-Riek, 2024-06-15:
The kernel comes from an Android source I could not mass enable wireless drivers due to a large amount of kernel errors during compiling. So I need to manually enable wireless drivers as problems come up or per user request. However recompiling the whole kernel is required.

Id be happy to enable this driver, ill get back to you when it's enabled.

Joshua-Riek, 2024-06-21:
This has been enabled in both kernels, I will update you when the kernel packages are upgraded this weekend.
https://github.com/Joshua-Riek/linux-rockchip/commit/41576ace65693da86345159bc658b4b5de52aa58
https://github.com/Joshua-Riek/linux-rockchip/commit/5be2a12f42e589cbdfd9da079ca1a7c33684c202

Joshua-Riek, 2024-06-24:
Update your system with apt and reboot, the driver has been added and kernel package uploaded.

closed 2024-06-24 by Joshua-Riek

alextrezvy, 2024-06-24:
It works! Thank you very much!
PS: wish to make a small donation, but don't know how to do it due to sanctions. Do you accept litecoin or zcash?

Joshua-Riek, 2024-06-25:
Much appreciated, currently I only accept donations through GitHub itself.

https://github.com/sponsors/Joshua-Riek