Joshua-Riek/ubuntu-rockchip #883: Request support LubanCat RK3566

AlexxIT, 2024-06-18:
Hi. This project already support LubanCat 4.

I have Thinkcore Pi TP-0W / LubanCat Zero W / RK3566.
I don't know if it difficult for you to add this support. Because images from LubanCat awful.

Some background: https://github.com/AlexxIT/go2rtc/issues/979#issuecomment-2171040892

Joshua-Riek, 2024-06-20:
Most of these boards have very bad software support so I need to dig through the poorly written code and port the support to my kernels. The RK3566 / RK3568 chip-set is not very good in my opinion due to it's lack of processing power, so I've been paying little attention. 

While I'd be happy to accept any pull request to add support for this device, I likely will not buy the device myself to add support for it.

AlexxIT, 2024-06-20:
Home Assistant developers chosen RK3566 as a base home server board. So I thought it would be good test board for me.
https://www.home-assistant.io/green/ 

But choosing a board from LubanCat was of course a big mistake.

closed 2024-11-29 by AlexxIT