Joshua-Riek/ubuntu-rockchip #884: problem about self chroot caused mpp_service missing

czyt, 2024-06-21:
hello,thanks for this great project.I have made my own chroot to setup the build,everything works as expect,but the mpp_service always failed:
```
mpp[8312]: vcodec_service: open vcodec_service /dev/mpp_service failed
mpp[8312]: hal_h264e_vepu541: hal_h264e_vepu541_init mpp_dev_init failed. ret: -1
```
I have tried in two way

1. build the project .include kernel\rootfs\image.failed.
2. using assets in this repo.and chroot in x86_64.the mpp_service  also missing.

what i have done in chroot is disable oem-config\set chinese language pack\add ibus and install extra packages.and i have tried not disable oem-config step\not set locale,the mpp_service is missing also.so could you give me some advice about the mpp_service ,which step i should look at.thank you.

Joshua-Riek, 2024-06-23:
I'm confused on this issue.. Please post a link to the exact changes you made to the source code.

czyt, 2024-06-24:
hello,thanks for reply.the changes can be found inf the repo.https://github.com/holomotion/factory-images

Joshua-Riek, 2024-06-24:
Hmm there are a lot of changes. What application are you using that does not work with MPP? Does 4k video playback work in Chrome?

czyt, 2024-06-25:
after not install ffmpeg package,the /dev/mpp_service became avaliable,but still not working

Joshua-Riek, 2024-06-25:
Sorry ive not had time to look into this yet, i suspect some of the packages you have installed cause a conflict.

czyt, 2024-06-25:
thanks anyway for your reply,the detail log:
```
mpp[2545]: mpi: mpp_create enter ctx 0xaaab1e749bc8 mpi 0xaaab1e749bd0
mpp[2545]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[2545]: mpi: mpp_create leave ret 0 ctx 0xaaab1e748840 mpi 0xffffaf5303b0
mpp[2545]: mpi: mpi_control enter ctx 0xaaab1e748840 cmd 200006 parm 0xffffd90025e4
mpp[2545]: mpi: mpi_control leave ctx 0xaaab1e748840 ret 0
mpp[2545]: mpi: mpi_control enter ctx 0xaaab1e748840 cmd 200007 parm 0xffffd90025e4
mpp[2545]: mpi: mpi_control leave ctx 0xaaab1e748840 ret 0
mpp[2545]: mpi: mpp_init enter ctx 0xaaab1e748840 type 1 coding 7
mpp[2545]: vcodec_service: open vcodec_service /dev/mpp_service failed
mpp[2545]: hal_h264e_vepu541: hal_h264e_vepu541_init mpp_dev_init failed. ret: -1
mpp[2545]: mpp_enc_hal: mpp_enc_hal_init hal hal_h264e init failed ret -1
mpp[2545]: mpp_enc_hal: mpp_enc_hal_init could not found coding type 7
mpp[2545]: mpp_enc: mpp_enc_init_v2 could not init enc hal
mpp[2545]: mpp: error found on mpp initialization
mpp[2545]: mpi: mpp_init leave ctx 0xaaab1e748840 ret -1
mpp_init-1
```

czyt, 2024-06-25:
the issue was caused by the /dev/mpp_service permit,after chown to the new user,all worked

closed 2024-06-25 by czyt