Joshua-Riek/ubuntu-rockchip #885: USB 3.0 port on orange pi 5 Pro has no power

fllay, 2024-06-21:
I installed both 22.04 and 24.04 for orange pi 5 Pro. However, there is no power for a device connected to USB 3.0 port. Anyone experience this? I tried official Ubuntu from Orange pi web and USB 3.0 port delivers power just fine.

Joshua-Riek, 2024-06-23:
Fixed with https://github.com/Joshua-Riek/linux-rockchip/commit/5ac9ae7ca056673d05f5b0543012991a3e3f5ebd, update will go out later today.

closed 2024-06-23 by Joshua-Riek

marosrr, 2024-07-23:
Hi,

I tried same with latest version 24.04 and USB 3.0 is still not working on Orangepi 5 Pro, any ideas?

Thank you

lsusb result
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 003: ID 4037:2804  2.4G Composite Devic
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

I also tried to change device tree to OP5 pro, but not helped
fdt /lib/firmware/6.1.0-1016-rockchip/device-tree/rockchip/rk3588s-orangepi-5-pro.dtb