Joshua-Riek/ubuntu-rockchip #886: radxa zero 3w periodic wlan termination / interruption with system freeze

Manu7788, 2024-06-22:
fresh image of either 22 or 24 ubuntu with a wifi connection (managed by either system.networkd or network-manager) will have periodic wifi disconnects on radxa zero 3w. the network manager loads the gmac driver periodically, which fails because there is no external phy on radxa zero 3w. if you use apt-get update while this bug is active the system hangs and freezes randomly.

Manu7788, 2024-06-22:
i used `fping -l -p 100 fritz.box` for testing

Joshua-Riek, 2024-06-23:
How much RAM does your system have? If it's less than 4GB the desktop version will run out of memory causing extreme instability.

Manu7788, 2024-06-23:
> How much RAM does your system have? If it's less than 4GB the desktop version will run out of memory causing extreme instability.

Its the same behaviour on all devices i tested. doesnt relate to desktop nor server version. where in the overlays i am able to disable the gmac driver for testing?

Manu7788, 2024-06-23:
i just disabled the ethernet interface with overlay radxa-zero3-disabled-ethernet.dtbo but still the same. my written c program `reported CLIENT: ERROR recv(): Resource temporarily unavailable` so the sockets are hanging too. might be a driver problem.

Joshua-Riek, 2024-06-23:
I see Radxa has updated the AIC8800 wireless driver, there is a chance this problem has been addressed. I can try to update the driver later today.

https://github.com/Joshua-Riek/aic8800/compare/main...radxa-pkg%3Aaic8800%3Amain

Manu7788, 2024-06-23:
let me know when you have updated. i can build and test the new image tomorrow.

dfloer, 2024-06-26:
I found my Zero 3W was extremely picky about its orientation, what was close to it, etc, with WiFi. About 2.5m away from an AP, sitting on a desk, performance was good. Move it 1 half meter up onto a metal shelf? Barely able to sustain 4Mb/s. Back onto the desk, speeds got better.

Not sure if this will apply to yours, but something to look out for.

closed 2025-02-10 by github-actions[bot]