Joshua-Riek/ubuntu-rockchip #888: share a wayland vulkan mali driver [maybe official]

zylo117, 2024-06-25:
share a  maybe-official wayland vulkan driver.

just `cat x** > rockchip-mali_1.9-12_arm64.deb`

I've try this on nano r6s with official mali driver on wayland ubuntu 22.04 and succeed to build and run ncnn with vulkan, by manually copying the wayland-vulkan related *.so to lib path. But vulkan won't work in x11 desktop.

Can you please try this on your 3588 devices to see if they are working? If so, it would be nice that you can integrate it into ubuntu-rockchip. This project is really awesome.

#617 

[xaa.zip](https://github.com/user-attachments/files/15973383/xaa.zip)
[xab.zip](https://github.com/user-attachments/files/15973389/xab.zip)
[xac.zip](https://github.com/user-attachments/files/15973392/xac.zip)

Joshua-Riek, 2024-06-25:
Where did these files come from? Are they directly from Rockchip?

Uploading random binaries seems suspicious to me.

ginkage, 2024-06-25:
OK, that's interesting!
```
ginkage@rock-5b:~/libmali-vk/usr/lib/aarch64-linux-gnu$ vulkaninfo
arm_release_ver of this libmali is 'g6p0-01eac0', rk_so_ver is '10'.
'DISPLAY' environment variable not set... skipping surface info
arm_release_ver of this libmali is 'g6p0-01eac0', rk_so_ver is '10'.
==========
VULKANINFO
==========

Vulkan Instance Version: 1.3.204
Instance Extensions: count = 10
===============================
        VK_EXT_debug_report                    : extension revision 10
        VK_EXT_debug_utils                     : extension revision 2
        VK_EXT_headless_surface                : extension revision 1
        VK_KHR_device_group_creation           : extension revision 1
        VK_KHR_display                         : extension revision 23
        VK_KHR_external_fence_capabilities     : extension revision 1
        VK_KHR_external_memory_capabilities    : extension revision 1
        VK_KHR_external_semaphore_capabilities : extension revision 1
        VK_KHR_get_physical_device_properties2 : extension revision 2
        VK_KHR_surface                         : extension revision 25

Layers: count = 2
=================
VK_LAYER_MESA_device_select (Linux device selection layer) Vulkan version 1.3.211, layer version 1:
        Layer Extensions: count = 0
        Devices: count = 1
                GPU id = 0 (Mali-LODX)
                Layer-Device Extensions: count = 0

VK_LAYER_MESA_overlay (Mesa Overlay layer) Vulkan version 1.3.211, layer version 1:
        Layer Extensions: count = 0
        Devices: count = 1
                GPU id = 0 (Mali-LODX)
                Layer-Device Extensions: count = 0
Device Groups:
==============
Group 0:
        Properties:
                physicalDevices: count = 1
                        Mali-LODX (ID: 0)
                subsetAllocation = 0

arm_release_ver of this libmali is 'g6p0-01eac0', rk_so_ver is '10'.
        Present Capabilities:
                Mali-LODX (ID: 0):
                        Can present images from the following devices: count = 1
                                Mali-LODX (ID: 0)
                Present modes: count = 1
                        DEVICE_GROUP_PRESENT_MODE_LOCAL_BIT_KHR
```
Of course it wouldn't work in X11, as it's a Wayland .so...

bagong, 2024-06-25:
Is it a closed source driver? How come the origin isn't revealed? Joshua couldn't distribute an illegal, possibly even malicious driver...

ginkage, 2024-06-25:
It seems to lack `VK_KHR_wayland_surface`, so most likely it's a headless-only driver.

ginkage, 2024-06-25:
Looks like this deb is also available to download here: https://repo.rock-chips.com/edge/debian-release-v2.0.0/pool/main/r/rockchip-mali/

Joshua-Riek, 2024-06-25:
Good find, I just wanted to make sure this is public in some form. Splitting the package into 3 parts was strange, but this seems legitimate.

zylo117, 2024-06-26:
> Good find, I just wanted to make sure this is public in some form. Splitting the package into 3 parts was strange, but this seems legitimate.

> Looks like this deb is also available to download here: https://repo.rock-chips.com/edge/debian-release-v2.0.0/pool/main/r/rockchip-mali/

Yes, this is the download link

zylo117, 2024-06-27:
@Joshua-Riek Is it possible to add vulkan support to ubuntu-rockchip?

Joshua-Riek, 2024-06-28:
I think it may be to early to add these blobs. Does anyone know if it's usable in any form or if there is a compatible drivers for this new vulkan blob?

zylo117, 2024-06-28:
> I think it may be to early to add these blobs. Does anyone know if it's usable in any form or if there is a compatible drivers for this new vulkan blob?

Of course, it's very useful in chromium. When you open chrome, it it says vulkan error. If there is a vk driver,  it can accelerate the browser.
And vulkan is very important for gaming and ai developing in linux.

ginkage, 2024-06-28:
The problem is, this blob is useless for actually accelerating graphics, as it lacks the necessary extension `VK_KHR_wayland_surface`.
It may be (probably) used for things like cnn, but this particular blob can not be used for games or browser, I'm afraid.

ginkage, 2024-06-28:
As in:
```
ginkage@rock-5c:~/Vulkan-Tools/build/cube$ ./vkcube-wayland 
vkEnumerateInstanceExtensionProperties failed to find the VK_KHR_wayland_surface extension.
```
Here's the `vulkaninfo` output for reference: https://pastebin.com/NYD1z027
If you think it's still useful for anything, do tell, but without the ability to actually render anything on the screen there's hardly any point to include it.

zylo117, 2024-06-29:
> As in:
>
> ```
> ginkage@rock-5c:~/Vulkan-Tools/build/cube$ ./vkcube-wayland
> vkEnumerateInstanceExtensionProperties failed to find the VK_KHR_wayland_surface extension.
> ```
>
> Here's the `vulkaninfo` [...]

well, then at least I get vulkan working in ncnn. I can do ai development 3588

nioroso-x3, 2024-07-09:
> well, then at least I get vulkan working in ncnn. I can do ai development 3588

Can you post the benchmark results to see if it is worth to bother installing this closed driver?

ginkage, 2024-07-09:
@nioroso-x3 What benchmark do you want me to run?

CendaL, 2024-07-14:
@zylo117 thanks for this driver.

I have Orange Pi 5 Pro and I was able to enable HW acceleration in both Jellyfin and Chromium browser with this driver!

Chromium acceleration works fine after ubuntu-rockchip installation but it stops working when you install Jellyfin's recommended ARM Mali OpenCL runtime (libmali) [v1.9-1-2d267b0](https://github.com/tsukumijima/libmali-rockchip/releases/download/v1.9-1-2d267b0/libmali-valhall-g610-g13p0-gbm_1.9-1_arm64.deb) (based on https://jellyfin.org/docs/general/administration/hardware-acceleration/rockchip#debian-and-ubuntu-linux).

Shivansps, 2024-07-21:
If it is headless only, it can still run mesa zink?

defencedog, 2024-08-06:
@CendaL thanks for explanation. Deom the same repo I downloaded userspace driver for `MaliG52` ...however there are many versions & I am unable to understand [syntax](https://github.com/tsukumijima/libmali-rockchip/issues/6) 

Whenever I install any `gbm` or non gbm release ... & I login to `gnome X11` session it breaks ... 

> Something went wrong...

ginkage, 2024-09-14:
After a few days' worth of digging: with some patches to the kernel, and using an extra WSI layer (hacked a little), and g21 DDK/firmware, and with g13 blob for the "main" GLES driver (no Panfork), this thing can actually be made to run PS2/GC emulation on Noble!

ginkage, 2024-09-14:
And `vulkaninfo` now looks like this: https://pastebin.com/35H2dvsR

Lolmc0587, 2024-09-17:
how to install it?

svesnav, 2024-09-26:
Please write guide how to install rockchip mali drivers for vulkan support!

Bleach665, 2024-10-28:
Just leave this here
https://www.collabora.com/news-and-blog/news-and-events/panvk-v10-support.html

Bleach665, 2024-10-28:
@zylo117 . Would you be so kind share u `vulkaninfo` output?
I install `rockchip-mali_1.9-12_arm64.deb` in `ubuntu-24.04-preinstalled-desktop-arm64-rock-5c.img.xz`, but unfortunately
<details> 

` vulkaninfo`
ERROR: [Loader Message] Code 0 : vkCreateInstance: Found no drivers!
Cannot create Vulkan instance.
This problem is often caused by a faulty installation of the Vulkan driver or attempting to use a GPU that does not support Vulkan.
ERROR at ./vulkaninfo/./vulkaninfo.h:458:vkCreateInstance failed with ERROR_INCOMPATIBLE_DRIVER
</details>

Maybe something else needs to be done besides installing the deb?

Bleach665, 2024-10-30:
This work for me. `vulkaninfo` output is correct, ncnn fork fine with vulkan backend.

https://github.com/Bleach665/Mali610Vulkan

GamingCity, 2024-10-30:
You can run zink over it?

Al3XKOoL, 2024-12-07:
Look this:
https://github.com/ROCKNIX/distribution/pull/656/commits

Rocknix have working vulkan for emulation

closed 2025-02-10 by github-actions[bot]