Joshua-Riek/ubuntu-rockchip #890: When RDP server is open, GPU is disabled!

thomasbergersen, 2024-06-26:
[image]

RDP client using mac version 'Thincast'.
[image]

In addition, the mac version of Microsoft Remote Desktop can't connect RDP server.
[image]

Joshua-Riek, 2024-06-26:
Can you install glmark2-es2-wayland and run it? Does it have high fps?

thomasbergersen, 2024-06-26:
> Can you install glmark2-es2-wayland and run it? Does it have high fps?

It looks like the gpu is involved in the computation

Hi @Joshua-Riek  This is test relust:

thomasbergersen@thomasbergersen-desktop:~$ glmark2-es2-wayland
warning: queue 0xaaaae04ed650 destroyed while proxies still attached:
  wl_display@1 still attached
=======================================================
    glmark2 2023.01
=======================================================
    OpenGL Information
    GL_VENDOR:      Mesa/X.org
    GL_RENDERER:    llvmpipe (LLVM 14.0.6, 128 bits)
    GL_VERSION:     OpenGL ES 3.2 Mesa 23.0.0-devel
    Surface Config: buf=32 r=8 g=8 b=8 a=8 depth=32 stencil=0 samples=0
    Surface Size:   800x600 windowed
=======================================================
[build] use-vbo=false: FPS: 47 FrameTime: 21.372 ms
[build] use-vbo=true: FPS: 47 FrameTime: 21.685 ms
[texture] texture-filter=nearest: FPS: 47 FrameTime: 21.385 ms
[texture] texture-filter=linear: FPS: 47 FrameTime: 21.402 ms
[texture] texture-filter=mipmap: FPS: 45 FrameTime: 22.452 ms
[shading] shading=gouraud: FPS: 51 FrameTime: 19.977 ms
[shading] shading=blinn-phong-inf: FPS: 49 FrameTime: 20.441 ms
[shading] shading=phong: FPS: 48 FrameTime: 20.871 ms
[shading] shading=cel: FPS: 32 FrameTime: 31.761 ms
[bump] bump-render=high-poly: FPS: 38 FrameTime: 26.508 ms
[bump] bump-render=normals: FPS: 49 FrameTime: 20.545 ms
[bump] bump-render=height: FPS: 55 FrameTime: 18.506 ms
[effect2d] kernel=0,1,0;1,-4,1;0,1,0;: FPS: 51 FrameTime: 19.672 ms
[effect2d] kernel=1,1,1,1,1;1,1,1,1,1;1,1,1,1,1;: FPS: 46 FrameTime: 21.924 ms
[pulsar] light=false:quads=5:texture=false: FPS: 29 FrameTime: 34.609 ms
[desktop] blur-radius=5:effect=blur:passes=1:separable=true:windows=4: FPS: 25 FrameTime: 40.425 ms
[desktop] effect=shadow:windows=4: FPS: 37 FrameTime: 27.216 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 41 FrameTime: 24.686 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=subdata: FPS: 28 FrameTime: 36.451 ms
[buffer] columns=200:interleave=true:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 42 FrameTime: 23.833 ms
[ideas] speed=duration: FPS: 36 FrameTime: 28.115 ms
[jellyfish] <default>: FPS: 29 FrameTime: 34.597 ms
[terrain] <default>: FPS: 5 FrameTime: 227.352 ms
[shadow] <default>: FPS: 39 FrameTime: 25.940 ms
[refract] <default>: FPS: 13 FrameTime: 80.855 ms
[conditionals] fragment-steps=0:vertex-steps=0: FPS: 41 FrameTime: 24.436 ms
[conditionals] fragment-steps=5:vertex-steps=0: FPS: 47 FrameTime: 21.334 ms
[conditionals] fragment-steps=0:vertex-steps=5: FPS: 48 FrameTime: 21.015 ms
[function] fragment-complexity=low:fragment-steps=5: FPS: 47 FrameTime: 21.329 ms
[function] fragment-complexity=medium:fragment-steps=5: FPS: 30 FrameTime: 34.452 ms
[loop] fragment-loop=false:fragment-steps=5:vertex-steps=5: FPS: 49 FrameTime: 20.572 ms
[loop] fragment-steps=5:fragment-uniform=false:vertex-steps=5: FPS: 34 FrameTime: 29.496 ms
[loop] fragment-steps=5:fragment-uniform=true:vertex-steps=5: FPS: 44 FrameTime: 22.768 ms
=======================================================
                                  glmark2 Score: 38 
=======================================================
warning: queue 0xaaaae04ee2d0 destroyed while proxies still attached:
  wl_display@1 still attached

Joshua-Riek, 2024-06-28:
Sadly the GPU driver is dead and has been for 2 years now, so if this is a GPU related bug it will likely never be fixed. 

However this could very well be an issue with the remote desktop protocol, but this is something I know very little about.

thomasbergersen, 2024-06-29:
Okay, it has been confirmed that the issue was with RDP connection！ Thank you.

closed 2024-06-29 by thomasbergersen