Joshua-Riek/ubuntu-rockchip #891: Radxa Zero 3W - Enable i2c

PhilShaw852, 2024-06-27:
Hi,

I am using the Radxa 3W server image from here: https://joshua-riek.github.io/ubuntu-rockchip-download/boards/radxa-zero3.html

I can see the port with i2cdetect -y 3 but my device is not detected.

I have seen posts for editing the ubuntuEnv.txt file but cannot find it on this image. There is no file in /boot, nor is there a /firmware folder.

Any help would be much appreciated.

Manu7788, 2024-06-27:
https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Ubuntu-22.04-LTS

Joshua-Riek, 2024-06-29:
Device tree overlays for RK3566 / RK3568 need to be added to the kernel.

chengguizi, 2024-07-09:
To add on this, when I use debugger to get UART console log, I get:

[   33.025641] rk3x-i2c fe5c0000.i2c: timeout, ipd: 0x00, state: 1              
[   33.158947] rk3x-i2c fe5c0000.i2c: timeout, ipd: 0x00, state: 1              
[   33.262334] rk3x-i2c fe5c0000.i2c: timeout, ipd: 0x00, state: 1              
[   33.405597] rk3x-i2c fe5c0000.i2c: timeout, ipd: 0x00, state: 1              
[   33.509006] rk3x-i2c fe5c0000.i2c: timeout, ipd: 0x00, state: 1              
[   33.642295] rk3x-i2c fe5c0000.i2c: timeout, ipd: 0x00, state: 1              
[   33.748912] rk3x-i2c fe5c0000.i2c: timeout, ipd: 0x00, state: 1              
[   33.892273] rk3x-i2c fe5c0000.i2c: timeout, ipd: 0x00, state: 1              
[   33.995635] rk3x-i2c fe5c0000.i2c: timeout, ipd: 0x00, state: 1              
[   34.128939] rk3x-i2c fe5c0000.i2c: timeout, ipd: 0x00, state: 1

May I know which overlays to apply to the kernel to fix this?

mikecarr, 2024-09-18:
> To add on this, when I use debugger to get UART console log, I get:
>
> [ 33.025641] rk3x-i2c fe5c0000.i2c: timeout, ipd: 0x00, state: 1 [ 33.158947] rk3x-i2c fe5c0000.i2c: timeout, ipd: 0x00, state: 1 [...]

Did you ever figure out what this is?

chengguizi, 2024-09-21:
Not yet..

closed 2025-02-10 by github-actions[bot]