Joshua-Riek/ubuntu-rockchip #893: WiringPi support for Rock 5B

myc-ab, 2024-06-28:
Hi!
How do I properly configure WiringPi for the Rock 5B? After installing the `wiringpi` package from apt, the `gpio readall` command returns an error:
```
Oops: Unable to determine board revision from /proc/device-tree/system/linux,revision
or from /proc/cpuinfo
 -> No "Hardware" line
 ->  You'd best google the error to find out why.
```

Joshua-Riek, 2024-06-28:
From my understanding WiringPi does not support the Rock 5B.

RadxaStephen, 2024-07-09:
> Hi! How do I properly configure WiringPi for the Rock 5B? After installing the `wiringpi` package from apt, the `gpio readall` command returns an error:
> 
> ```
> Oops: Unable to determine board revision from /proc/device-tree/system/linux,revision
> or from /proc/cpuinfo
>  -> No "Hardware" line
>  ->  You'd best google the error to find out why.
> ```
> 
> System: Ubuntu Server 24.04 with 6.1.0-1017-rockchip

How about WiringX? WiringX supports ROCK 5B now. https://github.com/wiringX/wiringX/

closed 2025-02-10 by github-actions[bot]