Joshua-Riek/ubuntu-rockchip #894: can't enable hdmirx on orangepi5 plus

michelehokuto, 2024-06-28:
Morning, 
I'm new to this project. I would like to record on the HDMI input port but I saw that the device /dev/video0 does not exist.

Reading I understood that it was disabled by default:
[https://github.com/armbian/linux-rockchip/pull/173](url)
[https://github.com/Joshua-Riek/ubuntu-rockchip/issues/758](url)

I tried adding an overlay to enable it but it doesn't work, here are the things I tried:

**add overlay on file: /etc/default/u-boot** 

##/etc/default/u-boot - configuration file for u-boot-update(8)+
#U_BOOT_UPDATE="true"
#U_BOOT_ALTERNATIVES="default recovery"
#U_BOOT_DEFAULT="l0"
#U_BOOT_PROMPT="1"
#U_BOOT_ENTRIES="all"
#U_BOOT_MENU_LABEL="Debian GNU/Linux"
#U_BOOT_PARAMETERS="ro earlycon"
#U_BOOT_ROOT=""
#U_BOOT_TIMEOUT="50"
#U_BOOT_FDT=""
#U_BOOT_FDT_DIR="/lib/firmware/"
#U_BOOT_FDT_OVERLAYS="device-tree/rockchip/overlay/rk3588-hdmirx.dtbo"
#U_BOOT_FDT_OVERLAYS_DIR="/lib/firmware/"
#U_BOOT_SYNC_DTBS="false"

**update u-boot configuration**

sudo u-boot-update 

P: Checking for EXTLINUX directory... found.
P: Writing config for vmlinuz-6.1.0-1017-rockchip...
P: Writing config for vmlinuz-6.1.0-1016-rockchip...

**check extlinux.conf**
`cat /boot/extlinux/extlinux.conf | grep hdmi`
result: 'hdmi' there isn't in the file

I also checked if the /dev/video0 device exists after reboot but it doesn't.

What should I do? Thank you

N.B. I also updated the whole system, the kernel is:
 `uname -a

Linux orange-desktop 6.1.0-1017-rockchip #17-Ubuntu SMP Sun Jun 23 19:14:56 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux
`

Joshua-Riek, 2024-06-28:
You need to uncomment `U_BOOT_FDT_OVERLAYS`, remove the "#".
```diff
13c13
< #U_BOOT_FDT_OVERLAYS="device-tree/rockchip/overlay/rk3588-hdmirx.dtbo"
---
> U_BOOT_FDT_OVERLAYS="device-tree/rockchip/overlay/rk3588-hdmirx.dtbo"
```

michelehokuto, 2024-06-30:
sorry for late, now it work. Device /dev/video0 exist, but I still can't record on obs, nothing comes out on the screen.
I followed the instructions in this video, I installed obs-studio and obs-gstreamer but it doesn't work, any advice?
[https://www.youtube.com/watch?v=i_L0mGepOtI](url)

Joshua-Riek, 2024-06-30:
OBS can not record from HDMI input, this is a problem with the BSP kernel from Rockchip.

Please see https://github.com/Joshua-Riek/ubuntu-rockchip/issues/830

michelehokuto, 2024-06-30:
yes, I was reading the issue. It seems like it works on obs, maybe I'm wrong [#830](https://github.com/Joshua-Riek/ubuntu-rockchip/issues/830#issuecomment-2170478491) , I'm not an expert in the field. My goal is to record 1920x1080 60fps, is there an easy way to do this? I'll tag @MaximKurakin too, maybe he can give me a hand given his experience. Thank you

Joshua-Riek, 2024-06-30:
Using gstreamer via a command line, the below issue has some useful info:

https://github.com/Joshua-Riek/ubuntu-rockchip/issues/252#issuecomment-1629302255

michelehokuto, 2024-06-30:
with gstreamer it works, but the frame rate is low. This is the command:
`gst-launch-1.0 v4l2src device=/dev/video0 io-mode=4 ! videoconvert ! video/x-raw,format=NV12,width=1920,height=1080,framerate=30/1 ! videoscale ! video/x-raw,width=1280,height=720 ! autovideosink sync=false 2>&1 > /dev/null`

Joshua-Riek, 2024-06-30:
Your limiting the frame rate it looks like, you should read the documentation and look at how other users are capturing input on RK3588 devices.

Joshua-Riek, 2024-06-30:
Anyway closing, as hdmirx can be enabled.

closed 2024-06-30 by Joshua-Riek