Joshua-Riek/ubuntu-rockchip #896: Orange Pi 5 Pro won't boot, memory errors on serial output - Help needed!

thisismz, 2024-06-29:
Hey everyone, I'm having trouble with my Orange Pi 5 Pro and could use some help. Here's what's happening: - When I insert the SD card, the board doesn't boot

- Only the red LED lights up
- I'm getting error messages through the serial port Here's the serial output:
`
DDR 9fffbe1e78 cym 24/02/04-10:09:20,fwver: v1.16
LPDDR5, 2400MHz
channel[0] BW=16 Col=10 Bk=16 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=4096MB
channel[1] BW=16 Col=10 Bk=16 CS0 Row=16 CS1 Row=16 CS=2 Die BW=channel[2] BW=16 Col=10 Bk=16 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=4096MB
channel[3] BW=16 Col=10 Bk=16 CS0 Row=16 CS=1 Die BW=16 Size=2048MB
Manufacturer ID:0xff
ch:3 dq0 fail,write:0x1,read:0x0
ch:3 dq1 fail,write:0x2,read:0x0
write:0x8,read:0x0e:0x4,read:0x0
ch:3 dq4 fail,write:0x10,read:0x0
ch:3 dq5 fail,write:0x20,read:0x0
ch:3 dq6 fail,write:0x40,read:0x0
ch:3 dq7 fail,write:0x80,read:0x0
ch:3 dq8 fail,write:0x100,read:0h:3 dq9 fail,write:0x200,read:0x0
ch:3 dq10 fail,write:0x400,read:0x0
ch:3 dq11 fail,write:0x8ch:3 dq12 fail,write:0x1000,read:0x0
ch:3 dq13 fail,write:0x2000,read:0x0
ch:3 dq14 fail,writech:3 dq15 fail,write:0x8000,read:0x0
ch:3 dq0 fail,write:0xfffffffe,read:0x0
ch:3 dq1 fail,write:0xfffffffd,read:0x0
ch:3 dq2 fail,write:0xfffffffb,read:0x0
ch:3 dq3 fail,write:0xfffffff7,ch:3 dq4 fail,write:0xffffffef,read:0x0
ch:3 dq5 fail,write:0xffffffdf,read:0x0
ch:3 dq6 fail,ad:0x0
ch:3 dq7 fail,write:0xffffff7f,read:0x0
ch:3 dq8 fail,write:0xfffffeff,read:0x0
ch:3 dq9 fail,write:0xfffffdff,read:0x0
ch:3 dq10 fail,write:0xfffffbff,read:0x0
ch:3 dq11 fail,writfff7ff,read:0x0
ch:3 dq12 fail,write:0xffffefff,read:0x0
ch:3 dq13 fail,write:0xffffdfff,read:write:0xffffbfff,read:0x0
ch:3 dq15 fail,write:0xffff7fff,read:0x0
error
ERR
`

It looks like there might be issues with the LPDDR5 memory, especially in channel 3. All data lines (DQ0 to DQ15) are failing read/write tests.
What I've tried so far:

1. Checked for any visible physical damage (none found)
2. Made sure I'm using a stable power supply
3. Tested with a different SD card

Has anyone encountered something similar or have any ideas on what might be causing this? Any suggestions on further troubleshooting steps? Thanks in advance for any help!

Joshua-Riek, 2024-06-29:
I would test with a 5v 4a power supply and make sure that no SPI or eMMC is attached. 

But this is likely a problem with power or a hardware issue with your board, the DDR blob can't see 2GB of ram from channel 3.

Here is the output from my Orange Pi 5 Pro:
```
DDR 9fffbe1e78 cym 24/02/04-10:09:20,fwver: v1.16
LPDDR5, 2400MHz
channel[0] BW=16 Col=10 Bk=16 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=4096MB
channel[1] BW=16 Col=10 Bk=16 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=4096MB
channel[2] BW=16 Col=10 Bk=16 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=4096MB
channel[3] BW=16 Col=10 Bk=16 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=4096MB
Manufacturer ID:0xff
CH0 RX Vref:25.8%, TX Vref:20.0%,20.0%
CH1 RX Vref:27.9%, TX Vref:19.0%,19.0%
CH2 RX Vref:26.7%, TX Vref:21.0%,20.0%
CH3 RX Vref:28.5%, TX Vref:18.0%,19.0%
change to F1: 534MHz
change to F2: 1320MHz
change to F3: 1968MHz
change to F0: 2400MHz
out
U-Boot SPL board init
U-Boot SPL 2017.09 (Jun 12 2024 - 21:12:51)
unknown raw ID 0 0 0
unrecognized JEDEC id bytes: 00, 00, 00
Trying to boot from MMC1
Trying fit image at 0x4000 sector
## Verified-boot: 0
## Checking atf-1 0x00040000 ... sha256(a7d1d8d191...) + OK
## Checking uboot 0x00200000 ... sha256(eb0ce6dacd...) + OK
## Checking fdt 0x0034a300 ... sha256(1b12611ccd...) + OK
## Checking atf-2 0xff100000 ... sha256(4b2065349b...) + OK
## Checking atf-3 0x000f0000 ... sha256(aa71013e72...) + OK
Jumping to U-Boot(0x00200000) via ARM Trusted Firmware(0x00040000)
Total: 232.55 ms

INFO:    Preloader serial: 2
NOTICE:  BL31: v2.3():v2.3-682-g4ca8a8422:derrick.huang, fwver: v1.45
NOTICE:  BL31: Built : 10:11:21, Dec 27 2023
INFO:    spec: 0x13
INFO:    code: 0x88
INFO:    ext 32k is valid
INFO:    ddr: stride-en 4CH
INFO:    GICv3 without legacy support detected.
INFO:    ARM GICv3 driver initialized in EL3
INFO:    valid_cpu_msk=0xff bcore0_rst = 0x0, bcore1_rst = 0x0
INFO:    l3 cache partition cfg-0
INFO:    system boots from cpu-hwid-0
INFO:    disable memory repair
INFO:    idle_st=0x21fff, pd_st=0x11fff9, repair_st=0xfff70001
INFO:    dfs DDR fsp_params[0].freq_mhz= 2400MHz
INFO:    dfs DDR fsp_params[1].freq_mhz= 534MHz
INFO:    dfs DDR fsp_params[2].freq_mhz= 1320MHz
INFO:    dfs DDR fsp_params[3].freq_mhz= 1968MHz
INFO:    BL31: Initialising Exception Handling Framework
INFO:    BL31: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE initialization. SMC`s destined for OPTEE will return SMC_UNK
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x200000
INFO:    SPSR = 0x3c9

U-Boot 2017.09 (Jun 12 2024 - 21:12:51 +0000)

Model: RK3588 Orange Pi 5 Pro
PreSerial: 2, raw, 0xfeb50000
DRAM:  16 GiB
Sysmem: init
Relocation Offset: eda2c000
Relocation fdt: eb9f8fc8 - eb9fecc8
CR: M/C/I
Using default environment

mmc@fe2c0000: 0, mmc@fe2e0000: 1
Bootdev(atags): mmc 0
MMC0: Legacy, 52Mhz
PartType: EFI
DM: v2
boot mode: None
Model: RK3588 Orange Pi 5 Pro
CLK: (sync kernel. arm: enter 1008000 KHz, init 1008000 KHz, kernel 0N/A)
  b0pll 24000 KHz
  b1pll 24000 KHz
  lpll 24000 KHz
  v0pll 24000 KHz
  aupll 24000 KHz
  cpll 1500000 KHz
  gpll 1188000 KHz
  npll 24000 KHz
  ppll 1100000 KHz
  aclk_center_root 702000 KHz
  pclk_center_root 100000 KHz
  hclk_center_root 396000 KHz
  aclk_center_low_root 500000 KHz
  aclk_top_root 750000 KHz
  pclk_top_root 100000 KHz
  aclk_low_top_root 396000 KHz
Net:   No ethernet found.
Hit key to stop autoboot('CTRL+C'):  0 
switch to partitions #0, OK
mmc0 is current device
mmc@fe2c0000: 0 (SD)
mmc@fe2e0000: 1
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:2...
Found /boot/extlinux/extlinux.conf
Retrieving file: /boot/extlinux/extlinux.conf
991 bytes read in 17 ms (56.6 KiB/s)
U-Boot menu
1:	Ubuntu 22.04.4 LTS 5.10.0-1007-rockchip
2:	Ubuntu 22.04.4 LTS 5.10.0-1007-rockchip (rescue target)
Enter choice: 1:	Ubuntu 22.04.4 LTS 5.10.0-1007-rockchip
Retrieving file: /boot/initrd.img-5.10.0-1007-rockchip
16149072 bytes read in 1311 ms (11.7 MiB/s)
Retrieving file: /boot/vmlinuz-5.10.0-1007-rockchip
39258624 bytes read in 3173 ms (11.8 MiB/s)
append: root=UUID=0e84247b-eb64-44b9-a976-b86aa05f95c2 rootwait rw console=ttyS2,1500000 console=tty1 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory
Retrieving file: /lib/firmware/5.10.0-1007-rockchip/device-tree/rockchip/rk3588s-orangepi-5-pro.dtb
242544 bytes read in 871 ms (271.5 KiB/s)
Fdt Ramdisk skip relocation
## Flattened Device Tree blob at 0x0a100000
   Booting using the fdt blob at 0x0a100000
  'reserved-memory' ramoops@110000: addr=110000 size=f0000
   Using Device Tree in place at 000000000a100000, end 000000000a13e36f
Adding bank: 0x00200000 - 0xf0000000 (size: 0xefe00000)
Adding bank: 0x100000000 - 0x3fc000000 (size: 0x2fc000000)
Adding bank: 0x3fc500000 - 0x3fff00000 (size: 0x03a00000)
Adding bank: 0x4f0000000 - 0x500000000 (size: 0x10000000)
Total: 9933.866 ms

Starting kernel ...
```

thisismz, 2024-06-30:
My Orange Pi has 16 GB of RAM. In reality, we should have 4 channels of 4 gigabytes each, but we have 3 channels of 4 GB and one channel of 2 GB. I don't know if there's a solution to disable one of these channels.

Joshua-Riek, 2024-06-30:
I would argue your device is DOA and you should try to get a replacement.

closed 2024-07-03 by Joshua-Riek