Joshua-Riek/ubuntu-rockchip #898: 关于opi5-pro的一些系统错误

m2kall, 2024-06-30:
你好 
  我安装使用的是ubuntu24.04内核6.1.0-1017的桌面wayland版本，安装该系统的机器型号为orange pi 5 Pro
  发现了一些小问题，比如说wifi/bt5 会使用几分钟或者更长时间会失去联系，右上角的wifi跟蓝牙图标都没有。
查询了下系统的日志，发现了以下内容。另外今天的有线网络也失去联系好多次，不排除是我做了其他改动。
以下为系统日志内容，请查看：

19:27
<error> [1719746839.6599] platform-linux: error reading net:/sys/class/net/veth024b70e/dev_id: error reading 4096 bytes from file descriptor: Invalid argument
NetworkManager
18:50
<error> [1719744651.9067] platform-linux: error reading net:/sys/class/net/vetha1f3e5e/dev_id: error reading 4096 bytes from file descriptor: Invalid argument
NetworkManager
18:39
rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx phy lane locked!
kernel
18:39
rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx phy pll locked!
kernel
18:27
<error> [1719743230.5026] platform-linux: error reading net:/sys/class/net/vethf153f2d/dev_id: error reading 4096 bytes from file descriptor: Invalid argument
NetworkManager
18:19
Failed to start fwupd-refresh.service - Refresh fwupd metadata and update motd.
systemd
17:49
Failed to start app-gnome-ubuntu\x2dreport\x2don\x2dupgrade-5293.scope - Application launched by gnome-session-binary.
systemd
17:49
gkr-pam: unable to locate daemon control file
gdm-password]
17:45
Failed to set mode: Failed (0x03)
bluetoothd
17:45
sap-server: Operation not permitted (1)
bluetoothd
17:45
profiles/sap/server.c:sap_server_register() Sap driver initialization failed.
bluetoothd
17:45
of_dma_request_slave_channel: dma-names property of node '/serial@febc0000' missing or empty
kernel
17:45
rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx phy lane locked!
kernel
17:45
rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx phy pll locked!
kernel
17:45
debugfs: File 'constraint_flags' in directory '/' already present!
kernel
17:45
debugfs: File 'max_uV' in directory '/' already present!
kernel
17:45
debugfs: File 'min_uV' in directory '/' already present!
kernel
17:45
debugfs: File 'uA_load' in directory '/' already present!
kernel
17:45
debugfs: Directory 'fdab0000.npu-rknpu' with parent 'vdd_npu_s0' already present!
kernel
2发生
17:45
debugfs: File 'constraint_flags' in directory '/' already present!
kernel
17:45
debugfs: File 'max_uV' in directory '/' already present!
kernel
17:45
debugfs: File 'min_uV' in directory '/' already present!
kernel
17:45
debugfs: File 'uA_load' in directory '/' already present!
kernel
17:45
RKNPU fdab0000.npu: can't request region for resource [mem 0xfdad0000-0xfdadffff]
kernel
17:45
RKNPU fdab0000.npu: can't request region for resource [mem 0xfdac0000-0xfdacffff]
kernel
17:45
RKNPU fdab0000.npu: can't request region for resource [mem 0xfdab0000-0xfdabffff]
kernel
17:45
debugfs: File 'constraint_flags' in directory '/' already present!
kernel
17:45
debugfs: File 'max_uV' in directory '/' already present!
kernel
17:45
debugfs: File 'min_uV' in directory '/' already present!
kernel
17:45
debugfs: File 'uA_load' in directory '/' already present!
kernel
17:45
debugfs: Directory 'fb000000.gpu-mali' with parent 'vdd_gpu_s0' already present!
kernel
17:45
rockchip-dmc dmc: failed to get vop pn to msch rl
kernel
17:45
rockchip-dmc dmc: failed to get vop bandwidth to dmc rate
kernel
17:45
debugfs: Directory 'fb000000.gpu-mali' with parent 'vdd_gpu_s0' already present!
kernel
17:45
rkvenc_init:1870: failed to add venc devfreq
kernel
17:45
mpp_rkvenc2 fdbe0000.rkvenc-core: failed to init_opp_table
kernel
17:45
mpp_rkvenc2 fdbe0000.rkvenc-core: failed to init opp info
kernel
17:45
mpp_rkvenc2 fdbe0000.rkvenc-core: failed to set opp config
kernel
17:45
mpp_rkvenc2 fdbe0000.rkvenc-core: error -ENODEV: _opp_set_regulators: no regulator (venc) found
kernel
17:45
mpp_rkvenc2 fdbe0000.rkvenc-core: rockchip_init_read_margin: no regulator (venc) found: -19
kernel
17:45
rkvenc_init:1870: failed to add venc devfreq
kernel
17:45
mpp_rkvenc2 fdbd0000.rkvenc-core: failed to init_opp_table
kernel
17:45
mpp_rkvenc2 fdbd0000.rkvenc-core: failed to init opp info
kernel
17:45
mpp_rkvenc2 fdbd0000.rkvenc-core: failed to set opp config
kernel
17:45
mpp_rkvenc2 fdbd0000.rkvenc-core: error -ENODEV: _opp_set_regulators: no regulator (venc) found
kernel
17:45
mpp_rkvenc2 fdbd0000.rkvenc-core: rockchip_init_read_margin: no regulator (venc) found: -19
kernel
17:45
rkvdec2_init:985: No niu hclk reset resource define
kernel
17:45
mpp_rkvdec2 fdc48100.rkvdec-core: shared_niu_h is not found!
kernel
17:45
rkvdec2_init:982: No niu aclk reset resource define
kernel
17:45
mpp_rkvdec2 fdc48100.rkvdec-core: shared_niu_a is not found!
kernel
17:45
rkvdec2_init:985: No niu hclk reset resource define
kernel
17:45
mpp_rkvdec2 fdc38100.rkvdec-core: shared_niu_h is not found!
kernel
17:45
rkvdec2_init:982: No niu aclk reset resource define
kernel
17:45
mpp_rkvdec2 fdc38100.rkvdec-core: shared_niu_a is not found!
kernel
17:45
mpp-iep2 fdbb0000.iep: allocate roi buffer failed
kernel
17:45
ES8323 3-0011: ASoC: error at soc_component_write_no_lock on ES8323.3-0011 for register: [0x00000035] -5
kernel
17:45
debugfs: File 'Capture' in directory 'dapm' already present!
kernel
17:45
arm-scmi firmware:scmi: Failed. SCMI protocol 17 not active.
kernel
17:45
rk806 spi2.0: no dvs-setting pinctrl state
kernel
17:45
rk806 spi2.0: no reset-setting pinctrl state
kernel
17:45
rk806 spi2.0: no sleep-setting state
kernel
17:45
rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx phy lane locked!
kernel
17:45
rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx phy pll locked!
kernel
17:45
rockchip-vop2 fdd90000.vop: failed to init_opp_table
kernel
17:45
rockchip-vop2 fdd90000.vop: failed to init opp info
kernel
17:45
rockchip-vop2 fdd90000.vop: failed to set opp config
kernel
17:45
rockchip-vop2 fdd90000.vop: error -ENODEV: _opp_set_regulators: no regulator (vop) found
kernel
17:45
dw-pcie fe190000.pcie: Failed to initialize host
kernel
17:45
dw-pcie fe190000.pcie: invalid resource
kernel
17:45
dw-pcie fe180000.pcie: Failed to initialize host
kernel
17:45
dw-pcie fe180000.pcie: invalid resource
kernel
17:45
fiq_debugger_probe: could not install nmi irq handler
kernel
17:45
fiq_debugger fiq_debugger.0: error -ENXIO: IRQ wakeup not found
kernel
17:45
fiq_debugger fiq_debugger.0: error -ENXIO: IRQ fiq not found
kernel
17:45
OF: fdt: Reserved memory: failed to reserve memory for node 'drm-cubic-lut@00000000': base 0x0000000000000000, size 0 MiB
kernel
17:45
OF: fdt: Reserved memory: failed to reserve memory for node 'drm-logo@00000000': base 0x0000000000000000, size 0 MiB
[image]

https://github.com/Joshua-Riek/ubuntu-rockchip/assets/42733818/1b730620-c6d4-4dc7-9c64-4b6eb9113582

Joshua-Riek, 2024-06-30:
Hi, this is a duplicate of https://github.com/Joshua-Riek/ubuntu-rockchip/issues/841 and due to conflicting drivers this is almost impossible to fix without creating a second kernel which I will not do for one device.

closed 2024-06-30 by Joshua-Riek

m2kall, 2024-06-30:
> Hi, this is a duplicate of #841 and due to conflicting drivers this is almost impossible to fix without creating a second kernel which I will not do for one device.嗨，这是的 #841 副本，由于驱动程序冲突，如果不创建第二个内核，这几乎不可能修复，我不会为一台设备做。

 好的，还是谢谢你