Joshua-Riek/ubuntu-rockchip #899: APT Initramfs-tools error

cody8764, 2024-06-30:
I was using ubuntu normally but I then did an apt upgrade. I suddenly got an error saying: 

Processing triggers for initramfs-tools (0.142ubuntu25.1) ...
update-initramfs: Generating /boot/initrd.img-6.1.0-1017-rockchip
Unsupported platform 'Orange Pi 5 Plus'.
run-parts: /etc/initramfs/post-update.d//flash-kernel exited with return code 1
dpkg: error processing package initramfs-tools (--configure):
 installed initramfs-tools package post-installation script subprocess returned 
error exit status 1
Errors were encountered while processing:
 initramfs-tools
E: Sub-process /usr/bin/dpkg returned an error code (1)

How do I fix this error so that I can use APT normally again?

Joshua-Riek, 2024-06-30:
For some reason you installed `flash-kernel`. Purge the package.

cody8764, 2024-07-01:
Thanks, It solved my issue!

closed 2024-07-01 by cody8764

alexorex, 2025-02-05:
I deployed 22.04 Server, upgraded, then installed minimal Gnome and got the same "Unsupported platform 'Orange Pi 5 Plus'" error having installed mpv, which was resolved after purging the flash-kernel.
Can't the package be somehow forbidden in O-Pi distro version?
Thanks for the solution anyway!