Joshua-Riek/ubuntu-rockchip #900: No overlay map after system upgrade

mrslowfloww, 2024-07-01:
After the latest system upgrade (to 6.1.0-1018), there's no overlay map in /lib/firmware/6.1.0-1018-rockchip/device-tree/rockchip . So can't use GPIO, display or anything else that requires overlay files. 

(Orange Pi 5 Plus)

Joshua-Riek, 2024-07-01:
Crap, looking into it now

Joshua-Riek, 2024-07-01:
Should be fixed with https://github.com/Joshua-Riek/linux-rockchip/commit/bd004a7dc104f12c13557baf92058d51b0febb8a. 

Im uploading the kernel now, but i wont be able to upload the meta package until tonight after work.

mrslowfloww, 2024-07-01:
> Should be fixed with [Joshua-Riek/linux-rockchip@bd004a7](https://github.com/Joshua-Riek/linux-rockchip/commit/bd004a7dc104f12c13557baf92058d51b0febb8a).
> 
> Im uploading the kernel now, but i wont be able to upload the meta package until tonight after work.

It's alright. I temporarily switched back to 6.1.0-1017 until it will be resolved.

Joshua-Riek, 2024-07-01:
The update has been released, and this should be fixed now.

closed 2024-07-01 by Joshua-Riek