Joshua-Riek/ubuntu-rockchip #902: Orange pi 5 pro random freeze

marcoexpo, 2024-07-01:
Hi, I have a brand new Orange pi 5 pro with Ubuntu 24.04 running. I experience random freezing while performing different tasks: browsing, watching videos and so on. I am attaching the system log just after a random freeze. Thank you for the help!
[sys.log](https://github.com/user-attachments/files/16057754/sys.log)

Joshua-Riek, 2024-07-01:
What power supply are you using? %99 of all crashing is a bad power supply

marcoexpo, 2024-07-01:
I am using the power supply bought from the producer of orange pi

Joshua-Riek, 2024-07-01:
That means little. Use something that is 5v 4a.

marcoexpo, 2024-07-01:
Could it be a Bluetooth speaker? I disconnected it and now I am running the orange pi for a while and no freezing has happened.

Joshua-Riek, 2024-07-01:
Yes, wifi and Bluetooth should not be used on this device due to a driver conflict.

The wrong kernel driver is being used, and there is no way to fix this without spending a lot of time hacking the driver out of tree into a dkms module. Or building a second kernel, which I will not do for one device to compensate for the hardware oversight from OrangePi.

See #841

closed 2024-07-01 by Joshua-Riek

marcoexpo, 2024-07-01:
Thank you for the answer! Just one more clarification. Do you mean that Wi-Fi and Bluetooth shouldn't be used simultaneously or that they shouldn't be used at all? Thank you again.

Joshua-Riek, 2024-07-01:
Both WiFi and Bluetooth should not be used at all. I may issue an update to completely disable them to avoid any confusion.

marcoexpo, 2024-07-06:
I tried to deactivate Wifi and Bluetooth through the Ubuntu settings but it doesn't seem to solve the issue. Should I try to deactivate them another way? How could I do it? By the way,  I tried also with a 5v 4a power supply, but nothing changes. Thank you very much again for your help!

marcoexpo, 2024-07-30:
Update: The random freeze was due to the excessive heating of the orange pi 5 pro. I have installed a cooling system and everything is running perfect.