Joshua-Riek/ubuntu-rockchip #907: fdt: Reserverd memory:failed to reserve menory for node 'drm-logo@0':base 0x0000000000000000,size 0 MiB

m2kall, 2024-07-03:
你好，上次你说的需要更新内核，我尝试了下更新到6.1.0-1019.经常会卡顿，fork资源不可用
于是查看了以下内容，帮我看看什么情况好吗，谢谢🙏
[image]

Joshua-Riek, 2024-07-03:
drm-logo is for a splash screen image on boot, but this is not used. 

What software lags?

m2kall, 2024-07-03:
Terminal and browser
I don't know English, so the description using Google Translate may not be very good. So I recorded 3 videos to recreate the scene.
part1. When the traffic proxy software is not used, the browser will turn white in sections (this whitening situation only occurs in this version of Chrominu, not in other versions), and after opening other tasks, the browser will have no service. Sometimes there is no service even if no service is opened.
part2. After using the traffic proxy software clash, the browser has no service more frequently, sometimes the folder cannot be opened, the browser cannot enter text, and the terminal will often bash:fork:retry: resource is temporarily unavailable (this situation often occurs when I did not install the traffic proxy software before)
part3. In the third video, I restarted this orange pi 5-pro and opened these services, without using any traffic proxy. The browser also has no service.

I originally recorded several segmented videos, but the files are too large to upload

https://github.com/Joshua-Riek/ubuntu-rockchip/assets/42733818/0dd61338-9b97-475e-8066-310a3bdbe104

[image]
[image]
[image]
[image]

closed 2024-07-09 by m2kall

cliangtime, 2025-05-22:
我也遇到了一样的问题，请问是怎么解决的呀

m2kall, 2025-05-25:
> 我也遇到了一样的问题，请问是怎么解决的呀

后面我忘了怎么操作的，现在我的系统挺稳定的。

[image]

我现在的系统是这样，有大半年没折腾了，你下载最新的镜像试试。