Joshua-Riek/ubuntu-rockchip #908: Updating u-boot radxa and problems on nanopc-t6

wmport, 2024-07-04:
dmesg 24.04 https://pastebin.com/Af3GJxgE
dmesg 22.04 https://pastebin.com/pZc6wi7P

wmport, 2024-07-04:
The dmesg input for 22.04 does not contain any special errors
[   13.571990] optee: probing for conduit method.
[   13.571997] optee: api uid mismatch
[   13.572006] optee: probe of firmware:optee failed with error -22

[   51.187487] [drm:vop2_plane_atomic_check] *ERROR* Esmart0-win0 is invisible(src: pos[0, 0] rect[64 x 64] dst: pos[2912, 1444] rect[64 x 64]
[   91.126715] EXT4-fs (mmcblk0p1): mounted filesystem with ordered data mode. Opts: errors=remount-ro
[  127.715064] EXT4-fs (nvme0n1p1): mounted filesystem with ordered data mode. Opts: errors=remount-ro

asoc-simple-card rt5616-sound: ASoC: DAPM unknown pin Headphones
[   13.707064] rk-pcie fe170000.pcie: invalid resource
[   13.712060] input: realtek,rt5616-codec Headphones as /devices/platform/rt5616-sound/sound/card4/input4
[   13.713489] rk-pcie fe190000.pcie: invalid prsnt-gpios property in node
[   13.713498] rk-pcie fe190000.pcie: Looking up vpcie3v3-supply from device tree
[   13.713697] vccio_sd_s0: ramp_delay not set
[   13.713899] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up sdio-supply from device tree
[   13.713937] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up sdio-supply property in node /power-management@fd8d8000/power-controller failed
[   13.713996] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up gmac-supply from device tree
[   13.714033] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up gmac-supply property in node /power-management@fd8d8000/power-controller failed
[   13.714092] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up npu-supply from device tree
[   13.714129] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up npu-supply property in node /power-management@fd8d8000/power-controller failed
[   13.714338] rk-pcie fe190000.pcie: IRQ msi not found
[   13.714388] rk-pcie fe190000.pcie: use outband MSI support
[   13.714392] rk-pcie fe190000.pcie: Missing *config* reg space
[   13.714443] rk-pcie fe190000.pcie: host bridge /pcie@fe190000 ranges:
[   13.714456] rk-pcie fe190000.pcie:      err 0x00f4000000..0x00f40fffff -> 0x00f4000000
[   13.714464] rk-pcie fe190000.pcie:       IO 0x00f4100000..0x00f41fffff -> 0x00f4100000
[   13.714472] rk-pcie fe190000.pcie:      MEM 0x00f4200000..0x00f4ffffff -> 0x00f4200000
[   13.714477] rk-pcie fe190000.pcie:      MEM 0x0a00000000..0x0a3fffffff -> 0x0a00000000
[   13.714505] rk-pcie fe190000.pcie: Missing *config* reg space
[   13.714574] rk-pcie fe190000.pcie: invalid resource
[   13.717516] RKNPU fdab0000.npu: Adding to iommu group 0
[   13.717683] RKNPU fdab0000.npu: RKNPU: rknpu iommu is enabled, using iommu mode
[   13.717782] RKNPU fdab0000.npu: Looking up rknpu-supply from device tree
[   13.718432] RKNPU fdab0000.npu: Looking up mem-supply from device tree
[   13.719008] RKNPU fdab0000.npu: can't request region for resource [mem 0xfdab0000-0xfdabffff]
[   13.719161] RKNPU fdab0000.npu: can't request region for resource [mem 0xfdac0000-0xfdacffff]
[   13.719292] RKNPU fdab0000.npu: can't request region for resource [mem 0xfdad0000-0xfdadffff]
[   13.719920] [drm] Initialized rknpu 0.9.2 20231018 for fdab0000.npu on minor 1
[   13.720476] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up nputop-supply from device tree
[   13.720508] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up nputop-supply property in node /power-management@fd8d8000/power-controller failed
[   13.720557] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up npu1-supply from device tree
[   13.720587] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up npu1-supply property in node /power-management@fd8d8000/power-controller failed
[   13.720623] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up npu2-supply from device tree
[   13.720653] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up npu2-supply property in node /power-management@fd8d8000/power-controller failed

Joshua-Riek, 2024-07-04:
I suspect the `route_hdmi0` and `route_hdmi1` nodes causing problems, since they allow for a splash screen to be displayed in U-Boot. I will see if this is the case.

Joshua-Riek, 2024-07-04:
I found the problem, a newly implemented logging function can not handle disabled planes and throws a warning. Since this function is purely for logging purposes we can safely remove the warning and ignore it. A proper solution would be to handle disabled planes correctly in the logging function, but this can be done at a later date.

Fixed with commit: https://github.com/Joshua-Riek/linux-rockchip/commit/a987c818e6d9f0b884d0d3a44cd7e771f07901f0

Kernel log:
```
[   12.518517] WARNING: CPU: 7 PID: 115 at drivers/gpu/drm/rockchip/rockchip_drm_vop2.c:13761 vop2_bind+0xcc0/0x1054
[   12.518521] Modules linked in:
[   12.518525] CPU: 7 PID: 115 Comm: kworker/u16:4 Not tainted 6.1.0-1019-rockchip #19-Ubuntu
[   12.518528] Hardware name: FriendlyElec NanoPC-T6 (DT)
[   12.518530] Workqueue: events_unbound deferred_probe_work_func
[   12.518536] pstate: a0400009 (NzCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   12.518539] pc : vop2_bind+0xcc0/0x1054
[   12.518541] lr : vop2_bind+0xc6c/0x1054
[   12.518543] sp : ffff80000b54b870
[   12.518545] x29: ffff80000b54b910 x28: ffff0003fd83c400 x27: ffff0004f43e4080
[   12.518549] x26: ffff0004f4061000 x25: ffff0004f43e459c x24: ffff8000099ad2f1
[   12.518553] x23: ffff0004f43e5080 x22: ffff8000093bdcc8 x21: ffff0003fd83c410
[   12.518557] x20: 0000000000000003 x19: ffff0004f43e0080 x18: 0000000000000000
[   12.518561] x17: 73616d20656e616c x16: 70206e6769737361 x15: 20327076205d646e
[   12.518566] x14: 69625f32706f763a x13: 5d375b3372657473 x12: 756c43203a646920
[   12.518570] x11: 79687020656e616c x10: 70207972616d6972 x9 : 0000000000000000
[   12.518574] x8 : ffff0004f435bd80 x7 : ffff80000a67be20 x6 : 000000000000003f
[   12.518578] x5 : 0000000000000000 x4 : ffff0004f435bd00 x3 : 0000000000000001
[   12.518582] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffffffffffffffff
[   12.518586] Call trace:
[   12.518588]  vop2_bind+0xcc0/0x1054
[   12.518590]  component_bind+0xc8/0x164
[   12.518594]  component_bind_all+0xb4/0x104
[   12.518597]  rockchip_drm_bind.part.0+0x1b0/0x548
[   12.518602]  rockchip_drm_bind+0x74/0x7c
[   12.518605]  try_to_bring_up_aggregate_device+0xf0/0x140
[   12.518608]  __component_add+0xec/0x128
[   12.518611]  component_add+0x1c/0x28
[   12.518614]  dw_dp_probe+0x6dc/0x738
[   12.518617]  platform_probe+0x70/0xc0
[   12.518621]  really_probe+0x1cc/0x390
[   12.518624]  __driver_probe_device+0x13c/0x154
[   12.518627]  driver_probe_device+0x48/0xd0
[   12.518630]  __device_attach_driver+0x100/0x114
[   12.518633]  bus_for_each_drv+0xa4/0xd0
[   12.518636]  __device_attach+0xf0/0x170
[   12.518639]  device_initial_probe+0x1c/0x28
[   12.518642]  bus_probe_device+0x38/0x9c
[   12.518645]  deferred_probe_work_func+0xd0/0xe4
[   12.518648]  process_one_work+0x1a0/0x250
[   12.518652]  process_scheduled_works+0x4c/0x50
[   12.518655]  worker_thread+0x1f0/0x248
[   12.518658]  kthread+0xc0/0xd0
[   12.518661]  ret_from_fork+0x10/0x20
```

closed 2024-07-04 by Joshua-Riek