Joshua-Riek/ubuntu-rockchip #910: Question: 24.04 - how to build linux-tools-6.1.0-1019-rockchip (and other tools) to improve r8169 driver?

ewaldc, 2024-07-04:
First of all: congratulations on your Noble release with 6.1. kernel.  It has been super stable for my use case, most things run remarkebly faster.
Only the new Realtek r8169 driver has given me a few crashes and data corruptions (Orange Pi 5+'s as heavy use NAS with Samba/NFS and Cubernetes cluster).  I have helped to identify some root causes/fixes upstream specifically for the 8125B related issues.

I forked [ubuntu-linux](https://github.com/ewaldc/linux-rockchip), created a build script to build custom kernels (including support for incremental builds  vs. (slow) clean builds) and custom packages e.g. linux-buildinfo-6.1.0-1019+ewald_arm64.deb and managed to backport the 6.8 version of the r8169 driver (6.10 version requires too many changes in the 6.1 networking stack) with some critical post-6.8 fixes added.  Everything works great: no more crashes and getting 118MB/s Samba read/write to Windows 10/11 on 1Gigabit and ~200MB/s on 2.5Gb.  

As a final action before contributing this back, I wanted to use the "perf" tools to also improve the performance for 2.5Gb networks.  However, I could not find a way to create the linux-tools-6.1.0-1019-rockchip and linux-rockchip-tools-6.1.0-1019 packages (and linux-modules-extra-6.1.0-1019-rockchip), required to run the "perf" tool (gcov works). 
`fakeroot debian/rules binary-headers binary-rockchip do_mainline_build=true` creates buildinfo, headers, image, modules and rockchip headers.  

But how to create the tools and extra packages ?  I know you are very busy, it's not urgent (problem already fixed) and hopefully a simple question for you. I scanned the debian rules files, but nothing seemed to work.

Once I finish the perfomance tuning, I would be happy to contribute the updated driver back.  What works best for you?  A patch file would probably be the easiest as it would also (likely) apply to a newer 6.1.x version.  If so, I can add a patch folder, and post the patches there with release notes. 
 
Thanks in advance.

Joshua-Riek, 2024-07-05:
Thanks, its been quite a fun road so far!

There are a lot of parts reverse-engineered from Ubuntu for Raspberry Pi and Ubuntu Asahi as Canonical does not really document the kernel packaging process. To build the full kernel package you must be running on an arm64 host system and remove the `do_mainline_build=true` flag. As this flag will not build DKMS support and a few other packages as you have observed, however its not friendly to cross compiling.

To compile everything you can use: `fakeroot debian/rules clean binary`

---

There could be a packaging conflict due to how meta packages are handled, if there is an error when installing the tools package try to install it first from apt:
```bash
sudo apt-get install -y linux-tools-rockchip linux-tools-6.1.0-1019-rockchip
```

Joshua-Riek, 2024-07-05:
> Once I finish the perfomance tuning, I would be happy to contribute the updated driver back. What works best for you? A patch file would probably be the easiest as it would also (likely) apply to a newer 6.1.x version. If so, I can add a patch folder, and post the patches there with release notes.

If you can send a pull request with your changes to https://github.com/Joshua-Riek/linux-rockchip I would be happy to review and merge them into the kernel. I would like to ask for some clean / informative commit messages (no need for a paragraph or anything, a simple sentence works perfectly). This will help in the future when the kernel must be rebased on-top of the next Rockchip SDK.

ewaldc, 2024-07-05:
Mega thanks for such a fast answer.  Impressive how you got that all working, quite sad that the ubuntu kernel build process comes without documentation...
Pull request is fine, it can be turned into a patch anyhow.
`sudo apt-get install -y linux-tools-rockchip linux-tools-6.1.0-1019-rockchip` works to install `linux-tools-rockchip` but as a side effect, it pulls the whole official `6.1.0-1019` kernel.  By itself that is not an issue as my packages have unique naming (thanks to the `+xxx` tag), it just requires a small post-process as the official kernel ends up being the first/default in the uboot menu.  But to run `perf`, it still demands `linux-tools-6.1.0-1019+ewald-rockchip`.
Today I will try your `fakeroot debian/rules clean binary` suggestion on fresh copy as I use cross compiling.

ewaldc, 2024-07-06:
Unfortunately it's not working.  The debian rules/makefile don't support cross compiling the tools packages.  I managed the get the extra modules package to build though. Had to make a few changes to the rules/makefiles and delete the contents of `dkms-versions` (which contains just two modules) as the build was complaining about duplicate `zfs`.   In the end, there are three possible solutions IMHO:
- drop the tags (e.g. `+ewald`) in the package names, and leverage the `linux-tools-rockchip linux-tools-6.1.0-1019-rockchip` that is already in the ppa. This works when using the same base kernel, since the dependencies will be met and the tools are independent of the kernel config.  Overwriting the official `6.1.0-1019` kernel with custom packages of the same name is a quick but rather dirty fix, which also does not allow to run both kernels side by side.
- keep the tags and repackage `linux-tools-6.1.0-1019-rockchip` to `linux-tools-6.1.0-1019+ewald-rockchip`, so the dependencies are met.  Since both packages contain the exact same files, it can get equally messy e.g. when removing either kernel.
- keep the tags, build only the base custom packages and leverage/install the existing `linux-tools-6.1.0-1019-rockchip`.  At first glance this is the cleanest solution.  The only disadvantage is that the tagged kernel comes into second place, so it would require some global variable to `/etc/kernel/postinst.d/zz-u-boot-update` and/or `u-boot-update` to set the u-boot default to l1 instead of l0.

closed 2024-07-11 by ewaldc